espressif/esp32-camera #463: esp32-camera build problems help please - first trial od ESP

SoniaZotz, 2022-10-14:
Building in: E:\esp-idf-4.4.2\components\esp32-camera\build
Configuring in: E:\esp-idf-4.4.2\components\esp32-camera\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=E:\esp-idf-4.4.2\esp-idf-v4.4.2\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 E:\esp-idf-4.4.2\components\esp32-camera
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at E:/esp-idf-4.4.2/esp-idf-v4.4.2/tools/cmake/utilities.cmake:375 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-mlongcalls;-Wno-frame-address;].
Call Stack (most recent call first):
  E:/esp-idf-4.4.2/esp-idf-v4.4.2/tools/cmake/toolchain-esp32.cmake:10 (remove_duplicated_flags)
  C:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at E:/esp-idf-4.4.2/esp-idf-v4.4.2/tools/cmake/utilities.cmake:375 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-mlongcalls;-Wno-frame-address;].
Call Stack (most recent call first):
  E:/esp-idf-4.4.2/esp-idf-v4.4.2/tools/cmake/toolchain-esp32.cmake:12 (remove_duplicated_flags)
  C:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at E:/esp-idf-4.4.2/esp-idf-v4.4.2/tools/cmake/utilities.cmake:375 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-mlongcalls;].
Call Stack (most recent call first):
  E:/esp-idf-4.4.2/esp-idf-v4.4.2/tools/cmake/toolchain-esp32.cmake:14 (remove_duplicated_flags)
  C:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:87 (register_component):-- Configuring incomplete, errors occurred!

See also "E:/esp-idf-4.4.2/components/esp32-camera/build/CMakeFiles/CMakeOutput.log".
  Unknown CMake command "register_component".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.23)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.

Build complete (0 errors, 0 warnings): E:\esp-idf-4.4.2\components\esp32-camera\build
Total time taken to build the project: 6,848 ms

me-no-dev, 2022-10-14:
your project is not properly setup. Please check ESP-IDF's documentation on how to start a project and what it needs in it's configuration files.

SoniaZotz, 2022-10-15:
using esp-idf-4.4.2/
Building in: C:\Espressif\frameworks\**esp-idf-v4.4.2**\workspace\esp32-camera-master\build
Configuring in: C:\Espressif\frameworks\esp-idf-v4.4.2\workspace\esp32-camera-master\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=E:\esp-idf-v4.4.2\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 C:\Espressif\frameworks\esp-idf-v4.4.2\workspace\esp32-camera-master
-- The C compiler identification is GNU 8.4.0

Unknown CMake command "register_component". 

[SP-IDF Programming Guide ](https://docs.espressif.com/projects/esp-idf/en/v3.2.2/index.html)Logo
v3.2.2
has  reference to

Minimal Component CMakeLists

The minimal component CMakeLists.txt file is as follows:

set(COMPONENT_SRCS "foo.c")
set(COMPONENT_ADD_INCLUDEDIRS "include")
register_component()

but it is not found in esp-idf-4.4.2/

SoniaZotz, 2022-10-29:
> Building in: E:\esp-idf-4.4.2\components\esp32-camera\build Configuring in: E:\esp-idf-4.4.2\components\esp32-camera\build cmake -G Ninja [...]

closed 2022-10-29 by SoniaZotz

ziyan0302, 2024-07-24:
hi, the error shows that there is no CMakeLists.txt looks like this  in the current location:
`cmake_minimum_required(VERSION 3.5)

set(COMPONENTS main)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(camera_example)
`
I execute 'idf.py build' at esp32-camera/examples/camera_example. Here is where the project locates.