espressif/esp32-camera #464: Using OV2640 for raw Bayer data

Dwight-Choi, 2022-10-17:
Hi. 
I've been using ESP32-CAM with OV2640 for my project for a while.
And for better image processing, I need to get bayer raw data(data from the image sensor of camera). According to the data sheet of OV2640, 8-/10-byte raw image output is supported. And there are some idea(from ESP32 forum) that I can set PIX_FORMAT as raw should be working for this. But, when I excute the code, there was error like "not supported image format".
So, I really need to get raw data from the camera at this time. Any idea related this are really helpful to me.
1. PIXFORMAT_RAW is not supported for ESP32-CAM?
2. If not supported, Is there other way to get raw data ? like setting register of camera or other way?
Thank you in advance.

WangYuxin-esp, 2022-10-19:
OV2640 does support the output of RAW data, but it is not implemented in the driver at present. Because the PCLK supported by ESP32 is not large, ESP32 cannot support large resolution when the sensor outputs RAW data. As far as I know, ESP32S3 performs better than others in camera related projects, so I recommend you to use ESP32S3 for development. In addition, it seems that this sensor has been discontinued, so it is recommended that you choose a sensor in production to complete your project.

Dwight-Choi, 2022-10-19:
Thank you for your reply, but there is one thing that I can't understand. As I know, raw data has only 8 bits per each pixels. Then, I assume that the size of the data is not matter. So, this problem caused by not enough hardware specification of ESP32-CAM?

WangYuxin-esp, 2022-10-20:
This problem is not a limitation caused by hardware, but a purely software problem. To output RAW data, you need to rewrite the register list of OV2640.

Dwight-Choi, 2022-10-20:
Understand. Thank you for your info. Then, is there any reference or manual to rewrite the register? Or is there no plan to update the library? That's because users of ESP32-CAM can misunderstand they can get 8-/10-bit raw output with OV2640 according to your information in the readme.

closed 2022-12-28 by github-actions[bot]