espressif/esp32-camera #465: your project is not properly setup. Please check ESP-IDF's documentation on how to start a project and what it needs in it's configuration files.

SoniaZotz, 2022-10-22:
your project is not properly setup. Please check ESP-IDF's documentation on how to start a project and what it needs in it's configuration files.

_Originally posted by @me-no-dev in https://github.com/espressif/esp32-camera/issues/463#issuecomment-1278649322_

SoniaZotz, 2022-10-22:
are there any examples of how this should be set up please help 

- i imported a copy to the workspace
- tried to compile the project and I get  errors 
- have no idea what I should do next
- other build examples fromthe samples buildppperly
thank you

me-no-dev, 2022-10-24:
You can have a look [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-cmakelists-files) for more info

closed 2022-10-24 by me-no-dev