espressif/esp32-camera #466: PlatformIO espressif/esp32-camera@^2.0.0 linker error on Ubuntu

shehabattia96, 2022-10-24:
Recently updated my project to use `espressif/esp32-camera@^2.0.0` from `espressif/esp32-camera@^1.0.0`, but now linking fails. Reverting back to `espressif/esp32-camera@^1.0.0` makes build succeed.

Notes:

- Kconfig.projbuild in `src/` is soft linked to the one in `.pio/libdeps/esp32-camera` with absolute paths.
- git cloning the repo and using `-I../lib/esp32-camera` in build_flags as suggested in the README does not make a difference. i.e. same error.
- Reverting to `espressif/esp32-camera@^1.0.0` makes the project build.

```
; platformio.ini
[env:aithink_esp32]
build_type = debug
platform = espressif32
framework = espidf
board = esp32cam
upload_protocol = esptool
monitor_speed = 115200
board_build.partitions = partitions_singleapp.csv
build_flags = 
	-DCONFIG_RTCIO_SUPPORT_RTC_GPIO_DESC
	-DCONFIG_ESP32_SPIRAM_SUPPORT
	-DCONFIG_SCCB_CLK_FREQ
lib_deps = 
	espressif/esp32-camera@^2.0.0

```

```
 *  Executing task: platformio run 

Processing aithink_esp32 (platform: espressif32; framework: espidf; board: esp32cam)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (4.4.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-ninja @ 1.7.1 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32-camera @ 2.0.0
Building in debug mode
Linking .pio/build/aithink_esp32/firmware.elf
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(esp_camera.o):(.literal.camera_probe+0x14): undefined reference to `camera_enable_out_clock'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(esp_camera.o):(.literal.camera_probe+0x18): undefined reference to `camera_disable_out_clock'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(esp_camera.o): in function `camera_probe':
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/esp_camera.c:138: undefined reference to `camera_enable_out_clock'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/esp_camera.c:169: undefined reference to `camera_disable_out_clock'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/esp_camera.c:184: undefined reference to `camera_disable_out_clock'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(esp_camera.o): in function `esp_camera_deinit':
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/esp_camera.c:211: undefined reference to `camera_disable_out_clock'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_start_frame+0x4): undefined reference to `ll_cam_start'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_start_frame+0x8): undefined reference to `ll_cam_do_vsync'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_dma_config+0x38): undefined reference to `ll_cam_dma_sizes'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_dma_config+0x3c): undefined reference to `ll_cam_get_dma_align'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_init+0x14): undefined reference to `ll_cam_set_pin'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_init+0x18): undefined reference to `ll_cam_config'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_stop+0x0): undefined reference to `ll_cam_vsync_intr_enable'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o):(.literal.cam_stop+0x4): undefined reference to `ll_cam_stop'
[... 11 lines cut ...]
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:315: undefined reference to `ll_cam_set_pin'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:317: undefined reference to `ll_cam_config'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o): in function `cam_stop':
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:329: undefined reference to `ll_cam_vsync_intr_enable'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:329: undefined reference to `ll_cam_stop'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o): in function `cam_deinit':
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:425: undefined reference to `ll_cam_deinit'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o): in function `cam_config':
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:343: undefined reference to `ll_cam_set_sample_mode'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:374: undefined reference to `ll_cam_init_isr'
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o): in function `cam_start':
[... cut ...]
/home/me/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/aithink_esp32/libb66/libesp32-camera.a(cam_hal.o): in function `cam_take':
/home/me/projects/esp32-cam-mqtt/.pio/libdeps/aithink_esp32/esp32-camera/driver/cam_hal.c:464: undefined reference to `ll_cam_memcpy'
collect2: error: ld returned 1 exit status
*** [.pio/build/aithink_esp32/firmware.elf] Error 1
```

Surprisingly, these "undefined reference" messages are yellow, not red. Also, these warnings do NOT show up in 1.0.0.

shehabattia96, 2022-10-24:
Also to note: building the examples instead of my own code with 2.0.0 results in the same linking error.

shehabattia96, 2022-10-24:
using Espressif32 platform v4.4.0

shehabattia96, 2022-10-24:
fyi, Upgrading to Espressif32 platform v5.2.0 complains that `CONFIG_SCCB_CLK_FREQ` is redefined, but removing it from my platform.ini build flags makes that error go away, and I'm stuck with the same linking error.

me-no-dev, 2022-10-24:
I will suggest that you ask Platformio folks on how to setup additional IDF components. It seems to me that something is not OK in the build process. We recommend ESP-IDF v4.4 (not sure what that corresponds to in PIO)

shehabattia96, 2022-10-24:
> I will suggest that you ask Platformio folks on how to setup additional IDF components.

Which components should I be looking at? I am not aware that I am missing any dependencies.

> We recommend ESP-IDF v4.4 (not sure what that corresponds to in PIO)

From logs: PLATFORM: Espressif 32 (4.4.0)
From my testing, there is no difference between building with 4.4.0 and 5.2.0.

Jason2866, 2022-11-11:
Probably your camera src code needs to be udated to be compatible with v2

AgainPsychoX, 2023-01-03:
You might want to check out https://github.com/platformio/platformio-docs/issues/281 - you can make it work nicely, the PlatformIO docs are just outdated I guess...

shehabattia96, 2023-01-03:
Interesting, I'll give your suggestion a try and get back to you @AgainPsychoX. Thanks!

closed 2023-02-08 by shehabattia96