espressif/esp32-camera #467: 请问用ov5640摄像头发热严重是什么原因

chengxg, 2022-10-24:
esp32cam用ov5640摄像头发热严重, 而且清晰度很差, 是什么原因

WangYuxin-esp, 2022-10-25:
发热问题为摄像头本身的品质问题。清晰度问题可能为品质和参数导致。

chengxg, 2022-10-27:
> 发热问题为摄像头本身的品质问题。清晰度问题可能为品质和参数导致。

但是我用一个K210的板子来驱动, 就没有这个问题

WangYuxin-esp, 2022-10-27:
这个 OV5640 的线序和 ESP32cam 的线序匹配吗？发热问题应该和 sensor 工作的频率有关，工作的频率高则容易发热。总体来看，应该是 sensor 的寄存器配置不同，导致 sensor 的工作频率、成像质量的差异。不管是 K210还是 ESP32 其实都是配置 sensor，然后从 sensor 接收数据，一定程度上，它们只是数据接收器。

chengxg, 2022-10-27:
> 
我就是从某宝上买的OV5640, 都是dvp接口, 我看线序跟ov2640的是一样的, 应该是驱动有点问题

WangYuxin-esp, 2022-10-27:
是的，应该是对 ov5640 写入的寄存器的值不同。你可以替换掉 ov5640.c 中 reset() 中写的寄存器值看看是否有改善。

chengxg, 2022-10-27:
> 是的，应该是对 ov5640 写入的寄存器的值不同。你可以替换掉 ov5640.c 中 reset() 中写的寄存器值看看是否有改善。

感谢指点, 等我试试

closed 2022-11-01 by chengxg

mglwolf, 2023-05-15:
> reset()

能否请教一下reset()中应该修改为什么值比较合适？是以下这段吗？谢谢
static int reset(sensor_t *sensor)
{
    //dump_regs(sensor);
    vTaskDelay(100 / portTICK_PERIOD_MS);
    int ret = 0;
    // Software Reset: clear all registers and reset them to their default values
    ret = write_reg(sensor->slv_addr, SYSTEM_CTROL0, 0x82);
    if(ret){
        ESP_LOGE(TAG, "Software Reset FAILED!");
        return ret;
    }

WangYuxin-esp, 2023-05-15:
ESP32 通过 I2C 总线配置 sensor 的寄存器。在 reset() 函数中配置了 sensor 的关键的寄存器。
对于 ov5640，应该主要关注 `ret = write_regs(sensor->slv_addr, sensor_default_regs);` 中的 sensor_dafault_regs数组，其包含了 sensor 需要的绝大部分寄存器配置。

NoNameCanUse9, 2023-07-30:
你好，我想问一下，那个寄存器地址应该怎么设置

WangYuxin-esp, 2023-07-31:
> 你好，我想问一下，那个寄存器地址应该怎么设置
请新建一个issue 讨论新的问题。