espressif/esp32-camera #468: ov2640 in QQVGA frame size always cause "cam_hal: NO-EOI" and "cam_hal: FB-OVF"

MasterFeige, 2022-10-25:
==============================================
ov2640 configuration  (pins config is right,and will be omitted):
```c        
#define CONFIG_OV2640_SUPPORT                       1
#define CONFIG_SCCB_CLK_FREQ                        100000
#define CONFIG_CAMERA_CONVERTER_ENABLED             1
#define LCD_CAM_CONV_FULL_RANGE_ENABLED             0
#define CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX           32768

        .jpeg_quality     = 10,
        .fb_count         = 2,
        .pixel_format     = PIXFORMAT_JPEG,
        .frame_size       = FRAMESIZE_QQVGA,// 160*120
        .xclk_freq_hz     = 20000000
```
camera thread function is:
```c
static void camera_capture_thread(void*param)
{
    uint8_t  fps = 20;
    uint32_t time_gaps = (1000/fps);
    
    uint32_t len;
    bool      ret;
    uint8_t*  rgbbuf = heap_caps_malloc(160*120*3,MALLOC_CAP_SPIRAM);
    while(1){

        
        uint64_t total_time = esp_timer_get_time();

        user_fb = esp_camera_fb_get();
        if(!user_fb){
            vTaskDelay(100);
            continue;
        }

        ret = fmt2rgb888(user_fb->buf,user_fb->len,user_fb->format,rgbbuf);
	if(ret==ESP_OK){
            ESP_LOGI(TAG,"decode success!!");
        }else{
            ESP_LOGI(TAG,"decode error!!");
        }
        esp_camera_fb_return(user_fb);

        vTaskDelay(100);
    }
}
```
**output info as:**
```shell
I (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (27) boot: compile time 15:27:10
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1d1bch (119228) map
I (153) esp_image: segment 1: paddr=0002d1e4 vaddr=3ffb0000 size=02e34h ( 11828) load
I (157) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=85e4ch (548428) map
I (357) esp_image: segment 3: paddr=000b5e74 vaddr=3ffb2e34 size=00de4h (  3556) load
I (358) esp_image: segment 4: paddr=000b6c60 vaddr=40080000 size=1ae0ch (110092) load
I (408) esp_image: segment 5: paddr=000d1a74 vaddr=50000000 size=00010h (    16) load
I (421) boot: Loaded app from partition at offset 0x10000
I (421) boot: Disabling RNG early entropy source...
I (433) psram: This chip is ESP32-D0WD
I (435) spiram: Found 64MBit SPI RAM device
I (435) spiram: SPI RAM mode: flash 40m sram 40m
I (438) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (445) cpu_start: Pro cpu up.
I (449) cpu_start: Starting app cpu, entry point is 0x4008155c
0x4008155c: call_start_cpu1 at xxx/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (1340) spiram: SPI SRAM memory test OK
I (1348) cpu_start: Pro cpu start user code
I (1348) cpu_start: cpu freq: 240000000
I (1348) cpu_start: Application information:
I (1351) cpu_start: Project name:     camera
I (1356) cpu_start: App version:      09be8b6-dirty
I (1362) cpu_start: Compile time:     Oct 25 2022 16:05:13
I (1368) cpu_start: ELF file SHA256:  e6188e592a67e104...
I (1374) cpu_start: ESP-IDF:          v4.4.2-dirty
I (1380) heap_init: Initializing. RAM available for dynamic allocation:
I (1387) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1393) heap_init: At 3FFB85C0 len 00027A40 (158 KiB): DRAM
I (1399) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1405) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1412) heap_init: At 4009AE0C len 000051F4 (20 KiB): IRAM
I (1418) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1427) spi_flash: detected chip: generic
I (1431) spi_flash: flash io: dio
I (1436) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1525) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (1525) cam_hal: cam init ok
I (1525) sccb: pin_sda 26 pin_scl 27
I (1535) sccb: sccb_i2c_port=0

I (1535) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1575) camera: Detected camera at address=0x30
I (1575) camera: Detected OV2640 camera
I (1575) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1655) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 0
I (1655) cam_hal: Allocating 3840 Byte frame buffer in PSRAM
I (1655) cam_hal: Allocating 3840 Byte frame buffer in PSRAM
I (1665) cam_hal: cam config ok
I (1665) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
I (1755) wifi:wifi driver task: 3ffcc5e4, prio:23, stack:6656, core=0
I (1755) system_api: Base MAC address is not set
I (1755) system_api: read default base MAC address from EFUSE
cam_hal: EV-EOF-OVF
W (1775) cam_hal: NO-SOI
I (1775) wifi:wifi firmware version: eeaa27d
I (1775) wifi:wifi certification version: v7.0
I (1775) wifi:config NVS flash: enabled
I (1775) wifi:config nano formating: disabled
I (1785) wifi:Init data frame dynamic rx buffer num: 32
I (1785) wifi:Init management frame dynamic rx buffer num: 32
I (1795) wifi:Init management short buffer num: 32
I (1795) wifi:Init static tx buffer num: 16
W (1795) cam_hal: FBcam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
-OVF
I (1795) wifi:Init tx cache buffer num: 32
I (1815) wifi:Init static rx buffer size: 1600
I (1815) wifi:Init static rx buffer num: 10
I (1815) wifi:Init dynamic rx buffer num: 32
I (1825) wifi_init: rx ba win: 6
I (1825) wifi_init: tcpip mbox: 32
W (1825) cam_hal: FB-OVF
W (1835) cam_hal: FB-OVF
W (1835) cam_hal: FB-OVF
I (1835) wifi_init: udp mbox: 6
W (1845) cam_hal: FB-OVF
I (1845) wifi_init: tcp mbox: 6
W (1845) cam_hal: NO-SOI
I (1855) wifi_init: tcp tx win: 5744
I (1855) wifi_init: tcp rx win: 5744
I (1865) wifi_init: tcp mss: 1440
I (1865) wifi_init: WiFi/LWIP prefer SPIRAM
W (1865) cam_hal: FB-OVF
W (1875) cam_hal: FB-OVF
W (1875) cam_hal: FB-OVF
W (1885) cam_hal: FB-OVF
W (1885) cam_hal: FB-OVF
I (1885) wifi_init: WiFi IRAM OP enabled
I (1895) wifi_init: WiFi RX IRAM OP enabled
I (1895) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1995) wifi:mode : sta (c8:f0:9e:9b:26:88)
I (1995) wifi:enable tsf
I (1995) wifi:Set ps type: 0

I (1995) main: Connecting to "xxx",password : "xxx"

I (1995) main: event handler hanppend===>SYSTEM_EVENT_STA_START
I (2005) main: Connected
I (2015) main: start stream server
I (2015) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
I (2025) wifi:state: init -> auth (b0)
I (2025) wifi:state: auth -> assoc (0)
I (2135) wifi:state: assoc -> run (10)
I (2225) wifi:connected with ChinaNet-A4H7, aid = 1, channel 9, BW20, bssid = ce:08:fb:1c:b6:fc
I (2225) wifi:security: WPA2-PSK, phy: bgn, rssi: -48
I (2235) wifi:pm start, type: 0

W (2255) wifi:<ba-add>idx:0 (ifx:0, ce:08:fb:1c:b6:fc), tid:0, ssn:569, winSize:64
I (2325) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3245) esp_netif_handlers: sta ip: 192.168.x.xx6, mask: 255.255.255.0, gw: 192.168.2.1
I (3245) main: event handler hanppend===>SYSTEM_EVENT_STA_GOT_IP
I (3245) main: http://192.168.x.xx6
I (4005) main: start stream handle
W (4015) cam_hal: NO-EOI
W (4015) cam_hal: NO-EOI
W (4025) cam_hal: FB-OVF
```

**idf version is 4.4.2**

**I use QQVGA configuration in Ardunio's CameraWebServer and it works fine**

## My questions are: 
### 1. Why the same configuration, the current library will report an error, but the Arduino will not?
### 2. I tried to lower the frequency of xclk, but to no avail, is there any solution?
### 3.When the cam_take method encounters an error, it will make a recursive call, which will cause a stack overflow. Will the official fix it?
### 4.What does the jpg_scale_t parameter in the jpg2rgb888 method mean?
Looking forward to your answer, thank you

WangYuxin-esp, 2022-10-26:
Whether the sensor can work normally depends on the list of registers written by esp32 to the sensor's registers during initialization. Therefore, the difference may be caused by their register lists. In addition, the last parameter `jpg_scale_t` of `jpg2rgb888` represents scaling. For example, if scaling is 2 * 2, the length and width of the final data will be reduced to half of the original length and width.
In this library, the resolution of QVGA and VGA should work normally. You can try [it](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/pic_server).
For the stack overflow problem, you can try to increase the size of the stack for which the problem occurs.

MasterFeige, 2022-10-27:
@WangYuxin-esp Thank you for your answer.I used AI-Thinker 'camera library' and it worked well.' the resolution of QVGA and VGA should work normally' I tried ,but it doesn't work.
In my opinion,'For the stack overflow problem' I think it should be return NULL.
Thanks again for your answer. My question has been solved.

closed 2022-10-27 by MasterFeige