espressif/esp32-camera #471: Custom camera integration

dumarjo, 2022-10-27:
Hi,

We are integrating a new camera. we have made some modification to the library. Now the communication on I2C is working. The camera have the modclock integrated. I would like to know if it's possible to use this library if the ESP32S3 don't provide the modclck(xclk) ?

We only need VSYNC, HSYNC, PCLK and D0:7

We are integrating this camera https://www.espros.com/downloads/01_Chips/Datasheet_epc635.pdf

Regards

me-no-dev, 2022-10-28:
This should not be a problem, as long as the pixel clock is within the capabilities of the ESP camera bus. Frequencies of 10MHz or lower would be optimal.

jonathandumaresq, 2022-10-28:
We are looking for a 5fps. From the datasheet, I think the dclk(pclk) is at 20 mhz. I will lokk if we can get it lower

Should we set the xclk to 20 Mhz if the pclk is 10Mhz ?

dumarjo, 2022-10-28:
We are looking for a 5fps. From the datasheet, I think the dclk(pclk) is at 20 mhz. I will look if we can get it lower

Should we set the xclk to 20 Mhz if the pclk is 10Mhz ?

me-no-dev, 2022-10-28:
Since you are providing XCLK externally, it would not matter what is the frequency set in the driver. 20MHz PCLK should work, but it's better if you can half it. Would still be plenty for 5FPS

dumarjo, 2022-10-28:
Ok thanx.

closed 2022-10-28 by dumarjo