espressif/esp32-camera #472: Complier Errors in the Examples

SoniaZotz, 2022-11-06:
### Trying to compile examples without success  am I missing something 

**BMP-HTTP-Capture**

E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c: In function 'bmp_httpd_handler':
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:14:9: error: implicit declaration of function 'ESP_LOGE'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
         ESP_LOGE(TAG, "Camera capture failed");
         ^~~~~~~~
         ESP_OK
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:14:18: error: 'TAG' undeclared (first use in this function)
         ESP_LOGE(TAG, "Camera capture failed");
                  ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:14:18: note: each undeclared identifier is reported only once for each function it appears in
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:34:5: error: implicit declaration of function 'ESP_LOGI'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
     ESP_LOGI(TAG, "BMP: %uKB %ums", (uint32_t)(buf_len/1024), (uint32_t)((fr_end - fr_start)/1000));
     ^~~~~~~~
     ESP_OK
cc1.exe: some warnings being treated as errors

**BMP-HTTP-Capture**

E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c: In function 'bmp_httpd_handler':
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:14:9: error: implicit declaration of function 'ESP_LOGE'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
         ESP_LOGE(TAG, "Camera capture failed");
         ^~~~~~~~
         ESP_OK
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:14:18: error: 'TAG' undeclared (first use in this function)
         ESP_LOGE(TAG, "Camera capture failed");
                  ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:14:18: note: each undeclared identifier is reported only once for each function it appears in
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/BMP-HTTP-Capture/main/main.c:34:5: error: implicit declaration of function 'ESP_LOGI'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
     ESP_LOGI(TAG, "BMP: %uKB %ums", (uint32_t)(buf_len/1024), (uint32_t)((fr_end - fr_start)/1000));
     ^~~~~~~~
     ESP_OK
cc1.exe: some warnings being treated as errors

**JPEG-HTTP-Stream**

E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/JPEG-HTTP-Stream/main/main.c: In function 'jpg_stream_httpd_handler':
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/JPEG-HTTP-Stream/main/main.c:29:13: error: implicit declaration of function 'ESP_LOGE'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
             ESP_LOGE(TAG, "Camera capture failed");
             ^~~~~~~~
             ESP_OK
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/JPEG-HTTP-Stream/main/main.c:29:22: error: 'TAG' undeclared (first use in this function)
             ESP_LOGE(TAG, "Camera capture failed");
                      ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/JPEG-HTTP-Stream/main/main.c:29:22: note: each undeclared identifier is reported only once for each function it appears in
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/JPEG-HTTP-Stream/main/main.c:67:9: error: implicit declaration of function 'ESP_LOGI'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
         ESP_LOGI(TAG, "MJPG: %uKB %ums (%.1ffps)",
         ^~~~~~~~
         ESP_OK
cc1.exe: some warnings being treated as errors

**Initialization**

E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:56:9: error: implicit declaration of function 'pinMode' [-Werror=implicit-function-declaration]
         pinMode(CAM_PIN_PWDN, OUTPUT);
         ^~~~~~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:56:31: error: 'OUTPUT' undeclared (first use in this function)
         pinMode(CAM_PIN_PWDN, OUTPUT);
                               ^~~~~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:56:31: note: each undeclared identifier is reported only once for each function it appears in
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:57:9: error: implicit declaration of function 'digitalWrite' [-Werror=implicit-function-declaration]
         digitalWrite(CAM_PIN_PWDN, LOW);
         ^~~~~~~~~~~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:57:36: error: 'LOW' undeclared (first use in this function)
         digitalWrite(CAM_PIN_PWDN, LOW);
                                    ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:63:9: error: implicit declaration of function 'ESP_LOGE'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
         ESP_LOGE(TAG, "Camera Init Failed");
         ^~~~~~~~
         ESP_OK
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:63:18: error: 'TAG' undeclared (first use in this function)
         ESP_LOGE(TAG, "Camera Init Failed");
                  ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c: In function 'camera_capture':
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:74:18: error: 'TAG' undeclared (first use in this function)
         ESP_LOGE(TAG, "Camera Capture Failed");
                  ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:78:5: error: implicit declaration of function 'process_image' [-Werror=implicit-function-declaration]
     process_image(fb->width, fb->height, fb->format, fb->buf, fb->len);
     ^~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors

me-no-dev, 2022-11-06:
seems that you are trying to compile Arduino code in ESP-IDF. Also you did not give any code to check your initialization. For the `LOG_x` errors, you are probably missing an `#include "esp_log.h"`

SoniaZotz, 2022-11-06:
the code is in this same repo https://github.com/espressif/esp32-camera trying to compile the examples

SoniaZotz, 2022-11-06:
Yes  want to run this in ESP_IDE   not Arduino  - how do I do this ?   trying to understand this process 

```
#include "esp_camera.h"
#include "esp_http_server.h"
#include "esp_timer.h"

#include "esp_log.h"
static const char* TAG = "BMP-HTTP-Capture";

esp_err_t bmp_httpd_handler(httpd_req_t *req){
    camera_fb_t * fb = NULL;
    esp_err_t res = ESP_OK;
    int64_t fr_start = esp_timer_get_time();

    fb = esp_camera_fb_get();
    if (!fb) {
        ESP_LOGE(TAG, "Camera capture failed");
        httpd_resp_send_500(req);
        return ESP_FAIL;
    }

    uint8_t * buf = NULL;
    size_t buf_len = 0;
    bool converted = frame2bmp(fb, &buf, &buf_len);
    esp_camera_fb_return(fb);
    if(!converted){
        ESP_LOGE(TAG, "BMP conversion failed");
        httpd_resp_send_500(req);
        return ESP_FAIL;
    }

    res = httpd_resp_set_type(req, "image/x-windows-bmp")
       || httpd_resp_set_hdr(req, "Content-Disposition", "inline; filename=capture.bmp")
       || httpd_resp_send(req, (const char *)buf, buf_len);
    free(buf);
    int64_t fr_end = esp_timer_get_time();
    ESP_LOGI(TAG, "BMP: %uKB %ums", (uint32_t)(buf_len/1024), (uint32_t)((fr_end - fr_start)/1000));
    return res;
}
```
Seems to compile clean  however the issue is that 

```
cmd.exe /C "cd . && E:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\BMP-HTTP-Capture.elf.rsp -o BMP-HTTP-Capture.elf  && cd ."
e:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x1c): undefined reference to `app_main'
e:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
E:/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/port_common.c:135: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

SoniaZotz, 2022-11-06:
_**Need help with this please**_

```
#include "esp_camera.h"

#include "esp_log.h"
static const char* TAG = "BMP-HTTP-Capture";

//WROVER-KIT PIN Map
#define CAM_PIN_PWDN    -1 //power down is not used
#define CAM_PIN_RESET   -1 //software reset will be performed
#define CAM_PIN_XCLK    21
#define CAM_PIN_SIOD    26
#define CAM_PIN_SIOC    27

#define CAM_PIN_D7      35
#define CAM_PIN_D6      34
#define CAM_PIN_D5      39
#define CAM_PIN_D4      36
#define CAM_PIN_D3      19
#define CAM_PIN_D2      18
#define CAM_PIN_D1       5
#define CAM_PIN_D0       4
#define CAM_PIN_VSYNC   25
#define CAM_PIN_HREF    23
#define CAM_PIN_PCLK    22

static camera_config_t camera_config = {
    .pin_pwdn  = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    .xclk_freq_hz = 20000000,//EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,//YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_UXGA,//QQVGA-UXGA, For ESP32, do not use sizes above QVGA when not JPEG. The performance of the ESP32-S series has improved a lot, but JPEG mode always gives better frame rates.

    .jpeg_quality = 12, //0-63, for OV series camera sensors, lower number means higher quality
    .fb_count = 1, //When jpeg mode is used, if fb_count more than one, the driver will work in continuous mode.
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY//CAMERA_GRAB_LATEST. Sets when buffers should be filled
};

esp_err_t camera_init(){
    //power up the camera if PWDN pin is defined
    if(CAM_PIN_PWDN != -1){
        pinMode(CAM_PIN_PWDN, OUTPUT);
        digitalWrite(CAM_PIN_PWDN, LOW);
    }

    //initialize the camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
    }

    return ESP_OK;
}

esp_err_t camera_capture(){
    //acquire a frame
    camera_fb_t * fb = esp_camera_fb_get();
    if (!fb) {
        ESP_LOGE(TAG, "Camera Capture Failed");
        return ESP_FAIL;
    }
    //replace this with your own function
    process_image(fb->width, fb->height, fb->format, fb->buf, fb->len);

    //return the frame buffer back to the driver for reuse
    esp_camera_fb_return(fb);
    return ESP_OK;
}
```

```

E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c: In function 'camera_init':
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:61:9: error: implicit declaration of function 'pinMode' [-Werror=implicit-function-declaration]
         pinMode(CAM_PIN_PWDN, OUTPUT);
         ^~~~~~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:61:31: error: 'OUTPUT' undeclared (first use in this function)
         pinMode(CAM_PIN_PWDN, OUTPUT);
                               ^~~~~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:61:31: note: each undeclared identifier is reported only once for each function it appears in
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:62:9: error: implicit declaration of function 'digitalWrite' [-Werror=implicit-function-declaration]
         digitalWrite(CAM_PIN_PWDN, LOW);
         ^~~~~~~~~~~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:62:36: error: 'LOW' undeclared (first use in this function)
         digitalWrite(CAM_PIN_PWDN, LOW);
                                    ^~~
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c: In function 'camera_capture':
E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/Initialization/main/main.c:83:5: error: implicit declaration of function 'process_image' [-Werror=implicit-function-declaration]
     process_image(fb->width, fb->height, fb->format, fb->buf, fb->len);
     ^~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
```

me-no-dev, 2022-11-07:
You can start with the example in this repo: https://github.com/espressif/esp32-camera/tree/master/examples

closed 2023-01-14 by github-actions[bot]