espressif/esp32-camera #473: NT99141 sensor can not work on ESP32-S3-EYE

heculess, 2022-11-09:
I use NT99141 sensor on official development boards ESP32-S3-EYE. But  I can't get any image from it. 
[image]

WangYuxin-esp, 2022-11-14:
This is very normal. It is caused by the incorrect configuration of the sensor. According to my observation, this sensor seems to work in the test mode by default, so it cannot plot normally. If you plan to use this sensor, you'd better take a look at the register configuration in this sensor.

closed 2023-01-21 by github-actions[bot]

Raven888888, 2024-10-02:
@WangYuxin-esp 请问register configuration要怎么设置才能使用NT99141呢？我也遇到一样的问题
谢谢

WangYuxin-esp, 2024-10-08:
Firstly, you can try using the example [here](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/pic_server) to configure the parameters. Please use: Xclk=20M，JPEG format、HD resolution.  
Secondly, the sensor defaults to loading register configurations from [here](https://github.com/espressif/esp32-camera/blob/master/sensors/private_include/nt99141_settings.h#L18).