espressif/esp32-camera #481: https://github.com/espressif/esp32-camera/blob/master/test/CMakeLists.txt

SoniaZotz, 2022-11-19:
**### > **Unknown CMake command "idf_component_register".****

```
df_component_register(SRC_DIRS .
                       PRIV_INCLUDE_DIRS .
                       PRIV_REQUIRES test_utils esp32-camera nvs_flash 
                       EMBED_TXTFILES pictures/testimg.jpeg pictures/test_outside.jpeg pictures/test_inside.jpeg)
 project(TEST)
 cmake_minimum_required(VERSION 3.23)

Building in: E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\test\build
Configuring in: E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\test\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=E:\Espressif\frameworks\esp-idf-v4.4.2\tools\cmake\toolchain-esp32s3.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32s3 E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\test
CMake Error at CMakeLists.txt:1 (idf_component_register):-- Configuring incomplete, errors occurred!

  Unknown CMake command "idf_component_register".

cmake --build . -- -v
Error: could not find CMAKE_PROJECT_NAME in Cache
Build complete (1 errors, 0 warnings): E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\test\build
Total time taken to build the project: 1,259 ms
```

WangYuxin-esp, 2022-11-24:
[Here](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera) are some examples that can be run directly. Maybe you can refer to the examples [here](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera).

SoniaZotz, 2022-11-26:
thank you for your help i will try it

closed 2023-02-02 by github-actions[bot]