espressif/esp32-camera #482: Compilation Errors and unresolved inclusions

RatTrap85, 2022-11-30:
I hate having to bug people online when I can't solve a problem but after 12 hours of banging my head against the wall I have no other choice.  

heres the situation.  I am trying to use the ESP32-camera code in a porject I am working on.  I have tried repeatedly over and over again different methods to add this to my project, and every time I get hundreds of errors that are all caused by the fact that the most of the #includes in the header files can't resolve.  I have modified the make files hundreds of different ways in an attempt to correct this.  I have downloaded and unzipped to add it,  I have cloned the git, I have tried just manually typing it all in.  No matter what I do I end up in the same situation.   

the kicker is that I can get it to compile as long as I don't open the either the driver/include/esp_camera.h or driver/include/sensor.h.
if I even open either one of those it stops compiling and gives those errors again.  

its like the header files aren't there, which of course they are, and it can find them as long as I don't try to open them.  
[image]

could someone show me mercy and tell me how in the hell to use this.   I am not new to coding but am very new to ESP-IDF and am really missing how to simply add a library in this IDE.

thank you so much for any help.

igrr, 2022-11-30:
If you are using Eclipse with IDF plugin (or the "Espressif-IDE"), the simplest way is probably as follows:
1. In project explorer, right-click anywhere and choose "New" > "Espressif IDF project"
2. Enter the project name (e.g. "camera-test"), click "Finish"
3. Right-click the project in the project explorer, choose "Install ESP-IDF components"
4. In the window with a list of components which opens, find esp32-camera and click the "Install" button next to it.
5. Open main.c, add `#include "esp_camera.h"`
6. Click the "hammer" icon on the top left to build the project.

igrr, 2022-11-30:
Another option, without using the "Install ESP-IDF components" option and by downloading the component manually using Git:
1. Create a new IDF project, same as in the previous approach
2. Open a terminal, go to the directory of the newly created project
3. Create a `components` directory
4. Go into the `components` directory
5. Clone esp32-camera: `git clone https://github.com/espressif/esp32-camera.git`
6. Close the terminal
7. Open main/CMakeLists.txt file of the project
8. Add `esp32-camera` after `REQUIRES`
9. Add `#include "esp_camera.h` in main/main.c
10. Click the hammer icon to build the project

RatTrap85, 2022-12-01:
thank you sir.  I will give those two a shot.  if I may ask how do I use this type of resource in an existing project.  I hate to use the example of Arduino or Atmel studio, but I am thinking of using it more like a library than just using their example code.  I am sorry if I sound stupid but I'm still very much learning the IDE.  I feel like a moron for not understanding how to add it as a resource.

igrr, 2022-12-01:
From your screenshot, it looks like you got it almost right with your existing project!

The problem seems to be that the esp32-camera directory is in the root of your project directory. However, ESP-IDF by default expects project components to be present in the `components/` subdirectory. You can read more about the structure of ESP-IDF build system projects here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#example-project.

So, based on this, I think it's not a problem to use either of the above methods with an existing project:

1. Prerequisite: make sure you can build your project in the IDE without any camera-related stuff added.
2. Add the esp32-camera component to the project:
   - Either using the automated method, via the "Install ESP-IDF components", as in method 1
   - Or manually, by cloning esp32-camera repository into components/esp32-camera subdirectory of the project. In this case, you will also have to manually add a dependency on the esp32-camera component in CMakeLists.txt, as explained in the previous reply.
3. Add `#include "esp_camera.h"` to your main source file.
4. Try to build.

RatTrap85, 2022-12-01:
thank you so much for the help  I really appreciate it.  I really need to take a class on CMake files because they don't make any sense to me at all.  I suppose that is the problem with trying to teach myself.  I did as you instructed  

this is the result of trying to do the install 
[image]

and here is the result of trying it with the Git clone into a components subdirectory, but there is no doubt in my mind that I have screwed up on the make file.

[image]

and here is my attempt to add the components to the make file.
[image]

RatTrap85, 2022-12-01:
I keep searching online for a coherent example of the make file but just get redirected back to the espresssif resources which despite my most sincere efforts I just don't understand

igrr, 2022-12-01:
Could you please also show the contents of main/CMakeLists.txt?

RatTrap85, 2022-12-01:
certainly

[image]

igrr, 2022-12-01:
From what I see in the screenshots, you have done everything correctly!

I have tried to repeat all the steps with IDF v4.4.2 (which you are using) and I was still able to get the project built. That's really odd. I can't figure out what is the difference that causes the build to fail on your side.

When you build the project, do you get the same error message about the missing esp_camera.h in the "Console" window at the bottom? Or is the error only shown in the Problems tab?

Could you please attach the following files from the 'build' directory after an unsuccessful build:
- build/compile_commands.json
- build/build.ninja
- build/project_description.json

We'll try to figure out what's going on based on these files.

RatTrap85, 2022-12-02:
sorry for the delay.  I got it working for a while but then lost internet.

I did both the git clone and the managed components and it worked for a while.  Then a moment ago it stopped working, I did a quick git clone and makefile change and it just started working.  idk

I don't know how to get the error again. but if it does happen I will send you the files per your request

RatTrap85, 2022-12-02:
maybe it will happen again and I can get you the files to find and crush that bug.  Thank you for all of your help and for not making me feel anymore like an idiot than I already do.  Sorry that it stopped failing before I could get those bug files for you.

igrr, 2022-12-02:
No worries, at least it works for you now, which is probably not a bad outcome. If this happens again, please reopen the issue and @-mention me, I'll take a look.

closed 2022-12-02 by igrr