espressif/esp32-camera #485: i2c_ll_disable_intr_mask Core  0 panic'ed (LoadProhibited) when calling esp_camera_init multiple times

petercipov, 2022-12-08:
## Problem
```
I (24) boot: ESP-IDF v5.1-dev-2066-g7869f4e151 2nd stage bootloader
I (24) boot: compile time Dec  7 2022 20:47:47
I (25) boot: chip revision: v0.1
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0f9b0h ( 63920) map
I (109) esp_image: segment 1: paddr=0001f9d8 vaddr=3fc94500 size=00640h (  1600) load
I (110) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=30fd0h (200656) map
I (147) esp_image: segment 3: paddr=00050ff8 vaddr=3fc94b40 size=037d4h ( 14292) load
I (150) esp_image: segment 4: paddr=000547d4 vaddr=40374000 size=1040ch ( 66572) load
I (172) boot: Loaded app from partition at offset 0x10000
I (172) boot: Disabling RNG early entropy source...
I (173) octal_psram: vendor id    : 0x0d (AP)
I (178) octal_psram: dev id       : 0x02 (generation 3)
I (183) octal_psram: density      : 0x03 (64 Mbit)
I (189) octal_psram: good-die     : 0x01 (Pass)
I (194) octal_psram: Latency      : 0x01 (Fixed)
I (199) octal_psram: VCC          : 0x01 (3V)
I (204) octal_psram: SRF          : 0x01 (Fast Refresh)
I (210) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (216) octal_psram: BurstLen     : 0x01 (32 Byte)
I (222) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (228) octal_psram: DriveStrength: 0x00 (1/1)
W (233) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (244) esp_psram: Found 8MB PSRAM device
I (248) esp_psram: Speed: 80MHz
I (252) cpu_start: Pro cpu up.
I (255) cpu_start: Starting app cpu, entry point is 0x40375548
0x40375548: call_start_cpu1 at /Users/petercipov/esp/esp-idf/components/esp_system/port/cpu_start.c:145

I (0) cpu_start: App cpu up.
I (583) esp_psram: SPI SRAM memory test OK
I (592) cpu_start: Pro cpu start user code
I (592) cpu_start: cpu freq: 240000000 Hz
I (592) cpu_start: Application information:
I (592) cpu_start: Project name:     main_test
I (592) cpu_start: App version:      38f24e5-dirty
I (592) cpu_start: Compile time:     Dec  7 2022 21:09:08
I (593) cpu_start: ELF file SHA256:  801fd3d13956245e...
I (593) cpu_start: ESP-IDF:          v5.1-dev-2066-g7869f4e151
I (593) cpu_start: Min chip rev:     v0.0
I (593) cpu_start: Max chip rev:     v0.99
I (593) cpu_start: Chip rev:         v0.1
I (594) heap_init: Initializing. RAM available for dynamic allocation:
I (594) heap_init: At 3FC98F40 len 000507D0 (321 KiB): D/IRAM
I (594) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (594) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (595) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (596) spi_flash: detected chip: generic
I (596) spi_flash: flash io: qio
I (596) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Running Camera driver init, deinit test...
I (597) s3 ll_cam: DMA Channel=4
I (597) cam_hal: cam init ok
I (597) sccb: pin_sda 4 pin_scl 5
I (597) sccb: sccb_i2c_port=1

I (597) gpio: GPIO[43]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[... 196 lines cut ...]
        sensor_t *s = esp_camera_sensor_get();
        s->set_framesize(s, size_bak);
    }

    return ret;
}

TEST_CASE("Camera driver init, deinit test", "[camera]")
{
    TEST_ESP_OK(init_camera(20000000, PIXFORMAT_RGB565, FRAMESIZE_QVGA, 2, SIOD_GPIO_NUM, -1));

    TEST_ESP_OK(esp_camera_deinit());
}

TEST_CASE("Camera driver init 2, deinit test", "[camera]")
{
    TEST_ESP_OK(init_camera(20000000, PIXFORMAT_RGB565, FRAMESIZE_QVGA, 2, SIOD_GPIO_NUM, -1));

    TEST_ESP_OK(esp_camera_deinit());
}
``` 

## Expected behaviour
core does not crash
camara is initialized/deinitialized multiple times if needed

v44r, 2023-01-06:
Same here... and I suppose this is the only way to change pixformat, because sensor->set_pixformat causes problems with the frame buffer size; either "cam_hal: FB-SIZE: 7680 != 153600" when going from RGB565 to JPEG or "cam_hal: NO-SOI." when going from JPEG to RGB565.

filipbudisa, 2023-01-24:
Hey there! Had the same problem and after some debugging, I think I found the source of the crash.

It seems esp_camera_deinit ends up with a call to i2c_driver_delete, which deletes the I2C driver's p_i2c_obj[i2c_num] object. As far as I can tell, the object isn't created again anywhere by the time esp_camera_init ends up at the i2c_param_config call, which tries to read from the now deleted p_i2c_obj[i2c_num]. This is on IDF v5.1.0 (commit 4c98bee8a4417d92a387ef2416be14ff1fdfa92a ).

A workaround is to manually initialize the i2c driver once, and pass the i2c port by camera_config.sccb_i2c_port. Something like this:

```C++
	ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0));

	i2c_config_t conf = {};
	conf.mode = I2C_MODE_MASTER;
	conf.sda_io_num = cam_pin_sda;
	conf.scl_io_num = cam_pin_scl;
	conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
	conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
	conf.master.clk_speed = CONFIG_SCCB_CLK_FREQ;
	ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_0, &conf));
```
Replace cam_pin_sda and cam_pin_scl with your i2c pins.
Camera config then needs to be changed like this:
```C++
	camera_config.pin_sccb_sda = -1;
	camera_config.pin_sccb_scl = -1;
	config.sccb_i2c_port = I2C_NUM_0;
```
pin_sccb_sda is set to -1 so the camera driver uses the already initialized port. Change I2C_NUM_0 to I2C_NUM_1 in both the camera config and i2c config if you wish to use the other i2c port.

petercipov, 2023-01-28:
@filipbudisa I have checked whether this is bug in esp-idf i2c library. Indeed it was. In latest master (5.1.x) branch it was already fixed.

https://github.com/espressif/esp-idf/issues/10163

closed 2023-01-28 by petercipov