espressif/esp32-camera #488: cam_hal: Failed to get the frame on time! and multiple cam_hal: EV-VSYNC-OVF

MattiaDallaCosta, 2022-12-15:
board: esp32-cam (a clone of the ai thinker one)

#define CAM_PIN_PWDN 32
#define CAM_PIN_RESET -1 
#define CAM_PIN_XCLK 0
#define CAM_PIN_SIOD 26
#define CAM_PIN_SIOC 27

#define CAM_PIN_D7 35
#define CAM_PIN_D6 34
#define CAM_PIN_D5 39
#define CAM_PIN_D4 36
#define CAM_PIN_D3 21
#define CAM_PIN_D2 19
#define CAM_PIN_D1 18
#define CAM_PIN_D0 5
#define CAM_PIN_VSYNC 25
#define CAM_PIN_HREF 23
#define CAM_PIN_PCLK 22

pin_pwdn = CAM_PIN_PWDN,
pin_reset = CAM_PIN_RESET,
pin_xclk = CAM_PIN_XCLK,
pin_sccb_sda = CAM_PIN_SIOD,
pin_sccb_scl = CAM_PIN_SIOC,
pin_d7 = CAM_PIN_D7,
pin_d6 = CAM_PIN_D6,
pin_d5 = CAM_PIN_D5,
pin_d4 = CAM_PIN_D4,
pin_d3 = CAM_PIN_D3,
pin_d2 = CAM_PIN_D2,
pin_d1 = CAM_PIN_D1,
pin_d0 = CAM_PIN_D0,
pin_vsync = CAM_PIN_VSYNC,
pin_href = CAM_PIN_HREF,
pin_pclk = CAM_PIN_PCLK,
xclk_freq_hz = 2000000,
ledc_timer = LEDC_TIMER_0,
ledc_channel = LEDC_CHANNEL_0,
pixel_format = PIXFORMAT_JPEG,
frame_size = FRAMESIZE_96X96,    
jpeg_quality = 12, 
fb_count = 1,  
grab_mode = CAMERA_GRAB_LATEST,
fb_location = CAMERA_FB_IN_PSRAM

camera: ov2640

log:
I (0) cpu_start: App cpu up.
I (665) esp_psram: SPI SRAM memory test OK
I (673) cpu_start: Pro cpu start user code
I (673) cpu_start: cpu freq: 160000000 Hz
I (673) cpu_start: Application information:
I (676) cpu_start: Project name:     thesis-project
I (682) cpu_start: App version:      v5.1-dev-2066-g7869f4e151
I (688) cpu_start: Compile time:     Dec 13 2022 17:51:49
I (694) cpu_start: ELF file SHA256:  42754dc4ca6b034a...
I (700) cpu_start: ESP-IDF:          v5.1-dev-2066-g7869f4e151
I (707) cpu_start: Min chip rev:     v0.0
I (711) cpu_start: Max chip rev:     v3.99 
I (716) cpu_start: Chip rev:         v1.0
I (721) heap_init: Initializing. RAM available for dynamic allocation:
I (728) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (734) heap_init: At 3FFB2B90 len 0002D470 (181 KiB): DRAM
I (741) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (747) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (753) heap_init: At 4008F6D4 len 0001092C (66 KiB): IRAM
I (760) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (768) spi_flash: detected chip: generic
I (772) spi_flash: flash io: dio
I (787) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (788) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (798) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (798) cam_hal: cam init ok
I (808) sccb: pin_sda 26 pin_scl 27
I (808) sccb: sccb_i2c_port=0

I (818) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (848) camera: Detected camera at address=0x30
I (848) camera: Detected OV2640 camera
I (848) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (938) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 0
I (938) cam_hal: Allocating 1843 Byte frame buffer in PSRAM
I (938) cam_hal: cam config ok
I (938) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
I (1028) example:take_picture: Taking picture...
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
W (5028) cam_hal: Failed to get the frame on time!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

and then it restarts.
I don't have the slightest queue of what EV-VSYNC-OVF  the only thing i understood reading the source is that is written when the system isn't able to retrieve the frame buffer from the cam queue inside the ll_cam_send_event() function but i don't know what is the cause of this behaviour.

PS: I have also tried different clock speeds (now is at 2MHz) different pixel formats and different frame sizes and also tried to use multiple fbs (as suggested by MaxSvjatoha in a previous closed issue) but nothing worked

HomeACcessoryKid, 2023-01-22:
For what it is worth, I had the `EV-VSYNC-OVF` errors as well, and no idea where to look.
Turned out that re-seating the connector solved the issue!
I hope you are equally lucky.

DJG193, 2023-03-26:
> 将xclk_freq_hz = 2000000 减少到 1000000

is useful ! 谢谢

nikolausspence, 2023-03-29:
see pull requests:
https://github.com/espressif/esp-who/pull/250
https://github.com/espressif/esp32-camera/pull/523

Both together solve the problem.  Setting the XCLK_FREQ_HZ is helpful but does not solve overbearing issues.

12Mhz with x2 multiplier is better than 10Mhz and the examples contain a hard coding of 15Mhz.  This setting should be based on the sensor used.

closed 2023-06-05 by github-actions[bot]

Meowchine1, 2024-06-04:
Formatting sd card and this video help me:  https://www.youtube.com/watch?v=6KKHgyapl-c

RN0H, 2024-06-28:
> reduce xclk_freq_hz = 2000000 to 1000000
Works! Thanks!