espressif/esp32-camera #492: fast aid please::I have read all the related issues and still can't fix this error😑😀

majedjun17, 2022-12-21:
after downloading the last version of esp32 (2.0.5) I still get this error 
**E (291) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105**
I use this board https://mikroelectron.com/Product/ESP32-CAM-with-USB-Development-Board/
the sketch I use is the example sketch (Camera->CameraWebServer)
I really want to start the cam for my graduation project after 20 days😥

Jason2866, 2022-12-22:
Error 105 is typical for a bad hardware design. Often caused by bad power supply or loose cam cable. Get a S3 cam board. Never had error 105 with this boards.
It is NOT a software bug!

majedjun17, 2022-12-26:
> Error 105 is typical for a bad hardware design. Often caused by bad power supply or loose cam cable. Get a S3 cam board. Never had error 105 with this boards. It is NOT a software bug! Duplicate of #450

thank you, I bought the one you mentioned and it's working fine.

closed 2023-02-26 by majedjun17

Piyush-GG, 2024-08-17:
I am getting this error with the ESP32 S3 cam board, idk what to do!!!