espressif/esp32-camera #494: Please upload versions on the PlatformIO libraries registry

AgainPsychoX, 2022-12-31:
See https://registry.platformio.org/libraries/espressif/esp32-camera/versions

Currently only version 2.0.0 is listed at the registry.

me-no-dev, 2023-01-09:
we do not update the PIO registry ourselves. Please contact PlatformIO for that

closed 2023-01-09 by me-no-dev

AgainPsychoX, 2023-01-20:
What do you mean... Can't you publish new version?

https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing

AgainPsychoX, 2023-01-20:
If you don't have ownership over the registry entry of the library, you can claim it:

https://github.com/platformio/platformio-registry/issues/new?labels=namespace&template=claim-ownership.md&title=Claiming+namespace+[name]

me-no-dev, 2023-01-24:
@ivankravets @valeros any clues how/who added/updated the camera driver on your side? How should we proceed?

ivankravets, 2023-01-24:
@me-no-dev, thanks for pinging us.

There are 2 options on how to resolve this issue:

1) The "autosync" legacy mode is enabled for this library. The only problem is that https://github.com/espressif/esp32-camera/blob/master/library.json is outdated. The `version` field should match with the latest `release tag` (in your case ), `"version": "2.0.3". You could update the `library.json` manifest with the latest version but it will not work because the "tag" contains [invalid manifest](https://github.com/espressif/esp32-camera/blob/v2.0.3/library.json). It points to 2.0.0 instead of 2.0.3. If you update the `library.json` in the master branch and publish the new `2.0.4` - PlatformIO Registry will handle the new version automatically.
2) This option should be more comfortable for you. No need to release a new version. Clone the source code of `v2.0.3`, update `library.json => version` field, run `pio pkg publish --owner espressif /path/to/cloned/lib`.

Please note, if you use option 2, the option 1 will be permanently disabled. Next time you will always need to use option 2.

@me-no-dev, did this help you?

grbl-IO, 2023-06-06:
I still see 2.0.0 in PlatformIO :-)

ivankravets, 2023-06-06:
I manually published 2.0.4 version https://registry.platformio.org/libraries/espressif/esp32-camera

Please confirm that it works for you. The configuration might look as :

**platformio.ini**
```ini
[env]
platform = espressif32
framework = espidf
monitor_speed = 115200
lib_deps =
    espressif/esp32-camera@^2.0.4

[env:esp32dev]
board = esp32dev
```

**CMakeLists.txt** (in the root of the project)
```cmake
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
list(APPEND EXTRA_COMPONENT_DIRS "./pio/libdeps/esp32dev/esp32-camera")
project(camera_example)
```