espressif/esp32-camera #495: Error while using it with Arduino IDE

orhanemree, 2023-01-08:
Hi, I am tring to use this in Arduino but returns error
```
'process_image' was not declared in this scope
     process_image(fb->width, fb->height, fb->format, fb->buf, fb->len);
'camera_captue' was not declared in this scope
   camera_captue();
camera_config_t' has no non-static data member named 'pin_sccb_sda
```

I don't know is it about the library or not but how can I fix this?

me-no-dev, 2023-01-09:
please post this in Arduino's repository. It is not clear what version of esp32-Arduino or this driver you are using.

closed 2023-01-09 by me-no-dev

orhanemree, 2023-01-09:
I am using Espressif ESP32 based card and my sensor model is OV2640.
By Arduino I meant Arduino IDE. -not PlatformIO-
And still if its not about this repo where is the Arduino's repo that should I post?
Thanks

me-no-dev, 2023-01-10:
here it is: https://github.com/espressif/arduino-esp32