espressif/esp32-camera #498: How to fully disable `cam_hal` logging?

AgainPsychoX, 2023-01-22:
I think one can disable logs per component name using [`esp_log_level_set("cam_hal", ESP_LOG_INFO);`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#_CPPv417esp_log_level_setPKc15esp_log_level_t), however some of logs generated by this library don't use standard way of logging in ESP, but `ESP_CAMERA_ETS_PRINTF` macro which is `ets_printf`.

Example:
```
cam_hal: EV-EOF-OVF
W (29504) cam_hal: NO-SOI    // Can be disabled using the function described above
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
```
Another small issue is they look not as nice in the logs (missing the standard prefix).

The only instance of this is at `ll_cam_send_event` function:
https://github.com/espressif/esp32-camera/blob/2478a5fc7aca43137c8b112af5805f9f5b4d0fab/driver/cam_hal.c#L104-L111

I **guess** there maybe is no way to efficiently use standard logging (`ESP_LOGX`) inside this `IRAM_ATTR`-marked function, but how about making some macro (configurable form `Kconfig`?) as option to disable this one log?

igrr, 2023-01-22:
> I guess there maybe is no way to efficiently use standard logging (ESP_LOGX) inside this IRAM_ATTR-marked function

It should be possible to use [ESP_DRAM_LOGx](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#c.ESP_DRAM_LOGE) for this.

The only issue is that it was added in IDF v4.2, while this library has to support v4.1 and later, at least until February when IDF v4.1 will become EoL. I would probably suggest waiting until then, removing v4.1 support, and then replacing the logging macros with ESP_DRAM_LOGx.

closed 2023-03-30 by github-actions[bot]

ucfncouch, 2023-07-20:
I am trying to figure out how to full disable cam_hal messages. I am trying to view some output in the terminal put they are completely drowned out by constant cam_hal messages.