espressif/esp32-camera #499: Flickering with Waveshare OV5640 board (B) and ESP32-S3-DevKitC-1-N8R2

jayavanth, 2023-01-26:
Here's the video of the flickering: https://imgur.com/a/85QLVHk. The first one is when I set it to PIXFORMAT_MONOCHROME. The second one is for PIXFORMAT_JPEG.

The more I move it or change anything in the frame, the more it flickers. If I keep the scene and the camera still, there is minimal flickering.

These are the pins I'm using:
```
.pin_pwdn = -1,
.pin_reset = -1,
.pin_xclk = 47,
.pin_sscb_sda = 36,
.pin_sscb_scl = 37,
.pin_d7 = 21,
.pin_d6 = 20,
.pin_d5 = 7,
.pin_d4 = 15,
.pin_d3 = 8,
.pin_d2 = 46,
.pin_d1 = 12,
.pin_d0 = 13,
.pin_vsync = 40,
.pin_href = 39,
.pin_pclk = 48,
```

More information:
1. This happens for all image sizes. The module I'm using has 2MB PSRAM. Hopefully that's good enough for at least VGA
2. I have set `xclk_freq_hz` to 16MHz. I tried other frequencies ranging from 7MHz to 17MHz and they are all worse than 16MHz
3. The code I'm using for streaming is from here: https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/video_stream_server/ and I'm only using WiFi AP at the moment for streaming.
4. I get these warnings on the console frequently
    a. Always: `W (14174) cam_hal: Failed to get the frame on time!`
    b. Occasionally: `W (78408) cam_hal: NO-EOI`
5. The program crashes multiple times with these errors but eventually it starts streaming after 2-3 crashes
```
E (1774) sccb: W [3810]=00 0 fail

E (1774) ov5640: Setting framesize to: 1280x720 failed
E (1777) camera: Failed to set frame size
```
6. I'm using esp-idf `v5.1-dev` but I have tried `ESP-IDF v4.4.2-155` as well
7. Devkit: ESP32-S3-DevKitC-1-N8R2, Camera board: Waveshare OV5640 board (B)

jayavanth, 2023-01-26:
Looks like the first video (Monochrome) is because the stream server only handles JPEG

me-no-dev, 2023-01-27:
The problem is with the fact that you are using 16MHz XCLK. On S3, when you set HCLK to 16MHz, you turn on EDMA support on the camera bus and the hardware tries to write to the PSRAM directly. This is not working all that great, especially on QSPI PSRAM (R2 models). R8 models with OPI PSRAM do better. I suggest you change to another XCLK. The artifacts you see are exactly missing bytes in the frame.

jayavanth, 2023-01-28:
@me-no-dev Thanks! I tried other clock rates both higher and lower but still getting similar results. I ordered an R8 devkit and will update how it goes

jayavanth, 2023-02-03:
Problem seems a bit more complicated that thought previously. Here are some facts and some changes I made:

1. The OV5640 camera board is connected to ESP32-S3-DevKitC-1-N8R8 through a breadboard.
2. I added Korvo-2 to the mix to help with debugging.
3. Turns out that the code doesn't support OV5640 that well and framerates get capped at ~15fps with any image size. So I switched to OV3660 sensor on both boards. The observations below are for OV3660 both on Korvo-2 and N8R8.
4. I also changed the camera pin configuration on N8R8 to match that of Korvo-2 and that resulted in better performance and lesser `NO-SOI`, `NO-EOI` warnings.
5. JPEG quality is set at 30.

With these changes, I tried out these two examples: [test_framerate](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/test_framerate) and [video_stream_server](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/video_stream_server)

**Results on test_framerate:**

* Korvo-2: I can get good framerate close to theoretical so that's comforting. But I can't see the actual video with this example so I'm not sure what it looks like.
* ESP32-S3-DevKitC-1-N8R8: Framerate is half of that on Korvo-2. Examples: 22.26 fps at 15MHz for VGA, 17.4 fps at 20Mhz for HD.

**Results on video_stream_server:**

Unfortunately I can't use the same clock numbers for this example without getting SOI/EOI warnings probably because it streams the frames over Wi-Fi. I end up using about half the clock rate from previous example.

* Korvo-2: I don't see any flickering when using half the clock rates (7-10 MHz). With half the clock rate, I don't see the fps numbers seen in the previous example. When I increase the clock rates beyond 10MHz, it starts flickering.
* ESP32-S3-DevKitC-1-N8R8: Flickering all around. Unless I use QVGA and much lower clock rates of 5-7MHz

jayavanth, 2023-04-05:
Flickering was the result of interference from the wires sticking out of the breadboard connecting to the camera. Issues went away after PCB was fabricated

closed 2023-04-05 by jayavanth