espressif/esp32-camera #503: camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

achao1104, 2023-02-02:
idf4.3.2 and esp32s2 and ov5640

It runs correctly in most of time, but sometimes, it failed as below, if I restarted, it will recover again, what is the issue then? thanks in advanced.

I (1289849) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (1289849) cam_hal: cam init ok
I (1289849) sccb: pin_sda 38 pin_scl 37
E (1289929) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (1289929) camera: Camera Init Failed
I (1295029) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (1295029) cam_hal: cam init ok
I (1295029) sccb: pin_sda 38 pin_scl 37
E (1295109) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (1295109) camera: Camera Init Failed
I (1300209) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (1300209) cam_hal: cam init ok
I (1300209) sccb: pin_sda 38 pin_scl 37
E (1300289) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (1300289) camera: Camera Init Failed
I (1305389) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (1305389) cam_hal: cam init ok
I (1305389) sccb: pin_sda 38 pin_scl 37
E (1305469) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (1305469) camera: Camera Init Failed

WangYuxin-esp, 2023-02-03:
Can I know the clock frequency of I2C currently used? 100K is recommended. When the program is initialized, an I2C request will be sent to detect the camera sensor. This should be related to the hardware connection of I2C.

achao1104, 2023-02-03:
Thanks for your feedback, and where to set the I2C frequency? Is it followings:
    camera_config_t camera_config = {
        .pin_pwdn = -1,
        .pin_reset = -1,
        .pin_xclk = CAM_XCLK,
        .pin_sscb_sda = CAM_SDA,
        .pin_sscb_scl = CAM_SCL,
        .pin_d7 = CAM_D7,
        .pin_d6 = CAM_D6,
        .pin_d5 = CAM_D5,
        .pin_d4 = CAM_D4,
        .pin_d3 = CAM_D3,
        .pin_d2 = CAM_D2,
        .pin_d1 = CAM_D1,
        .pin_d0 = CAM_D0,
        .pin_vsync = CAM_VSYNC,
        .pin_href = CAM_HSYNC,
        .pin_pclk = CAM_PCLK,
        //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
		//.xclk_freq_hz = 20000000,
		.xclk_freq_hz = 10000000,
		//.xclk_freq_hz = 16000000,//EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
		//.xclk_freq_hz = 5000000,
        .ledc_timer = LEDC_TIMER_0,
        .ledc_channel = LEDC_CHANNEL_0,
        .pixel_format = PIXFORMAT_JPEG, //YUV422,GRAYSCALE,RGB565,JPEG
        .frame_size = size,//FRAMESIZE_UXGA,//size,    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG
        .jpeg_quality = (quality<10?10:quality),//(quality<8?8:quality),//quality, //0-63 lower number means higher quality
        .fb_count = 1,       //if more than one, i2s runs in continuous mode. Use only with JPEG
		.grab_mode = CAMERA_GRAB_WHEN_EMPTY//CAMERA_GRAB_LATEST. Sets when buffers should be filled
    };

WangYuxin-esp, 2023-02-03:
If you are using ESP-IDF, you can open the configuration menu through `idf.py menuconfig`. Then follow this path to find the place to configure I2C frequency: Component config ->Camera.

achao1104, 2023-02-04:
I am using idfd4.3.2, and it looks there is no i2c config in your mentioned path.
[image]

achao1104, 2023-02-04:
If you mean here, I think it is 100k
[image]

closed 2023-04-13 by github-actions[bot]

g0tmk, 2023-12-31:
In my case, this was caused by an unstable 5v0 or GND connection to my ESP32-CAM board. Apparently it was enough power to boot the ESP32, but not the camera module.

After replacing the jumper cables I was using for those two pins, the error went away with no changes to the code.