espressif/esp32-camera #506: Change register values of ov5640 camera

dume03, 2023-02-13:
Hi,

Our application requires usage of ESP32 with ov5640 camera. We are using esp32-camera library in our application.

We need to set some registers of ov5640 camera from ESP32. For example, we want to set register 0x3031 of ov5640 to some custom value. 
So is it possible to change registers from the library. If possible, then where can we set those values.
We also want to change the values of some other registers from the datasheet of ov5640 as per our requirements. 
Please let us know how we can set the custom values of the registers.

me-no-dev, 2023-02-13:
you can use [this method](https://github.com/espressif/esp32-camera/blob/master/driver/include/sensor.h#L249) to change register values.
```cpp
sensor_t * s = esp_camera_sensor_get();
s->set_reg(s, reg, mask, value);
```

macksal, 2023-02-14:
you can also fork the driver and modify `ov5640_settings.h` if necessary.

dume03, 2023-02-16:
Hi,
Thanks for the reply.
How do I get the mask value for different registers?
Is there any default value for mask?
Please let me know.

me-no-dev, 2023-02-17:
Mask is used if you want to modify only certain bits of the register. If you give it 0xFF it will change the whole register.

dume03, 2023-02-21:
Thanks for your response.

We have to set 0x3031 register at 0x08 to bypass the regulator as it is giving reverse current and default value of the register is 0x00.
So as per the set_reg() when we set mask as 0x00 the value set on the register is 0x00(logically) while when we use mask as 0xff then only the register value will be set to 0x08.

Please confirm and if there is an issue in this let me know.

me-no-dev, 2023-02-21:
The mask should be 0xFF if you want to set the value of the whole register. Mask of 0x00 will not change anything

dume03, 2023-02-21:
Thanks for your reply.
This issue is resolved now.

closed 2023-02-21 by dume03