espressif/esp32-camera #508: Detected camera not supported only after a software reset

nicksimonov, 2023-02-20:
Board: AI Thinker ESP32-CAM
IDE: Arduino

Description: setup() executes normally and the camera initializes correctly after a hardware reset. But if ESP.restart() is executed the same startup fails on esp_camera_init(&config) with:

[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

I tried executing esp_camera_deinit(); before esp_camera_init() and various timeouts but that did not change anything. What are the ways to make the software reset work?

nicksimonov, 2023-02-23:
> What triggers the “probe failed” is that the ESP32 sends an I2C signal, but does not receive an ACK. In the case of just rebooting and not being able to re-initialize again, I guess it's because the sensor is not ready to work properly.

Thank you. Can you offer any suggestions on how to make the sensor ready again programmatically and the software reset work?

nicksimonov, 2023-04-25:
Is there a programmatic way to emulate the hardware reset? I mean to set some pin value so it sends the hardware reset pulse?

WangYuxin-esp, 2023-04-25:
Some cameras can be reset through reset pin. If you have a manual for this sensor, you can also use `set_reg()` set its register to achieve reset.

olejeek, 2023-05-29:
I get this problem after deep sleep. Have anyone solve this problem?
I`m using OV2640

nicksimonov, 2023-05-30:
> I get this problem after deep sleep. Have anyone solve this problem? I`m using OV2640

No, sleep has nothing to do with it. ESP32 resets fine n matter what. The problem is with the camera reset

olejeek, 2023-05-30:
> > I get this problem after deep sleep. Have anyone solve this problem? I`m using OV2640
> 
> No, sleep has nothing to do with it. ESP32 resets fine no matter what. The problem is with the camera reset

I understand. I want to say I cant initialize camera after esp32 deep sleep.

nicksimonov, 2023-05-31:
> > > I get this problem after deep sleep. Have anyone solve this problem? I`m using OV2640
> > No, sleep has nothing to do with it. ESP32 resets fine no matter what. The problem is with the camera reset
> 
> I understand. I want to say I cant initialize camera after esp32 deep sleep.
It is likely the same issue then. I haven't solved the problem with a piece of code, thinking about just wiring the reset button.

olejeek, 2023-05-31:
I think you can try anything like this before softreset:
`
sensor_t * sensor =  esp_camera_sensor_get(void);
sesor->reset();
`
I can try it too some hours later.

closed 2023-08-08 by github-actions[bot]