espressif/esp32-camera #510: esp32S3 jpg编码解码速度如何？

weilian1977, 2023-02-21:
有没有堆esp32S3 jpg编码解码的速度极进行测试？

WangYuxin-esp, 2023-02-23:
与输入的图像格式、分辨率大小有关。

weilian1977, 2023-02-25:
> 
有没有大致的测试结果参考一下？

WangYuxin-esp, 2023-02-27:
按照我个人的测试，参考如下：
QVGA\YUV422 JPEG编码 33
VGA\YUV422 JPEG 编码 9

QVGA  JPEG解码 65
VGA JPEG 解码 16

weilian1977, 2023-02-27:
> 8bit 灰度图 VGA相当于仅有Y分量， JPEG 编码速度会不会快一些？

WangYuxin-esp, 2023-02-27:
当然。MONO\Only Y 的格式，编码更快。

weilian1977, 2023-03-06:
哪一款摄像头适合测试MONO格式JPEG编码速度呢？

WangYuxin-esp, 2023-03-06:
推荐 S3 + SC031GS /SC035HGS。

weilian1977, 2023-03-07:
> VGA\YUV422 JPEG 编码 9
 VGA\YUV422 JPEG 编码时需要对Y、Cb、Cr三个分量进行DCT变换、量化、扫描、编码，可以认为对于MONO\Only Y压缩运算量为YUV422压缩运算量的1/3?是否可以估计VGA\MONO\Only Y JPEG 编码 27？

WangYuxin-esp, 2023-03-08:
VGA 分辨率，Only Y 的 JPEG 编码应该是可以上 20fps+。
你可以直接伪造一个 Only Y VGA 的数据，在 S3 上测试下速率。

weilian1977, 2023-03-08:
真是太棒了！

WangYuxin-esp, 2023-03-09:
最好不涉及任何摄像头，只用一个假数据测试Only Y JPEG 编码的能力。
理论研究不太好估算。
我原先好像测试过 VGA OnlyY 的JPEG 编码。很流畅，是个视频项目。

weilian1977, 2023-03-11:
> 
重要价值！

WangYuxin-esp, 2023-03-13:
是的。15fps+

weilian1977, 2023-03-19:
我修改了esp who中human_face_detection_web项目代码，使用OV3660摄像头，设置像素格式为灰度，分辨率为VGA，在else if (!frame2jpg(frame, 90, &_jpg_buf, &_jpg_buf_len))前后加了信息打印代码，输出如下，好像达不到15帧？
I (21229) camera_httpd: JPEG compression begin
I (21369) camera_httpd: JPEG compression end
I (21469) camera_httpd: JPEG compression begin
I (21609) camera_httpd: JPEG compression end
I (21819) camera_httpd: JPEG compression begin
I (21959) camera_httpd: JPEG compression end
I (22049) camera_httpd: JPEG compression begin
I (22189) camera_httpd: JPEG compression end
I (22499) camera_httpd: JPEG compression begin
I (22639) camera_httpd: JPEG compression end
I (22719) camera_httpd: JPEG compression begin
I (22859) camera_httpd: JPEG compression end
I (23169) camera_httpd: JPEG compression begin
I (23319) camera_httpd: JPEG compression end
I (23399) camera_httpd: JPEG compression begin
I (23539) camera_httpd: JPEG compression end
I (23849) camera_httpd: JPEG compression begin
I (23989) camera_httpd: JPEG compression end
I (24079) camera_httpd: JPEG compression begin
I (24219) camera_httpd: JPEG compression end
I (24519) camera_httpd: JPEG compression begin
I (24669) camera_httpd: JPEG compression end
I (24749) camera_httpd: JPEG compression begin
I (24889) camera_httpd: JPEG compression end
I (25199) camera_httpd: JPEG compression begin
I (25339) camera_httpd: JPEG compression end
I (25419) camera_httpd: JPEG compression begin
I (25559) camera_httpd: JPEG compression end
I (25869) camera_httpd: JPEG compression begin
I (26009) camera_httpd: JPEG compression end
I (26099) camera_httpd: JPEG compression begin
I (26239) camera_httpd: JPEG compression end
I (26549) camera_httpd: JPEG compression begin
I (26689) camera_httpd: JPEG compression end
I (26779) camera_httpd: JPEG compression begin
I (26919) camera_httpd: JPEG compression end
I (27219) camera_httpd: JPEG compression begin
I (27369) camera_httpd: JPEG compression end
I (27449) camera_httpd: JPEG compression begin
I (27589) camera_httpd: JPEG compression end
I (27899) camera_httpd: JPEG compression begin
I (28039) camera_httpd: JPEG compression end
I (28129) camera_httpd: JPEG compression begin
I (28269) camera_httpd: JPEG compression end
I (28569) camera_httpd: JPEG compression begin
I (28719) camera_httpd: JPEG compression end
I (28799) camera_httpd: JPEG compression begin
I (28939) camera_httpd: JPEG compression end
I (29249) camera_httpd: JPEG compression begin
I (29389) camera_httpd: JPEG compression end
I (29479) camera_httpd: JPEG compression begin
I (29609) camera_httpd: JPEG compression end
I (29929) camera_httpd: JPEG compression begin
I (30069) camera_httpd: JPEG compression end
I (30149) camera_httpd: JPEG compression begin
I (30289) camera_httpd: JPEG compression end
I (30599) camera_httpd: JPEG compression begin
I (30739) camera_httpd: JPEG compression end
I (30829) camera_httpd: JPEG compression begin
I (30969) camera_httpd: JPEG compression end
I (31279) camera_httpd: JPEG compression begin
I (31419) camera_httpd: JPEG compression end
I (31499) camera_httpd: JPEG compression begin
I (31639) camera_httpd: JPEG compression end
I (31949) camera_httpd: JPEG compression begin
I (32089) camera_httpd: JPEG compression end
I (32179) camera_httpd: JPEG compression begin
I (32319) camera_httpd: JPEG compression end
I (32629) camera_httpd: JPEG compression begin
I (32769) camera_httpd: JPEG compression end
I (32849) camera_httpd: JPEG compression begin
I (32989) camera_httpd: JPEG compression end
I (33299) camera_httpd: JPEG compression begin
I (33439) camera_httpd: JPEG compression end
I (33529) camera_httpd: JPEG compression begin
I (33669) camera_httpd: JPEG compression end
I (33979) camera_httpd: JPEG compression begin
I (34119) camera_httpd: JPEG compression end

WangYuxin-esp, 2023-03-20:
不执行压缩。只是出帧的速率能达到 15fps 吗？

weilian1977, 2023-03-20:
这个没有评估，因为对OV3660摄像头并不熟悉，不好评估摄像头的工作参数和输出帧频。frame2jpg(frame, 90, &_jpg_buf, &_jpg_buf_len)这个函数应该就是对图像数据进行jpeg压缩的函数。

WangYuxin-esp, 2023-03-20:
你可以参考这里的代码，测试下摄像头输出的帧率是多少：
https://github.com/espressif/esp32-camera/blob/master/test/test_camera.c#L168
或者你给一个 VGA 的假数据，只测试压缩的帧率。

weilian1977, 2023-03-20:
ok！

weilian1977, 2023-03-21:
OV3660  4FPS

WangYuxin-esp, 2023-03-23:
压缩很消耗内存和时间的。当前的移植很粗糙，没有针对平台进行优化。  
这里有一个正在测试中的 jpeg lib。应该比当前 这里的压缩速率高一些。
https://github.com/espressif/esp-adf-libs/blob/master/esp_codec/include/codec/esp_jpeg_dec.h

weilian1977, 2023-03-23:
ESP-S3-EYE dev board V1.1  flash and spram仅支持80Mhz？

weilian1977, 2023-03-23:
> 压缩很消耗内存和时间的。当前的移植很粗糙，没有针对平台进行优化。 这里有一个正在测试中的 jpeg lib。应该比当前 这里的压缩速率高一些。 https://github.com/espressif/esp-adf-libs/blob/master/esp_codec/include/codec/esp_jpeg_dec.h

没有说明文档么

weilian1977, 2023-03-23:
> 按照我个人的测试，参考如下： 
QVGA\YUV422 JPEG编码 33 
VGA\YUV422 JPEG 编码 9

目前测试看，
QVGA 灰度JPEG编码 26 PSRAM@80M，
VGA\灰度 JPEG 编码 7 PSRAM@80M，
如果考虑内存速度的影响，推测
QVGA 灰度JPEG编码 39 PSRAM@120M？
VGA\灰度 JPEG 编码 10 PSRAM@120M？

但是ESP-S3-EYE dev board V1.1 flash and spram似乎仅支持80Mhz？

WangYuxin-esp, 2023-03-24:
灰度图的模式下，如果 sensor 输出灰度图的速率能跟得上，编码速率不应该这么慢的。  
我建议你可以再检查下这份配置，CPU、flash、PSRAM 的速率是不是已经最快了。  
https://github.com/espressif/esp-iot-solution/blob/master/examples/camera/pic_server/sdkconfig.defaults#L3

xubo92, 2023-03-24:
你好 @WangYuxin-esp 

看到了你之前的一些建议，很感谢。我最近也在测试jpeg的编码，目前的测试结果是在QVGA/RGB565的摄像头输入设置下，frame2jpg()这个jpeg压缩函数需要大约125ms才能压缩一张图片，我也测试了QVGA/GRAYSCALE的设置，大约需要46ms。这个速度是不是偏慢？还有哪些地方可以优化呢？我我已经使用了这份设置：https://github.com/espressif/esp-iot-solution/blob/master/examples/camera/pic_server/sdkconfig.defaults#L3 

谢谢！

weilian1977, 2023-03-26:
> 灰度图的模式下，如果 sensor 输出灰度图的速率能跟得上，编码速率不应该这么慢的。 我建议你可以再检查下这份配置，CPU、flash、PSRAM 的速率是不是已经最快了。 https://github.com/espressif/esp-iot-solution/blob/master/examples/camera/pic_server/sdkconfig.defaults#L3

核实了一下，基本是一致的，编码速度没有明显变化

xubo92, 2023-03-26:
> 如果直接WIFI传640*480 灰度 能到多少帧？

@weilian1977 这个我测过，很慢，因为灰度图大小也有几十k，比jpeg（2~3k）大得多，wifi传不动。

weilian1977, 2023-03-27:
> > 如果直接WIFI传640*480 灰度 能到多少帧？
> 
> @weilian1977 这个我测过，很慢，因为灰度图大小也有几十k，比jpeg（2~3k）大得多，wifi传不动。
@xubo92  wifi的速度是多少？分析起来感觉不对啊？官方手册“支持 1T1R 模式，数据速率高达 150 Mbps”，按照640*480=307200byte=300kbyte，(150 Mbit/8)/300kbyte=(150*1024kbit/8)/300kbyte=64?考虑到乱七八糟，20帧总应该有？

weilian1977, 2023-03-27:
https://github.com/espressif/esp32-camera/pull/419

weilian1977, 2023-03-29:
> 按照我个人的测试，参考如下： QVGA\YUV422 JPEG编码 33 VGA\YUV422 JPEG 编码 9
> 
> QVGA JPEG解码 65 VGA JPEG 解码 16

esp-idf版本？esp32-camera版本？

xubo92, 2023-03-29:
> @xubo92 wifi的速度是多少？分析起来感觉不对啊？官方手册“支持 1T1R 模式，数据速率高达 150 Mbps”，按照640×480=307200byte=300kbyte，(150 Mbit/8)/300kbyte=(150×1024kbit/8)/300kbyte=64?考虑到乱七八糟，20帧总应该有？

连的是家里的wifi 2.4GHz, ESP32的wifi没有做特别设置。按照理论算不成立。 很多帖子讲过esp32就算用原生支持的jpeg模式，也就30FPS上下不多，倒退算如果灰度图比jpeg图片大10倍以上，那么就只剩3FPS了。https://github.com/espressif/esp32-camera/issues/15

weilian1977, 2023-03-30:
> > @xubo92 wifi的速度是多少？分析起来感觉不对啊？官方手册“支持 1T1R 模式，数据速率高达 150 Mbps”，按照640×480=307200byte=300kbyte，(150 Mbit/8)/300kbyte=(150×1024kbit/8)/300kbyte=64?考虑到乱七八糟，20帧总应该有？
> 
> 连的是家里的wifi 2.4GHz, ESP32的wifi没有做特别设置。按照理论算不成立。 很多帖子讲过esp32就算用原生支持的jpeg模式，也就30FPS上下不多，倒退算如果灰度图比jpeg图片大10倍以上，那么就只剩3FPS了。#15

“esp32就算用原生支持的jpeg模式，也就30FPS上下”这个不一定是WIFI传输速度的问题，从很多应用的角度30FPS已经可以了，从传感器的角度30FPS对于非特殊场景的应用也差不多了

但是好像也确实有很多讲速度不高，或者只能几十M