espressif/esp32-camera #512: Can you support the sensor with SPI interface, such as gc6153, gc6133, etc

xingxingRealzyx, 2023-03-07:


WangYuxin-esp, 2023-03-08:
As far as I know, the camera with this SPI interface can be docked on the ESP32. However, it needs a lot of debugging and business access to start this development.

xingxingRealzyx, 2023-03-08:
Thank you for your reply. I'm trying to adapt the camera with SPI interface.

baijiaa, 2023-07-12:
> As far as I know, the camera with this SPI interface can be docked on the ESP32. However, it needs a lot of debugging and business access to start this development.

Excuse me, do you know the definition of frame format when SPI camera receives image data?
I'm a bit confused about the frame format defined on GC6133, its description is too limited

WangYuxin-esp, 2023-07-12:
> Excuse me, do you know the definition of frame format when SPI camera receives image data?
> I'm a bit confused about the frame format defined on GC6133, its description is too limited

Sorry, I am not familiar with the data output format of this sensor GC6133. Perhaps you can consult with the original factory of this sensor. But I know that ESP32 can interface with the BF20A6 camera sensor through the SPI interface.

xzc212121, 2024-07-26:
怎么样了？能够通过spi接收到图片数据了吗？

baijiaa, 2024-07-26:
可以了

xuxubaobao-6, 2025-08-21:
> Sorry, I am not familiar with the data output format of this sensor GC6133. Perhaps you can consult with the original factory of this sensor. But I know that ESP32 can interface with the BF20A6 [...]

大哥，esp32怎么驱动SPI接口的BF20A6摄像头传感器？我看esp32-camera里是大哥你添加的BF20A6的相关代码，但是是DVP接口的，想用SPI接口该怎么办？求助求助

xuxubaobao-6, 2025-08-21:
> 怎么样了？能够通过spi接收到图片数据了吗？

你也是在研究SPI接口摄像头吗，成功了吗？

WangYuxin-esp, 2025-08-21:
SPI camera support on https://github.com/espressif/esp-video-components/tree/master/esp_video

xuxubaobao-6, 2025-08-21:
> SPI camera support on https://github.com/espressif/esp-video-components/tree/master/esp_video

Thank you, I'll study it first.

xuxubaobao-6, 2025-08-26:
> SPI camera support on https://github.com/espressif/esp-video-components/tree/master/esp_video

你好，简单学习了下，发现当前库没有BF20A6的驱动，而且我使用摄像头的是SPI 2-bit模式，和库里的SPI传感器还有些区别。目前，我正在参考esp32-camera库里的BF20A6的驱动（虽然是DVP接口的）和espressif__esp_cam_sensor库里的bf3901传感器，两相结合，来完成BF20A6 的 SPI 驱动适配，然后进行SPI 2-bit模式的相关修改，如：行，帧同步等等。已经预感到会有很多问题了。。。。请问这个往方向走，对吗？或者有其他更好的解决办法吗？

xuxubaobao-6, 2025-08-26:
> 如果 BF3901 的 QVGA + 20fps 可以满足需求的话，那么不需要继续研究了，还是使用已经适配过的 sensor 为宜，加快开发。 如果需要的输出尺寸更大，那么选择 DVP、MIPI接口，帧率要求也高的话，那么选择 ESP32-P4。

应该是可以满足需求的，但是手上没有这款，只有定做的BF20A6，SPI 2-bit的摄像头。我先研究看看，能不能搞出来吧

xuxubaobao-6, 2025-09-10:
> SPI camera support on https://github.com/espressif/esp-video-components/tree/master/esp_video

> > > 如果 BF3901 的 QVGA + 20fps 可以满足需求的话，那么不需要继续研究了，还是使用已经适配过的 sensor 为宜，加快开发。 如果需要的输出尺寸更大，那么选择 DVP、MIPI接口，帧率要求也高的话，那么选择 ESP32-P4。
> > 
> > 
> > 应该是可以满足需求的，但是手上没有这款，只有定做的BF20A6，SPI 2-bit的摄像头。我先研究看看，能不能搞出来吧
> 
> 当前不支持 2-bit，不能在 SPI 接口支持 BF20A6。重新定制一个模组是最佳选择。这样软件免开发。

大佬，公司这边还是想用BF20A6，您看有没有兴趣和时间接下这个活，我们付费请你当外援

WangYuxin-esp, 2025-09-10:
> 大佬，公司这边还是想用BF20A6，您看有没有兴趣和时间接下这个活，我们付费请你当外援

不用 BF3901 SPI 的话，那用 BF20A6 的 DVP 接口也可以啊。

xuxubaobao-6, 2025-09-10:
> > > > > SPI camera support on https://github.com/espressif/esp-video-components/tree/master/esp_video
> > > > 
> > > > 
> > > > > > > 如果 BF3901 的 QVGA + 20fps 可以满足需求的话，那么不需要继续研究了，还是使用已经适配过的 sensor 为宜，加快开发。 如果需要的输出尺寸更大，那么选择 DVP、MIPI接口，帧率要求也高的话，那么选择 ESP32-P4。
> > > > > > 
> > > > > > 
> > > > > > 应该是可以满足需求的，但是手上没有这款，只有定做的BF20A6，SPI 2-bit的摄像头。我先研究看看，能不能搞出来吧
> > > > > 
> > > > > 
> > > > > 当前不支持 2-bit，不能在 SPI 接口支持 BF20A6。重新定制一个模组是最佳选择。这样软件免开发。
> > > > 
> > > > 
> > > > 大佬，公司这边还是想用BF20A6，您看有没有兴趣和时间接下这个活，我们付费请你当外援
> > > 
> > > 
> > > 不用 BF3901 SPI 的话，那用 BF20A6 的 DVP 接口也可以啊。
> > 
> > 
> > 引脚不够，其他功能占太多引脚了
> 
> 这不是是否能适配的问题，因为当前的芯片设计上没有方便的外设去处理 2-bit mode。强行适配只能作为探究性质的开发，不保证最终一定成功，另外你们需要按照要求去打样 camera sensor 模组，然后联系乐鑫商务进行商务对接，这种只能是配合着做些探究，试试能不能成。

那如果是1-bit模式呢，能成功吗？成功的话能达到多少fps。因为我们也联系了比亚迪的技术支持，他们给了一个组件，看着是从esp32-camera改的，专门适配BF20A6的SPI 1-bit，是他们以前客户量产的，但是和我们对接的不是当时那个人，对接的这个对esp平台，代码细节都不清楚，我们跑了组件里的basic例程，摄像头在发，但s3接收不到。

xuxubaobao-6, 2025-09-10:
@WangYuxin-esp 好的，谢谢，我们内部商量一下