espressif/esp32-camera #513: S3 higher framerates due to Cam interface?

SinanAkkoyun, 2023-03-08:
Hello!
Does the Esp32s3 eye support double the framerate? The datasheet says the dedicated camera port is 40MHz, as far as I know, the normal ESP32 software cam driver only allows 20MHz?

So, what framerate can I expect at 640x480?

Thank you very much

WangYuxin-esp, 2023-03-09:
VGA+RGB565\YUV422\JPEG+30fps is OK.

SinanAkkoyun, 2023-03-09:
So even though the s3 has it's own camera peripheral only 30 FPS are doable?

Thank you

WangYuxin-esp, 2023-03-09:
This depends on the format that the sensor can output. For VGA resolution, 30 fps is recommended for data in RGB format. For JPEG format data, the frame rate can be higher. Because the amount of data in JPEG format is relatively small.

macksal, 2023-05-02:
I'm having success with 72 MHz PCLK on an ESP32-S3 after making some changes to the driver.

The trick is to use EDMA mode and disable the caches while the DMA is running. The cache used for code from flash competes for the SPI0 bus, which is also used for PSRAM. Ensuring that only the GDMA is using the SPI0 bus allows it to consume the full bandwidth (80MHz in an ideal world).

I am using this for full-res stills though, the situation will be different if you need to access a different framebuffer while one is being written, or to use the CPU for other tasks during the frame capture (since you can only execute code in IRAM while the cache is disabled). This means that ongoing wifi/video transmission is likely impractical.

me-no-dev, 2023-05-02:
@macksal want to provide a PR for this? It could be limited to `fb = 1` and certain `XCLK` (much like EDMA is limited now)

macksal, 2023-05-02:
> @macksal want to provide a PR for this? It could be limited to `fb = 1` and certain `XCLK` (much like EDMA is limited now)

I can't get it merged as-is because I've changed the core functions quite a bit. I stole the `spi_flash_disable_interrupts_caches_and_other_cpu` from the (private) spi flash API, and it only works by disabling the freertos scheduler - so you can't call freertos APIs until it's enabled again. This means the IN_SUC_EOF event can't be sent via queue, I instead use a volatile bool flag which cam_task busy-waits on, and it's cleared in the ISR. I don't think this plays nice with the bounce-buffer mode used by default. There's a bit of work in cleaning things up so the driver can support either operation at once.

I'm occupied with the project that this is part of for the time being - but happy to provide more info and perhaps contribute something mergeable at a later date

closed 2023-07-09 by github-actions[bot]