espressif/esp32-camera #515: FPS decrease after writing to SD card

wouterdebie, 2023-03-15:
Hi,

I'm experiencing an odd case. I'm working on a small camera (AI Thinker cam) project (https://github.com/wouterdebie/mortycam) with PIO and FreeRTOS that records onto an SD card. I use both cores of the esp32, where the App core takes pictures and puts them on a queue that then get written to disk by a task running on the Pro core. I'm running with VGA settings and get a throughput of 25 fps. The whole thing is a small state machine where the two tasks advance state in order to do different things. 

Now the strange part is that when the esp32 starts recording, I get a throughput of 25 fps, but when I stop recording, close the file I'm writing to, reopen a new file and restart recording, fps drops to 12. And any consecutive recording session stays at 12 fps.

I've played with different settings (like fb_count, queue size, grab_mode, etc), but the behavior is the same. However, when I deinit the camera and initialize it again, I sustain 25 fps. Is there anything I'm missing?

jameszah, 2023-03-16:
My guess is that you haven't emptied the queue and released all the fb buffers.  You need an empty buffer for the camera to be transferring a frame before it is ready to give it to you ... in the "continuous mode" i think they call it.

wouterdebie, 2023-03-16:
I have. I call `esp_camera_fb_return()` after writing to the SD card. I can run the recording however long I like and fps doesn't drop. But as soon as I stop taking pictures, close the file and resume, the fps drops. After I stop taking pictures I drain the queue as well and return the framebuffers.

That said, the fps I end up with very similar to when I'm running with a single framebuffer.

wouterdebie, 2023-03-16:
This is the piece of code that writes:

```
static void write_frame(TickType_t *timestamp) {
    message_t *msg = NULL;
    TickType_t wait = portMAX_DELAY;

    // Only wait for 1000 ms max if we're draining the queue,
    // otherwise wait forever
    if (state == DRAINING) {
        wait = ms_to_ticks(1000);
    }
    xQueueReceive(msg_queue, &msg, wait);

    if (!msg) {
        ESP_LOGI(wa_tag, "No more pictures in queue");
        set_state(CLOSING_FILE);
        return;
    }

    gwavi_add_frame(gwavi, msg->fb->buf, msg->fb->len);
    esp_camera_fb_return(msg->fb);
    *timestamp = msg->timestamp;
}
```

Note that I wrap the `camera_fb_t` in `message_t`, where `message_t` looks like:
```
typedef struct {
    camera_fb_t *fb;
    TickType_t timestamp;
} message_t;
```
And is written to like this:
```
camera_fb_t *fb = esp_camera_fb_get();
                if (!fb) {
                    ESP_LOGW(rf_tag, "Camera capture failed");
                    continue;
                }

                if (fb->len == 0) {
                    ESP_LOGW(rf_tag, "Camera capture failed, len = 0");
                    esp_camera_fb_return(fb);
                    continue;
                }

                message_t msg = {
                    .fb = fb,
                    .timestamp = xTaskGetTickCount()};

                message_t *msg_ptr = &msg;
                if (xQueueSend(msg_queue, &msg_ptr, 100) != pdTRUE) {
                    ESP_LOGE(rf_tag, "Failed to send frame to queue within 100 ticks");
                    esp_camera_fb_return(fb);
                }
```

jameszah, 2023-03-16:
You could add a couple lines of code to allocate and release all your buffers, and count how many.  And see if the fresh drivers situation give you the same result as the post open/close file situation.

I did a similar thing a few years ago with a queue of jpegs connected to a camera task, and a separate sd-writer task pulling them off the queue.  I was working with a bad sd card that had very unpredictable delays and thought a predictable sequence of jpegs with a slow-then-fast sd writer would even it out.  But since you can only have 6-7 uxga frames in a 4mb memory, it didn't really have any benefit.  So I started again with a queue of 2, and a separate sd-writer task on the other core that signal back and forth when the 2 tasks are ready for new work.  I think even that was over-kill, as the camera and driver and always copying and new jpeg into memory, and the sd-writer is mostly dma'ing the previous jpeg onto the sd, so neither task is very busy.

Yes, esp_camera_fb_return is what I was referring to.

wouterdebie, 2023-03-16:
I ended up adding a `esp_camera_reset_buffers()` to my local fork where I just enable all the buffers. This seems to have done the trick. This might be functionality others want, so just in case, I'll create a PR.

closed 2023-03-16 by wouterdebie