espressif/esp32-camera #518: Premature VSYNC coming from OV2640 camera in high-resolution/high-quality capture

MichaelXt, 2023-03-16:
After workarounding #514 by initializing camera with UXGA resolution and setting `cam_obj->recv_size = cam_obj->width * cam_obj->height;` (removing divide by 5 factor), I've got JPEG images up to 300Kb coming from camera (yay!).
However if there are more details in the scene, we resulting JPEG will be even bigger, and now I'm hitting the premature VSYNC issue. The esp32camera driver state machine waits [src](https://github.com/espressif/esp32-camera/blob/2478a5fc7aca43137c8b112af5805f9f5b4d0fab/driver/cam_hal.c#L175) for VSYNC in order to complete the frame. In case with resulting image < 300Kb, the `cnt` will be in 200-300 range, (meaning that there happened 200-300 copies from DMA buffer), however as scene complexity increases, the resulting image is getting bigger, and at some point, number of copies per VSYNC `cnt` becomes in range 90-110, and inspecting the resulting dma buffer shows that it is only partially filled. It appears as if VSYNC was triggered way too early for all the data to be transferred.
I wonder if that is a OV2640 bug (where it never completes the frame, even if DMA buffer is big enough), or is there some configuration which needs to be adjusted so that DMA transfer completes the whole frame?

MichaelXt, 2023-03-16:
#317 appears to be related. @jameszah, I wonder if you have been able to get the full resolution / highest quality images?

jameszah, 2023-03-17:
Looking at some old uxga movies, quality 8-9, they seem to have 300kb frames, with an average of about 290kb.  I had not made any modifications to the drivers - the esp32-camera code.  I was not trying to maximize quality, and would increase the quality number (decrease frame size) when I would see those broken jpegs - which seemed to be complete, just missing enough bits to break the jpeg expander algorithm, but not cut off at the 60% or 80% mark.  Don't really know the issues of the vsync stuff.  I have noticed a 4 second limit sometimes in fetching a frame -- don't know if that is related.

MichaelXt, 2023-05-17:
It is still valid

closed 2023-07-25 by github-actions[bot]