espressif/esp32-camera #522: ESP32-CAM deep sleep - lower current than 5mA possible?

soleluna, 2023-03-26:
I want to use the ESP32-CAM v2 as a battery driven bird nest watcher using deep sleep between two snapshots.
Saving to SD-Card, like
https://randomnerdtutorials.com/esp32-cam-take-photo-save-microsd-card/
Finally saving to my FTP-Server via WiFi (already to be done).
Current Active   ca. 115mA / @5V
Current deep sleep     5mA / @5V
-> Is it possible, to lower the current more in deep sleep or other mode?
For ex. By SW  switching some GPIOs, CAM OFF, CAM to sleep, etc.?
For ex. By HW modification, 5V/3V3 regulator, CAM-Supply with FET, supply with 3V3, etc.

I am dreaming of 1mA or less in deep sleep, powering with 3V3 CR123 or 5V powerbank for weeks.
Would be great - thanks for any help in advance.

WangYuxin-esp, 2023-04-17:
For esp32, from the datasheet perspective, the minimum power consumption under deep sleep is at the uA level. Therefore, for devices with cameras, the main power consumption is concentrated in the camera sensor and RF module.

soleluna, 2023-04-17:
Thank you for your explanation. Without HW-mod no uA...
I will try to solar power the ESP32-CAM in combination with deep sleep.
I ended up at 3V3 / 3mA with 
esp_wifi_stop()     
adc_power_off()
esp_deep_sleep_start()

closed 2023-06-23 by github-actions[bot]