espressif/esp32-camera #526: Build warnings with pedantic flags

tore-espressif, 2023-04-03:
Pedantic build with following compile options fails
```
export PEDANTIC_FLAGS="-DIDF_CI_BUILD -Werror -Werror=deprecated-declarations -Werror=unused-variable -Werror=unused-but-set-variable -Werror=unused-function"
export PEDANTIC_CFLAGS="${PEDANTIC_FLAGS} -Wstrict-prototypes"
export PEDANTIC_CXXFLAGS="${PEDANTIC_FLAGS}"
export EXTRA_CFLAGS=${PEDANTIC_CFLAGS}
export EXTRA_CXXFLAGS=${PEDANTIC_CXXFLAGS}
```

Could you please fix the build warnings or disable the warnings (`-Wno-strict-prototypes` should help)

Related https://github.com/espressif/esp-bsp/issues/51

me-no-dev, 2023-04-18:
@tore-espressif can I add this somehow to our CI?Manually building it does not seem like ling term solution.

tore-espressif, 2023-04-19:
I prepared a PR here https://github.com/espressif/esp32-camera/pull/529

closed 2023-06-19 by tore-espressif