espressif/esp32-camera #535: esp32s3 adv7180 tinkering - EV-EOF-OVF

berkutta, 2023-05-23:
Hi

I'm trying to implement an ADV7180 FBAS to DVP converter into esp32-camera. This is my tinkering branch https://github.com/berkutta/esp32-camera/tree/adv7180-tinkering

I have implemented a very basic init procedure for this chip and hooked up the I2C lines, XCLK, VSYNC, HREF and PCLK.

Unfortunately when trying to take a Picture, the ESP is just sending `cam_hal: EV-EOF-OVF` in a loop.

What does this error indicate? Is it possible that the chip doesn't send more than specified pixels per line or something similar? Any idea where I should search next?

Log Output:
```
I (0) cpu_start: App cpu up.
I (432) spiram: SPI SRAM memory test OK
I (441) cpu_start: Pro cpu start user code
I (441) cpu_start: cpu freq: 240000000
I (441) cpu_start: Application information:
I (444) cpu_start: Project name:     camera_example
I (450) cpu_start: App version:      v2.0.4-1-g425d75d
I (456) cpu_start: Compile time:     May 23 2023 17:29:10
I (462) cpu_start: ELF file SHA256:  3a63a1c9ba020d45...
I (468) cpu_start: ESP-IDF:          v4.4.4
I (473) heap_init: Initializing. RAM available for dynamic allocation:
I (480) heap_init: At 3FC97EE8 len 00051828 (326 KiB): D/IRAM
I (486) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (493) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (499) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (505) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (514) spi_flash: detected chip: generic
I (518) spi_flash: flash io: qio
W (522) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (535) sleep: Configure to isolate all GPIO pins in sleep state
I (542) sleep: Enable automatic switching of GPIO sleep configuration
I (549) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (564) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (573) s3 ll_cam: DMA Channel=4
I (577) cam_hal: cam init ok
I (580) sccb: pin_sda 4 pin_scl 5
I (584) sccb: sccb_i2c_port=1
I (598) camera: Detected camera at address=0x21
E (599) sccb: SCCB_Read Failed addr:0x21, reg:0x0a, data:0x00, ret:-1
I (600) ov7725: Mismatch PID=0x0
E (605) sccb: SCCB_Read Failed addr:0x21, reg:0x0a, data:0x00, ret:-1
I (611) ov7760: Mismatch PID=0x0
E (615) sccb: SCCB_Read Failed addr:0x21, reg:0xf1, data:0x00, ret:-1
E (623) sccb: SCCB_Read Failed addr:0x21, reg:0xf0, data:0x00, ret:-1
I (629) gc032a: Mismatch PID=0x0
E (1633) sccb: SCCB_Write Failed addr:0x21, reg:0xfe, data:0x00, ret:263
E (1633) sccb: SCCB_Read Failed addr:0x21, reg:0x00, data:0x00, ret:-1
I (1636) gc0308: Mismatch PID=0x0
Working on ADV7180 probe 
I (1644) camera: Detected ADV7180 camera
I (1647) camera: Camera PID=0x1c VER=0x00 MIDL=0x00 MIDH=0x00
I (1664) camera: Frame size id: 22 

I (1664) s3 ll_cam: node_size: 2880, nodes_per_line: 1, lines_per_node: 2
I (1665) s3 ll_cam: dma_half_buffer_min:  2880, dma_half_buffer: 11520, lines_per_half_buffer:  8, dma_buffer_size: 23040
I (1677) cam_hal: buffer_size: 23040, half_buffer_size: 11520, node_buffer_size: 2880, node_cnt: 8, total_cnt: 72
I (1688) cam_hal: Allocating 829440 Byte frame buffer in PSRAM
I (1695) cam_hal: cam config ok
I (1698) camera: Setting frame size to 720x576
Width of 22 is 720 
Width of 8 is 640 
Passing 22 
Setting frame size W: 720, H: 576 
I (1711) example:take_picture: Taking picture...
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
```

WangYuxin-esp, 2023-06-21:
@berkutta , Hello, thanks for your question. From the log, it can be seen that the sensor has been properly initialized. The main reason for the prompt `cam_hal: EV-EOF-OVF` is due to the VSYNC signal being too fast. If your hardware connection is correct, please try reducing the value of the XCLK clock.

berkutta, 2023-06-21:
@WangYuxin-esp Thank you for your answer.

I have now tried it with different .xclk_freq_hz. Unfortunately, 20Mhz, 10Mhz and 5Mhz results in `cam_hal: EV-EOF-OVF`. Not sure if .xclk_freq_hz even has an effect on the ADV7180, as xclk isn't connected to it.

My connections are as follows, do you know if they are correct?

| ADV7180 | ESP32-S3 |
| ------------- | ------------- |
| HS - Horizontal Synchronization Output Signal | CAM_PIN_HREF |
| LLC - This is a line-locked output clock for the pixel data output by the ADV7180. It is nominally 27 MHz but varies up or down according to video line length. | CAM_PIN_PCLK |
| VS - Vertical Synchronization Output Signal. | CAM_PIN_VSYNC |

WangYuxin-esp, 2023-06-25:
> 

Yes, if the sensor is not connected to the XCLK on ESP32, changing the size of the XCLK output from ESP32 will not have any effect.
For ESP32, it requires three clock signals for synchronization:
1) VSYNC: One frame of image data triggers this signal once
2) HSYNC: This signal is triggered once for a row of image data
3) PCLK: That is, one pixel of data triggers the signal once

closed 2023-08-31 by github-actions[bot]