espressif/esp32-camera #543: Switch pixformat and pixformat --> wrong Frame Buffer

PatrikRindlisbacher, 2023-06-03:
Switching from pixformat results in an incorrect size of the framebuffer.
espressif/esp32-camera@^2.0.0 --> Visual Studio Code --> PlatformIO

bool change_Picture_Format_to_JPEG() {
  sensor_t *s = esp_camera_sensor_get();
  if (s == NULL) {
    Serial.println("Error: problem reading camera sensor settings");
    return 0;
  }
  s->set_pixformat(s,FRAMESIZE_VGA);
  s->set_pixformat(s,PIXFORMAT_JPEG) ;
  return 1;
}

bool change_Picture_Format_to_GRAYSCALE() {
  sensor_t *s = esp_camera_sensor_get();
  if (s == NULL) {
    Serial.println("Error: problem reading camera sensor settings");
    return 0;
  }
  s->set_framesize(s,FRAMESIZE_96X96);
  s->set_pixformat(s,PIXFORMAT_GRAYSCALE) ; 
  return 1;
}

me-no-dev, 2023-06-05:
This will be "fixed" in the near future.

PatrikRindlisbacher, 2023-06-05:
I guess I'm not the only one waiting for this feature. 
The bug was previously found with issue #433. 
Thank you for your work.

grbl-IO, 2023-06-06:
Have a time horizon.      Or you can get involved in testing.   Thanks

AgainPsychoX, 2024-02-25:
While current version of library might not support changing pixel format nor frame size on the fly, I have some success with re-initializing the camera. You need to use `esp_camera_deinit()` and `esp_camera_init(&camera_config)` with desired `pixel_format`/`frame_size` there.