espressif/esp32-camera #546: esp32s2 + ov2640 (v5.1-rc1-36-g4bc762621d) failed to init camera?

chopin1998, 2023-06-19:
[image]

i have setup pin config, then it shows like this
E (1211) ledc: ledc_channel_config(584): gpio_num argument is invalid
E (1217) camera_xclk: ledc_channel_config failed, rc=102
E (1223) cam_hal: cam_init(341): ll_cam initialize failed
E (1230) camera: Camera init failed with error 0xffffffff
E (1236) example:take_picture: Camera Init Failed

ledc is for camera sensor clock input? my ov2640 have internal clock.

how to resolve the problem?

chopin1998, 2023-06-20:
i force xclk to a unused pin agaist -1 , then it works, 

how can i use the driver without xclk output?

WangYuxin-esp, 2023-06-21:
If the sensor you are using is already connected to an external clock source, it is indeed necessary to configure the XCLK pin to -1.

chopin1998, 2023-06-21:
the problem is if i config the XCLK pin to -1, init camera will be failed....

WangYuxin-esp, 2023-06-21:
@me-no-dev Hello, for some reasons, some sensors may use an external clock source to provide XCLK. Can we take a look at how to improve this issue?

chopin1998, 2023-06-21:
[image]

maybe in ll_cam.c need add some if/else statment, if pin_xclk >= 0, then enable clock output

me-no-dev, 2023-06-21:
This should actually get handled in `camera_enable_out_clock`, which will fix this for ESP32 and ESP32-S2

me-no-dev, 2023-06-21:
@WangYuxin-esp feel free to add the PR, or I'll do it in a later time.

closed 2023-08-27 by github-actions[bot]