espressif/esp32-camera #548: No OTA (board reboot) if fb_count is > 1

GnomiBerlin, 2023-06-24:
When changing from Board Manager 1.0.6 to 2.0.9 and the required adaptations due to missing downward compatibility, OTA is causing a reboot at firmware upload.

For records:

- arduino ide, esp32 v2.0.9 board manager / 1.8.19 IDE
- CameraWebServer adapted example with disabled  CameraServer just to use single picture retrieval  with fb = esp_camera_fb_get();
- esp32 camera hardware (ESP32-S). Chip ESP32-D0WD-V3 (revision 3)
- ov2640 camera

OTA works but delivered  picture is not the last in buffer:
```
      config.fb_location = CAMERA_FB_IN_PSRAM;
      config.frame_size = FRAMESIZE_UXGA;
      config.jpeg_quality = 10;
      config.fb_count = 1; 
      config.grab_mode = CAMERA_GRAB_LATEST;
```

OTA causes board reboot during firmware upload, but delivers correct last picture:
```
      config.fb_location = CAMERA_FB_IN_PSRAM;
      config.frame_size = FRAMESIZE_UXGA;
      config.jpeg_quality = 10;
      config.fb_count = 2; //2 - 1
      config.grab_mode = CAMERA_GRAB_LATEST;
```
Thanks for any help.

WangYuxin-esp, 2023-06-26:
@GnomiBerlin When the driver is working in GRAB_LASTEST mode, esp32 always tries to cache the latest data to the framebuffer. In this mode, the `fb_count` must not be less than 2. Please find more explanations in this [issue.](https://github.com/espressif/esp32-camera/issues/545)

GnomiBerlin, 2023-06-26:
Thank you @WangYuxin-esp,
I changed the normal 
 ```
 fb = esp_camera_fb_get(); // used to get a single picture.
```
to
```
esp_camera_fb_return(esp_camera_fb_get()); //workaround @WangYuxin-esp due to not last picture delivery error
  fb = esp_camera_fb_get(); // used to get a single picture.

```
with 
```
config.fb_count = 1; 
config.grab_mode = CAMERA_GRAB_WHEN_EMPTY; //CAMERA_GRAB_LATEST;
```

Now the first picture content is correct, but when taking the second picture, my board is booting.

And if I understand you correct: the new coding DMA attempts causing this trouble, which was not happening in the old board manager 1.0.6?

GnomiBerlin, 2023-08-26:
Only workaround works - still no repair

> This issue appears to be stale. Please close it if its no longer valid.

closed 2023-11-03 by github-actions[bot]