espressif/esp32-camera #551: Compilation error in latest IDF

espzav, 2023-07-18:
Hello,
after I updated IDF to latest master, I have a compilation error in your component:

```
C:/WORK/GIT/esp-bsp/examples/display_camera/managed_components/espressif__esp32-camera/target/esp32s3/ll_cam.c: In function 'll_cam_dma_init':
C:/WORK/GIT/esp-bsp/examples/display_camera/managed_components/espressif__esp32-camera/target/esp32s3/ll_cam.c:144:19: error: 'SOC_GDMA_PAIRS_PER_GROUP' undeclared (first use in this function); did you mean 'SOC_GDMA_PAIRS_PER_GROUP_MAX'?
  144 |     for (int x = (SOC_GDMA_PAIRS_PER_GROUP - 1); x >= 0; x--) {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
      |                   SOC_GDMA_PAIRS_PER_GROUP_MAX
```

Please, could you fix it and upload to IDF component registry?
Thank you very much!

me-no-dev, 2023-07-18:
Thanks for the report @espzav . Fix is here: https://github.com/espressif/esp32-camera/commit/f595abfa0f04ec746ebdc7d21af058a11c00601e but would need a release for it to show in pacman. Please try the component from git and confirm that all is good before we do a relese

me-no-dev, 2023-07-18:
Actualy released this and more for v2.0.5 so you can re-try with the latest

closed 2023-07-18 by me-no-dev

espzav, 2023-07-18:
@me-no-dev Thank you very much for the fast fix! Compilation is ok now!