espressif/esp32-camera #552: How to fully disable cam_hal messages?

ucfncouch, 2023-07-20:
Hello, I am working in esp-idf trying to see output in the terminal from the program I wrote. It is essentially repeatedly acquiring frames until I have it stop and there are output in the terminal I want to see. The problem is that there is constant lines of **cam_hal: EV-VSYNC-OVF** being printed and what I am trying to look at is completely drowned out by these messages. Is there any way to completely disable these messages?

ucfncouch, 2023-07-24:
I will try this thank you!

ucfncouch, 2023-07-24:
I have tinkered around a bit and I am still encountering the issue. By cam_stop() is this a specific function that can be used in esp_idf? I tried to call it but says wasn't declared and I searching around online to find documentation on it and found nothing.

animevietsub, 2023-07-24:
It's a function of `"cam_hal.h"` (private_include folder) you can call `ESP_ERROR_CHECK(esp_camera_deinit()); ` instead

ucfncouch, 2023-07-25:
> It's a function of `"cam_hal.h"` (private_include folder) you can call `ESP_ERROR_CHECK(esp_camera_deinit()); ` instead

I ended up just commenting out the statement that prints that out. I am just doing some testing now so it work for the time being. I was not able to get it to stop in a different way.

Basically, what I am doing is repeatedly grabbing frames to work with them.

closed 2023-10-02 by github-actions[bot]