espressif/esp32-camera #556: FB-SIZE issue on ESP32-CAM with BF3005.

AthenasArch, 2023-08-01:
I'm having trouble configuring the BF3005 camera on the Arduino IDE with ESP32.

It returns the log:
E (8433) cam_hal: FB-SIZE: 149760 != 153600

And, if I understand correctly, the calculation is
FRAMESIZE_QVGA = 320 x 240 x 565bits color

then
320 x 240 x 2 = 153600, is that correct?

but I don't know why the camera is returning me: 149760, can you help me?

<pre>
```c
void setup() {
 
    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
    // put your setup code here, to run once:
    Serial.begin(115200);

    Serial.println("Inicializa comunicacao Serial");

    camera_config_t camera_config = {
        .pin_pwdn       = PWDN_GPIO_NUM,
        .pin_reset      = RESET_GPIO_NUM,
        .pin_xclk       = XCLK_GPIO_NUM,
        .pin_sccb_sda   = SIOD_GPIO_NUM,
        .pin_sccb_scl   = SIOC_GPIO_NUM,
        .pin_d7         = Y9_GPIO_NUM,
        .pin_d6         = Y8_GPIO_NUM,
        .pin_d5         = Y7_GPIO_NUM,
        .pin_d4         = Y6_GPIO_NUM,
        .pin_d3         = Y5_GPIO_NUM,
        .pin_d2         = Y4_GPIO_NUM,
        .pin_d1         = Y3_GPIO_NUM,
        .pin_d0         = Y2_GPIO_NUM,
        .pin_vsync      = VSYNC_GPIO_NUM,
        .pin_href       = HREF_GPIO_NUM,
        .pin_pclk       = PCLK_GPIO_NUM,

        .xclk_freq_hz   = 16000000,
        .ledc_timer     = LEDC_TIMER_0,
        .ledc_channel   = LEDC_CHANNEL_0,
        .pixel_format   = PIXFORMAT_RGB565, // PIXFORMAT_YUV422, // PIXFORMAT_JPEG,
        .frame_size     = FRAMESIZE_QVGA, // FRAMESIZE_240X240, // FRAMESIZE_QVGA, // FRAMESIZE_240X240,
        .jpeg_quality   = 6,
        .fb_count       = 2,
        .grab_mode      = CAMERA_GRAB_LATEST
    };
    camera_config.fb_location = CAMERA_FB_IN_PSRAM; 

    Serial.println("Define pinos camera");

    Serial.println("Obtem os bufers");

    // Init Camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK) {
        Serial.printf("Camera init failed with error 0x%x", err);
        return;
    }

    Serial.printf("Pixel Format: %d\n", camera_config.pixel_format);
    Serial.printf("Frame Size: %d\n", camera_config.frame_size);
    Serial.printf("JPEG Quality: %d\n", camera_config.jpeg_quality);

    tft.begin();
    tft.setRotation(0);  // 0 & 2 Portrait. 1 & 3 landscape (rotação de tela)
    tft.fillScreen(TFT_BLACK);

    // right here
    s = esp_camera_sensor_get();
    pinMode(12, INPUT_PULLUP);
    
    if (s->id.PID == BF3005_PID) {
        Serial.println("Camera BF3005 detectada");
    } else {
        Serial.print("Camera desconhecida detectada: ");
        Serial.println(s->id.PID);
        return; // Finaliza se a câmera não for a esperada
    }

    Serial.println("Configurando parametros da camera");
    //Comandos a baixo servem para controle do sensor Luz, imagem, posição... 

    Serial.println("Configuracao da camera concluida");
}

int specialEffect = 0;

void loop() {
    camera_fb_t * fb = NULL;
 
    if (buffer_ready) {
        // Swap the two frame buffers
        if (current_fb == fb1) {
        current_fb = fb2;
        } else {
        current_fb = fb1;
        }
        // Capture a new frame from the camera
        esp_camera_fb_return(current_fb);
        current_fb = esp_camera_fb_get();
    }

    if (buffer_ready) {
        // Display the current frame buffer using a fast DMA transfer
        tft.pushImageDMA(0, 0, current_fb->width, current_fb->height, (uint16_t*)current_fb->buf);
    }

    // Take Picture with Camera
    fb = esp_camera_fb_get(); 
    if(!fb) {
        Serial.println("Camera capture failed");
        return;
    }
    else{
        //tft.fillScreen(random(0xFFFF));
        fex.drawJpg((const uint8_t*)fb->buf, fb->len,0,0,240,240);
    }
    esp_camera_fb_return(fb); 
}

void dmaTransferComplete() {
    buffer_ready = true;
}
```
</pre>

E (8061) cam_hal: FB-SIZE: 149760 != 153600
E (8095) cam_hal: FB-SIZE: 149760 != 153600
E (8129) cam_hal: FB-SIZE: 149760 != 153600
E (8163) cam_hal: FB-SIZE: 149760 != 153600
E (8196) cam_hal: FB-SIZE: 149760 != 153600
E (8230) cam_hal: FB-SIZE: 149760 != 153600
E (8264) cam_hal: FB-SIZE: 149760 != 153600
E (8298) cam_hal: FB-SIZE: 149760 != 153600
E (8331) cam_hal: FB-SIZE: 149760 != 153600
E (8365) cam_hal: FB-SIZE: 149760 != 153600
E (8399) cam_hal: FB-SIZE: 149760 != 153600
E (8433) cam_hal: FB-SIZE: 149760 != 153600
E (8466) cam_hal: FB-SIZE: 149760 != 153600
E (8500) cam_hal: FB-SIZE: 149760 != 153600
E (8534) cam_hal: FB-SIZE: 149760 != 153600
E (8568) cam_hal: FB-SIZE: 149760 != 153600
E (8601) cam_hal: FB-SIZE: 149760 != 153600

WangYuxin-esp, 2023-08-08:
Hello, @AthenasArch , Can it work properly when configured as VGA ?

Abhay-2412, 2023-09-15:
Hello i'am also getting same error , i'am uisng OV5640 camera and trying to use `PIXFORMAT_YUV422` pixel format with frame size `FRAMESIZE_HVGA` below is the camera configuration

```
esp_err_t initCameraSettings(void)
{
  //Camera configuration
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 24000000;
  config.pixel_format = PIXFORMAT_YUV422;
  if (psramFound())
  {
    config.frame_size = FRAMESIZE_HVGA;
    config.jpeg_quality = 20;
    config.fb_count = 2;
    config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
    config.fb_location  = CAMERA_FB_IN_PSRAM;
  }
  else
  {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }
```

and i'am getting below logs
```

E (50637) cam_hal: FB-SIZE: 294400 != 960000
E (51459) cam_hal: FB-SIZE: 294400 != 960000
E (52281) cam_hal: FB-SIZE: 294400 != 960000
E (53103) cam_hal: FB-SIZE: 294400 != 960000
E (53925) cam_hal: FB-SIZE: 294400 != 960000
```

and an important thing to be noted is `PIXFORMAT_JPEG` works fine its not throwing any error and when i try to use `PIXFORMAT_YUV422` IN Arduino IDE it's work fine

AthenasArch, 2023-09-15:
Man, this could be your camera's resolution size setting. I adjusted the settings to what the datasheet says and it worked.

This software has been tested on the camera BF3005.

void setup() {
        WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
  Serial.begin(115200);
  delay(7000);
  Serial.setDebugOutput(true);
  Serial.println();
        Serial.println("V1.0.16 - Inicializa comunicacao Serial");
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 10000000;
  config.pixel_format = PIXFORMAT_RGB565; // PIXFORMAT_GRAYSCALE; //PIXFORMAT_YUV422; // PIXFORMAT_YUV420; // PIXFORMAT_RGB565;
//   config.sccb_i2c_port = 1;
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(psramFound()){
    Serial.println("SIM - ENCONTROU PSRAM");
    config.frame_size = FRAMESIZE_HQVGA;
    config.jpeg_quality = 4;
    config.fb_count = 6;
    config.grab_mode = CAMERA_GRAB_LATEST; 
    config.fb_location = CAMERA_FB_IN_PSRAM;
  } else {
    Serial.println("NAO - find PSRAM");
    config.frame_size = FRAMESIZE_VGA;
    config.jpeg_quality = 12;
    config.fb_count = 2;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  } else {
    Serial.println("Camera init OK.");
  }

  sensor_t * s = esp_camera_sensor_get();

  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    Serial.println("Modelo da camera OV3660_PID");
    s->set_vflip(s, 1); // flip it back
    s->set_brightness(s, 1); // up the brightness just a bit
    s->set_saturation(s, -2); // lower the saturation
  } else if (s->id.PID == BF3005_PID){
    Serial.println("Modelo da camera BF3005_PID");

  }
  // drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

Abhay-2412, 2023-09-18:
hello @AthenasArch  what about ov5640 its not suporting yuv422 format in esp idf but when i run same code in arduino ide and flash it it works perfectly do i need to change any menuconfig settings can you tell me and what's this error mean  `E (53925) cam_hal: FB-SIZE: 294400 != 960000` and why this `960000` is constant i think i need to change this but i donno how to

AthenasArch, 2023-09-19:
This error occurs when the format you are trying to send to the camera is different from the ideal format for the camera. It is usually related to camera resolution and frequency. I recommend you set a lower frequency, closer to 10000000

```C
     config.xclk_freq_hz = 10000000;
     // see datasheet to configure this: pixel_format. 
     config.pixel_format=PIXFORMAT_RGB565; //PIXFORMAT_GRAYSCALE; //PIXFORMAT_YUV422; //PIXFORMAT_YUV420; //PIXFORMAT_RGB565;
     // if PSRAM IC is present, boot with UXGA resolution and higher JPEG quality
     // for larger pre-allocated frame buffer.
     config.frame_size=FRAMESIZE_HQVGA;
     config.jpeg_quality = 4; // See ddatasheet
     config.fb_count = 6; // See ddatasheet
     config.grab_mode = CAMERA_GRAB_LATEST;
     config.fb_location = CAMERA_FB_IN_PSRAM; // if you use PSRAM.

Abhay-2412, 2023-09-20:
Hello @AthenasArch, I appreciate your response. Regarding the YUV422 format, I understand that a frequency of 24MHz is required. I have successfully implemented the same configuration in the Arduino IDE, and I am able to view photos. Therefore, I believe that there might be some adjustments needed in the menuconfig settings. If you require any additional information, please let me know. Thank you.

Abhay-2412, 2023-09-20:
Hello @AthenasArch i have tested what you have suggested by lowering the frequency still im getting same error no change