espressif/esp32-camera #561: Displaying the digitized data of an analog camera on an RGB LCD

farhadvahdani, 2023-08-30:
I intend to receive the digitized output of an analog camera through ESP32S3, which includes 8 data lines and also pclk, hsync, and vsync signals. This output is in BT.601 format. Then, I will use the LCD_CAM peripheral to receive the data and a YUV422 to RGB565 converter to convert the data to RGB565 format. Finally, I will display the RGB565 data on an RGB LCD screen.
I have modified the LCD_CAM code for my own purpose, but unfortunately I am receiving corrupted data with a length shorter than my expected data of 960*625. I am completely sure that the LCD part is correct, and the problem is definitely in the LCD_CAM or GDMA part. If you can help me, I would be grateful. My pclk is also 37 MHz and my data is 960*625 in YUV422 format, and Hs and Vs are also completely correct. If you can tell me what changes to make in the LCD_CAM and GDMA parts to get the data correctly, I would be grateful.

WangYuxin-esp, 2023-09-05:
Hi, @farhadvahdani , The 960*625 is not a common resolution size. I suggest using a commonly used resolution, such as 640 * 480, for testing first.

farhadvahdani, 2023-09-11:
Hi,WangYuxin-esp, My analog camera is PAL type with a resolution of 960x625 and is not changeable. I have given the digitalized input camera output to esp32s3, the format of digital data is also bt601.

closed 2023-09-11 by farhadvahdani