espressif/esp32-camera #562: camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

huangdiajd, 2023-09-06:
I (1843) cam_hal: cam init ok
I (1843) sccb: pin_sda 22 pin_scl 23
I (1853) sccb: sccb_i2c_port=1
I (1853) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:0
E (1893) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (1893) ESP32-CAM text: Camera Init Failed

If I use the lens of ov2640, there is no problem, but when I use the lens of sc031gs, I get this error. Of course, I also looked at other similar errors and modified them according to their solutions, but finally found that this error still appeared. Could you tell me how to solve this problem for sc031gs this lens should be?

WangYuxin-esp, 2023-09-06:
@huangdiajd , Hello, Please check your configuration menu to confirm that the sensor is enabled in components>Camera configuration.

huangdiajd, 2023-09-06:
> @huangdiajd , Hello, Please check your configuration menu to confirm that the sensor is enabled in components>Camera configuration.

Hello, thank you very much for your reply. My menu has already been set up, including checking the Camera configuration to Support SC031GS VGA, and setting the SCCB clk frequency to 100000. Is there anything else I need to change?

WangYuxin-esp, 2023-09-06:
Here is a simple [example](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_SC031GS_camera_sensor_support/examples/basic/main/take_picture.c) that you can compile and use between.

huangdiajd, 2023-09-11:
> Here is a simple [example](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_SC031GS_camera_sensor_support/examples/basic/main/take_picture.c) that you can compile and use between.

W (3388) ESP32-CAM text: fb get failed
W (3388) ESP32-CAM text: fb get failed
W (3398) ESP32-CAM text: fb get failed
W (3398) ESP32-CAM text: fb get failed
W (3408) ESP32-CAM text: fb get failed
W (3408) ESP32-CAM text: fb get failed
W (3408) ESP32-CAM text: fb get failed
W (3418) ESP32-CAM text: fb get failed
W (3418) ESP32-CAM text: fb get failed
I (3428) ESP32-CAM text: fps=225.886780, image_average_size=0
I (3428) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (3438) cam_hal: cam init ok
I (3448) sccb: pin_sda 22 pin_scl 23
I (3448) sccb: sccb_i2c_port=1
I (3458) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (3488) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (3488) ESP32-CAM text: Camera Init Failed

Hello, I also tried the example you gave me, but it still failed.

huangdiajd, 2023-09-11:
> Here is a simple [example](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_SC031GS_camera_sensor_support/examples/basic/main/take_picture.c) that you can compile and use between.
Hello, is there any need to configure the parameter configuration of the 031gs camera and ov2640, or is there any difference between the configuration?

huangdiajd, 2023-09-26:
1139) cpu_start: ELF file SHA256:  3b837f7a19004b17...
I (1145) cpu_start: ESP-IDF:          v5.1-dirty
I (1151) cpu_start: Min chip rev:     v0.0
I (1155) cpu_start: Max chip rev:     v0.99 
I (1160) cpu_start: Chip rev:         v0.2
I (1165) heap_init: Initializing. RAM available for dynamic allocation:
I (1172) heap_init: At 3FC97C58 len 00051AB8 (326 KiB): DRAM
I (1179) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1185) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1192) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (1198) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1206) spi_flash: detected chip: gd
I (1210) spi_flash: flash io: dio
W (1214) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (1227) sleep: Configure to isolate all GPIO pins in sleep state
I (1234) sleep: Enable automatic switching of GPIO sleep configuration
I (1242) app_start: Starting scheduler on CPU0
I (1246) app_start: Starting scheduler on CPU1
I (1246) main_task: Started on CPU0
I (1256) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1266) main_task: Calling app_main()
I (1266) s3 ll_cam: DMA Channel=4
I (1276) cam_hal: cam init ok
I (1276) sccb: pin_sda 21 pin_scl 14
I (1276) sccb: sccb_i2c_port=1
E (1296) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (1296) example:take_picture: init camrea sensor fail
I (1296) main_task: Returned from app_main()

Hello, I am using ESP32 S3 to drive the 031GS camera and still get this error. I have selected 031GS on menuconfig. However, I have tried to change the lens to ov2640 and there is no problem. Finally, I would like to ask, what parameter configuration of 031GS, such as xclk, pixel size, etc., should be configured?

huangdiajd, 2023-09-26:
#ifndef _ESP32_CONFIG__H_
#define _ESP32_CONFIG__H_

// Camera
#define CAMERA_PIN_PWDN                 -1
#define CAMERA_PIN_RESET                -1
#define CAMERA_PIN_XCLK                 GPIO_NUM_10
#define CAMERA_PIN_SIOD                 GPIO_NUM_21
#define CAMERA_PIN_SIOC                 GPIO_NUM_14
#define CAMERA_PIN_D7                   GPIO_NUM_11
#define CAMERA_PIN_D6                   GPIO_NUM_9
#define CAMERA_PIN_D5                   GPIO_NUM_8
#define CAMERA_PIN_D4                   GPIO_NUM_6
#define CAMERA_PIN_D3                   GPIO_NUM_4
#define CAMERA_PIN_D2                   GPIO_NUM_2
#define CAMERA_PIN_D1                   GPIO_NUM_3
#define CAMERA_PIN_D0                   GPIO_NUM_5
#define CAMERA_PIN_VSYNC                GPIO_NUM_13
#define CAMERA_PIN_HREF                 GPIO_NUM_12
#define CAMERA_PIN_PCLK                 GPIO_NUM_7

#endif

#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_timer.h"
#include "esp_camera.h"
#include "ESP32_config.h"

static const char *TAG = "example:take_picture";

static esp_err_t init_camera(uint32_t xclk_freq_hz, pixformat_t pixel_format, framesize_t frame_size, uint8_t fb_count)
{
    camera_config_t camera_config = {
        .pin_pwdn = CAMERA_PIN_PWDN,
        .pin_reset = CAMERA_PIN_RESET,
        .pin_xclk = CAMERA_PIN_XCLK,
        .pin_sscb_sda = CAMERA_PIN_SIOD,
        .pin_sscb_scl = CAMERA_PIN_SIOC,

        .pin_d7 = CAMERA_PIN_D7,
        .pin_d6 = CAMERA_PIN_D6,
        .pin_d5 = CAMERA_PIN_D5,
        .pin_d4 = CAMERA_PIN_D4,
        .pin_d3 = CAMERA_PIN_D3,
        .pin_d2 = CAMERA_PIN_D2,
        .pin_d1 = CAMERA_PIN_D1,
        .pin_d0 = CAMERA_PIN_D0,
        .pin_vsync = CAMERA_PIN_VSYNC,
        .pin_href = CAMERA_PIN_HREF,
        .pin_pclk = CAMERA_PIN_PCLK,

        //EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
        .xclk_freq_hz = xclk_freq_hz,
        .ledc_timer = LEDC_TIMER_0, // This is only valid on ESP32/ESP32-S2. ESP32-S3 use LCD_CAM interface.
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = pixel_format, // YUV422,GRAYSCALE,RGB565,JPEG
        .frame_size = frame_size,    // QQVGA-UXGA, sizes above QVGA are not been recommended when not JPEG format.

        .jpeg_quality = 12, //0-63
        .fb_count = fb_count,       // For ESP32/ESP32-S2, if more than one, i2s runs in continuous mode. Use only with JPEG.
        .fb_location = CAMERA_FB_IN_PSRAM
    };

    //initialize the camera
    esp_err_t ret = esp_camera_init(&camera_config);

    return ret;
}

static uint32_t camera_test_fps(uint16_t times)
{
    uint32_t image_size = 0;
    uint32_t ret;
    ESP_LOGI(TAG, "satrt to test fps");
    esp_camera_fb_return(esp_camera_fb_get());
    esp_camera_fb_return(esp_camera_fb_get());

    uint64_t total_time = esp_timer_get_time();
    for (size_t i = 0; i < times; i++) {
        camera_fb_t *pic = esp_camera_fb_get();
        if (NULL == pic) {
            ESP_LOGW(TAG, "fb get failed");
            continue;
        }

        image_size += pic->len;
        esp_camera_fb_return(pic);
    }
    total_time = esp_timer_get_time() - total_time;
    float fps = times / (total_time / 1000000.0f);
    ret = image_size / times;
    ESP_LOGI(TAG, "fps=%f, image_average_size=%lu", fps, ret);
    return ret;
}

void app_main()
{
    if (ESP_OK != init_camera(16 * 1000000, PIXFORMAT_GRAYSCALE, FRAMESIZE_96X96, 2)) {
        ESP_LOGE(TAG, "init camrea sensor fail");
        return;
    }

    camera_test_fps(300);

    while (1) {
        ESP_LOGI(TAG, "Taking picture...");
        camera_fb_t *pic = esp_camera_fb_get();

        // use pic->buf to access the image
        ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", pic->len);
        esp_camera_fb_return(pic); // to enable the frame buffer can be used again.

        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

This is the code you sent me for reference now

closed 2023-10-11 by huangdiajd