espressif/esp32-camera #563: Request : Support new resolution (640 x 480) for SC031GS

bbozdagli, 2023-09-07:
Can you improve SC031GS camera driver for  640x480 resolution?
Currently 200x200 resolution supporting for SC031GS camera.

WangYuxin-esp, 2023-09-07:
@bbozdagli Hi, Debugging a camera driver involves many parameters, so this is not an easy task. If you have a new idea or project, you can submit your requirements [here](https://www.espressif.com/zh-hans/contact-us/technical-inquiries).

WangYuxin-esp, 2023-10-10:
Hello, @bbozdagli I have obtained a VGA configuration from the original factory of the camera sensor, please refer to this [MR](https://github.com/espressif/esp32-camera/pull/581/files).

bbozdagli, 2023-10-11:
Thanks.

10 Eki 2023 Sal 11:31 tarihinde yuxinwww ***@***.***> şunu
yazdı:

Eduardo-Brandt, 2023-11-13:
> Hello, @bbozdagli I have obtained a VGA configuration from the original factory of the camera sensor, please refer to this [MR](https://github.com/espressif/esp32-camera/pull/581/files).

**Very cool work Mr. WangYuxin-esp!**, it really enables 640x480 resolution on SC031GS. **Thank you a lot!**
But I´ve got distorted images at 640x480 resolution. Can it be because of the high PCLK(Pixel Clock) of 25MHz? Any suggestion?

See some results I´ve gotten (Pictures from my oscilloscope):
pic1:
[image]
pic2:
[image]
pic3:
[image]

With lower resolutions, the results are better:
[image]

Eduardo-Brandt, 2023-11-16:
I have discovered that the problem was because of the BUS cable lenght. Mainly because of the PCLK (Pixel Clock) pin, that carries about 25MHz clock. I got very better results when I shortened the cables from 10cm to about 15mm of lenght. The 3.3V to 1.8V converters may worsen the problem also.

WangYuxin-esp, 2023-11-16:
Yes, the best method is to use FPC flexible cables smaller than 1cm. When the resolution and frame rate increase, the corresponding PCLK will increase significantly, making it the highest frequency clock signal.

nds-str, 2024-01-02:
Hi guys.
I'm trying to use the SC130GS sensor. Changed PID for SC031GS. All I need is a single frame with maximum resolution, the frequency is not important. But, I don’t see the PIXCLK signal from the sensor. What range does PIXCLK operate in? How can I select the minimum PIXCLK frequency? Can you recommend the register settings? Is this sensor defective?

Eduardo-Brandt, 2024-01-08:
> Hi guys. I'm trying to use the SC130GS sensor. Changed PID for SC031GS. All I need is a single frame with maximum resolution, the frequency is not important. But, I don’t see the PIXCLK signal from [...]

- I did it work using the ESP-IDF compiller ins VSCode (Visual Studio Code). 
- It is mandatory to use 3.3V to 1.8V voltage converters. SC130GS damage instantly if his BUS is powered with 3.3V.
- Maxgerad helpe me to put it working using ESp-IDF. It is necessary to use VSCode command **pio run -t menuconfig** to configure/Enable PSSam etc... See the post:  [https://community.platformio.org/t/arduino-esp-camera-h-lib-with-sc031gs-camera-have-anyone-tried/36246/5](url)
- Use the last version of Camera Lib, with the fix that allows 640x480 resolution. Otherwise it will work only on 200x200 resolution (that is not standard and you have to enable manually inside Camera lib files,... a problem).

Hope it helps. See Maxherald post info, that is very useful!

-

Eduardo-Brandt, 2024-01-08:
> Thank you, @Eduardo-Brandt Have you worked with the SC130GS?
> 
> I'm using ESP_IDF with VSCODE. And there are no problems using libraries. I'm trying to connect a circuit like https://ae04.alicdn.com/kf/H8942a452293740fe907b350cdfbdb752C.jpg to the ESP32-CAM module.
> 
> I had a problem assigning pins. The built-in library did not allow the "soft reset" function to be executed. I figured this out. I received the sensor response via I2C, I can configure the registers. The HSYNC and VSYNC signals change, I see frames and lines, but there is no pixel signal. The library gives the error "**cam_hal: EV-VSYNC-OVF**" even when I set the resolution to less than 200x200.
> 
> But, I realized that my sensor can only output PCLK ~37Mhz (if the input XCLK=27MHz). When the input frequency decreases, the PCLK signal disappears. I have a problem setting up the required PCLK signal. It is probably configured by setting registers. But the configuration files "[sc031gs_settings.h](https://github.com/espressif/esp32-camera/blob/master/sensors/private_include/sc031gs_settings.h)" for SC031GS don't work for me.
> 
> My datasheet for the SC130GS is too light and does not allow me to control the PCLK frequency. I have no experience working with video sensors. Perhaps there is more detailed documentation on the SC130GS sensor?
> 
> PS: I use an oscilloscope and can remove the built-in crystal oscillator on my SC130GS board. Maybe Mr. @WangYuxin-esp can help me?

I not use the built-in crystal oscillator. Have you tried that? I was able to reach as low as 3Mhz of Pclk (SC031 datasheet´s says it´s minimum is 6MHz).
 Do you have the link where you bought your board?

"cam_hal: EV-VSYNC-OVF" error means the ESP32 received a string larger than the expected packet. Example, a 200x200 resolutions means a 40kBytes packet(1 byte per pixel).

nds-str, 2024-01-09:
Hi @Eduardo-Brandt
I bought a board like this [aliexpress](https://fr.aliexpress.com/item/4000290476876.html?gatewayAdapt=glo2fra)

The built-in crystal has a frequency of 27MHz. I used it first.
I removed it and used an ESP32 generator. But the pixclk(PCLK) output is stable only at the input frequency ~27MHz(xclk_freq_hz) (it was ~37MHz(PCLK)).
PCLK became very noisy when lowered to 20MHz.

The sensor responds via I2C even at a frequency of 5MHz.
At 1MHz I2C does not respond. I haven't checked other frequencies.

Eduardo-Brandt, 2024-01-09:
> Hi @Eduardo-Brandt I bought a board like this [aliexpress](https://fr.aliexpress.com/item/4000290476876.html?gatewayAdapt=glo2fra)
>
> The built-in crystal has a frequency of 27MHz. I used it first. I [...]

Is it noisy even before the voltage level translator?

closed 2024-03-18 by github-actions[bot]