espressif/esp32-camera #565: Undefined Reference (IDFGH-11027)

Abhay-2412, 2023-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

iam trying to import and Arduino project to esp idf by using arduino as component but iam facing some undefiend reference issue and stuck with that i appreciate anyone who helps me 

[1/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.5\workspace\GAZZELE_MINI\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.5/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/build/bootloader/bootloader.bin"
Bootloader binary size 0x63f0 bytes. 0xc10 bytes (11%) free.
[2/4] Linking CXX executable GAZZELE_MINI.elf
FAILED: GAZZELE_MINI.elf 
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\GAZZELE_MINI.elf.rsp -o GAZZELE_MINI.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Camera_Settings.cpp.obj):(.literal.initCameraSettings+0x18): undefined reference to `esp_camera_init'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Camera_Settings.cpp.obj):(.literal.initCameraSettings+0x1c): undefined reference to `esp_camera_sensor_get'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Camera_Settings.cpp.obj):(.literal.capturePhotoSaveSpiffs+0x38): undefined reference to `esp_camera_fb_get'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Camera_Settings.cpp.obj):(.literal.capturePhotoSaveSpiffs+0x40): undefined reference to `esp_camera_fb_return'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Camera_Settings.cpp.obj): in function `initCameraSettings':
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/DriverModules/CamDrivers/Camera_Settings.cpp:56: undefined reference to `esp_camera_init'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/DriverModules/CamDrivers/Camera_Settings.cpp:56: undefined reference to `esp_camera_sensor_get'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/DriverModules/CamDrivers/Camera_Settings.cpp:72: undefined reference to `esp_camera_sensor_get'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Camera_Settings.cpp.obj): in function `capturePhotoSaveSpiffs':
C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/DriverModules/CamDrivers/Camera_Settings.cpp:194: undefined reference to `esp_camera_fb_get'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/DriverModules/CamDrivers/Camera_Settings.cpp:209: undefined reference to `esp_camera_fb_return'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.

igrr, 2023-09-08:
Have moved this from esp-idf repo. Looks like we need to add C++ guards in esp_camera.h and tag a new release.

Abhay-2412, 2023-09-08:
so what to do, this project works fine in Arduino or is there any other alternative

Abhay-2412, 2023-09-08:
can you specify what you want to check clearly ill provide  you with that

igrr, 2023-09-08:
Ah yes, my bad, I missed them when looking at the header file the first time:
https://github.com/espressif/esp32-camera/blob/d1c9c2cdb3fab523e81e8d953305c00ed54c834c/driver/include/esp_camera.h#L81-L83

@Abhay-2412 please specify:

- Which version of esp32-camera you are using
- How are you adding esp32-camera to your project?
  - Cloning it as a submodule to project components/ subdirectory?
    - If yes, please show the content of C:/Espressif/frameworks/esp-idf-v4.4.5/workspace/GAZZELE_MINI/main/CMakeLists.txt
  - Or, are you getting it from the component manager, using idf_component.yml file?
  - Or somehow else?

Abhay-2412, 2023-09-08:
I'am a newbie to all this thing i didn't get you what you meant by esp32-camera version and I cloned Arduino as a component to an ESP-IDF project in that project I gave file structure as shown below

```
main
└── DriverModules
    ├── CamDrivers
    │   ├── Camera_settings.cpp
    │   ├── Camera_settings.h
    │   ├── cam_hal.c
    │   ├── cam_hal.h
    │   ├── esp_camera.c
    │   ├── esp_camera.h
    │   ├── some_other_header1.h
    │   ├── some_other_header2.h
    │   └── some_other_header3.h
    ├── HVDCDriver
    │   ├── Digipot_MCP4017.cpp
    │   ├── Digipot_MCP4017.h
    │   ├── HVDC_ADC_Settings.cpp
    │   └── HVDC_ADC_Settings.h
    ├── LEDdimmerDriver
    │   ├── PCA9352_LED_dimmer.cpp
    │   └── PCA9352_LED_dimmer.h
    └── WifiDriver
        ├── WIFI_client_settings.cpp
        └── WIFI_client_settings.h
```
below is my CMakeLists.txt

```
idf_component_register(SRCS 
    "main.cpp" 
    "DriverModules/WifiDriver/WIFI_Client_Settings.cpp"
    "DriverModules/LEDdimmerDriver/PCA9352_LED_dimmer.cpp"
    "DriverModules/CamDrivers/Camera_Settings.cpp"
    "DriverModules/HVDCDriver/Digipot_MCP4017.cpp"
    "DriverModules/HVDCDriver/HVDC_ADC_Settings.cpp"
    "DriverModules/BatteryDriver/BQ25601RTWR.cpp"
    
    INCLUDE_DIRS
    "."
    "DriverModules/WifiDriver"
    "DriverModules/LEDdimmerDriver"
    "DriverModules/CamDrivers"
    "DriverModules/BatteryDriver"
    "DriverModules/HVDCDriver"
  
    )

```
I apologize for any mistakes I may make as I am still learning. Please let me know if there is any additional information you would like me to provide, as I am here to assist you.

Abhay-2412, 2023-09-11:
Iam so stuck at this point i donno how to proceed further I appreciate if anyone helps me with this

Abhay-2412, 2023-09-12:
> Okay, looks like you have added the source files but you haven't referenced them in CMakeLists.txt. This won't work because the files are not compiled. My suggestion is to simplify the whole thing, getting esp32-camera from the [Component Registry](https://components.espressif.com/components/espressif/esp32-camera).
> 
> 1. Delete these files:
>    ```
>     ├── CamDrivers
>     │   ├── Camera_settings.cpp
>     │   ├── Camera_settings.h
>     │   ├── cam_hal.c
>     │   ├── cam_hal.h
>     │   ├── esp_camera.c
>     │   ├── esp_camera.h
>    ```
> 2. Create `main/idf_component.yml` with the following content:
>    ```yaml
>     dependencies:
>       espressif/esp32-camera: "^2.0.5"
>    ```
> 3. Run `idf.py reconfigure` and then `idf.py build`.

> Okay, looks like you have added the source files but you haven't referenced them in CMakeLists.txt. This won't work because the files are not compiled. My suggestion is to simplify the whole thing, getting esp32-camera from the [Component Registry](https://components.espressif.com/components/espressif/esp32-camera).
> 
> 1. Delete these files:
>    ```
>     ├── CamDrivers
>     │   ├── Camera_settings.cpp
>     │   ├── Camera_settings.h
>     │   ├── cam_hal.c
>     │   ├── cam_hal.h
>     │   ├── esp_camera.c
>     │   ├── esp_camera.h
>    ```
> 2. Create `main/idf_component.yml` with the following content:
>    ```yaml
>     dependencies:
>       espressif/esp32-camera: "^2.0.5"
>    ```
> 3. Run `idf.py reconfigure` and then `idf.py build`.

X
  
    
    
 "I believe I referenced them in the following manner. 
 ```

     INCLUDE_DIRS
    "."
    "DriverModules/WifiDriver"
    "DriverModules/LEDdimmerDriver"
    "DriverModules/CamDrivers"
    "DriverModules/BatteryDriver"
    "DriverModules/HVDCDriver"
```
 Is that the correct approach? I will certainly try your suggestion, but I am also curious about the proper way to reference sources and compile them in this manner."

igrr, 2023-09-12:
> INCLUDE_DIRS
> Is that the correct approach?
INCLUDE_DIRS sets the list of include directories, more formally known as header file search directories — it's the list of directories this component provides where the header files are located. In other words, you can `#include` a header file located in one of these directories.

To set the list of sources, you need to add files to the SRCS list, which you already did for some of the other files.

Note though, unless you understand the library internals well, copying the source and header files into your project and adding them manually is not a reliable approach. There could be something in the library's CMakeLists.txt, for example, that you would be missing. You are also missing Kconfig file of this library this way, so all the options the this library relies upon won't be defined, and the resulting program will likely not work correctly.

Abhay-2412, 2023-09-12:
> Thank you for your prompt response. Based on your suggestion, I will proceed with the recommended approach. Currently, I am in the process of porting an Arduino project, which is functioning successfully, to the ESP-IDF platform. My approach involves directly transferring the camdrivers file from the original project to the new platform.

so is it fine to use them or should i go with your approach and what about `Camera_settings.cpp` i include `esp_camera.h` in this file so where should i add this file since you want me to delete below files
```

 ├── CamDrivers
 │   ├── Camera_settings.cpp

 │   ├── Camera_settings.h
 │   ├── cam_hal.c
 │   ├── cam_hal.h
 │   ├── esp_camera.c
 │   ├── esp_camera.h
```

igrr, 2023-09-12:
Basically, 1) delete all the source and header files of esp32-camera from your project, 2) add the project using component manager as explained above. That's the approach which will be the easiest for us to help you with.

Abhay-2412, 2023-09-12:
yeah

> Basically, 1) delete all the source and header files of esp32-camera from your project, 2) add esp32-camera using the component manager as explained above. That's the approach which will be the easiest for us to help you with.

yeah i get it, sorry if my last comment is unclear i meant i have written some camera function in `camera_settings.cpp` so should this file should be in same folder as `idf_component.yml` 

if so can i add this file in cam driver folder that is 

main/DriverModules/CamDrivers/idf_component.yml

Abhay-2412, 2023-09-12:
hello i have made some changes in `CMakeLists.txt` as shown below

```
idf_component_register(SRCS 
    "main.cpp" 
    "DriverModules/WifiDriver/WIFI_Client_Settings.cpp"
    "DriverModules/LEDdimmerDriver/PCA9352_LED_dimmer.cpp"
    "DriverModules/CamDrivers/Camera_Settings.cpp"
    "DriverModules/CamDrivers/cam_hal.c"
    "DriverModules/CamDrivers/esp_camera.c"
    "DriverModules/CamDrivers/ll_cam.c"
    "DriverModules/CamDrivers/ov2640.c"
    "DriverModules/CamDrivers/ov5640.c"
    "DriverModules/CamDrivers/sccb.c"
    "DriverModules/CamDrivers/sensor.c"
    "DriverModules/CamDrivers/xclk.c"
    "DriverModules/CamDrivers/Camera_Settings.cpp"
    "DriverModules/CamDrivers/Camera_Settings.cpp"
    "DriverModules/HVDCDriver/Digipot_MCP4017.cpp"
    "DriverModules/HVDCDriver/HVDC_ADC_Settings.cpp"
    "DriverModules/BatteryDriver/BQ25601RTWR.cpp"
    
    INCLUDE_DIRS
    "."
    "DriverModules/WifiDriver"
    "DriverModules/LEDdimmerDriver"
    "DriverModules/CamDrivers"
    "DriverModules/BatteryDriver"
    "DriverModules/HVDCDriver"
  
    )
    
   
```
   Again tried to build it got build successfully
   ```

   Total sizes:
Used static DRAM:   37440 bytes (  87140 remain, 30.1% used)
      .data size:   19104 bytes
      .bss  size:   18336 bytes
Used static IRAM:   96538 bytes (  34534 remain, 73.7% used)
      .text size:   95511 bytes
   .vectors size:    1027 bytes
Used Flash size :  692588 bytes
      .text     :  563123 bytes
      .rodata   :  129209 bytes
Total image size:  808230 bytes (.bin may be padded larger)
```
  i flashed the code try to run the device but this time the device is crashing and getting reboot everytime
  
```
[0;32mI (1200) cam_hal: cam init ok[0m
[0;32mI (1200) sccb: pin_sda 26 pin_scl 27[0m
[0;31mE (1202) i2c: i2c driver install error[0m
[0;32mI (1214) camera: Detected camera at address=0x3c[0m
[0;31mE (1215) ledc: ledc_stop(661): LEDC is not initialized[0m
[0;31mE (1215) camera: Detected camera not supported.[0m
[0;31mE (1217) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)[0m
[0;31mE (1221) ledc: ledc_stop(661): LEDC is not initialized[0m
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d8ef4  PS      : 0x00060e30  A0      : 0x800d82e9  A1      : 0x3ffb21e0  
A2      : 0x00000106  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x000186a0  
A6      : 0x3ffc2d5c  A7      : 0x00000001  A8      : 0x800d8ef2  A9      : 0x3ffb21c0  
A10     : 0x00000000  A11     : 0x000004c1  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x80000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000002  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9  

Backtrace: 0x400d8ef1:0x3ffb21e0 0x400d82e6:0x3ffb2220 0x400dc64a:0x3ffb2260 0x4008e199:0x3ffb2280

ELF file SHA256: 375623c71663f5ca

Rebooting...
```

Abhay-2412, 2023-09-12:
i just want to know is there any error at code side or not beacause this code work fine when compiled with arduino ide but troubling me a lot when im trying to use arduino as component in esp idf and this is my first time using it so help me out :(

Abhay-2412, 2023-09-13:
hello, I have observed that when I comment out the camera functionality in my code, everything works properly. Therefore, it appears that the crash is specifically related to the camera integration.

igrr, 2023-09-13:
> �[0;31mE (1217) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)�[0m

I have mentioned this problem in my previous comment: esp32-camera code depends on Kconfig options to enable sensor support. Since there is no Kconfig file, no options are defined, and hence support for no sensors gets added.

In Arduino-esp32, kconfig options are determined ahead of time, and already include all esp32-camera options. Therefore you haven't observed this issue there.

I would suggest following the recommendation above, removing source file copied from esp32-camera project, and adding esp33-camera via component manager.

Abhay-2412, 2023-09-13:
Hello @igrr I have attempted to implement your suggested solution for initializing the camera. However, despite successful initialization, I am encountering another issue with the `cam_hal`, and the application is crashing. I have included the relevant logs below for your reference and analysis.

```
Pullup: 0| Pulldown: 0| Intr:0
I (359) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (360) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (363) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (368) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (373) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (378) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (382) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (387) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (392) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (397) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (402) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (406) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (411) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (416) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (421) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (425) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (430) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (435) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (440) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (445) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (449) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (454) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (459) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (464) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (468) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (473) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (478) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (483) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (488) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (492) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (497) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (502) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
I (507) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
0
[... 85 lines cut ...]
├── CMakeLists.txt
└── main.cpp
```

Please find below the updated `CMakeLists.txt` file I have created:

```
idf_component_register(SRCS 
    "main.cpp" 
    "DriverModules/WifiDriver/WIFI_Client_Settings.cpp"
    "DriverModules/LEDdimmerDriver/PCA9352_LED_dimmer.cpp"
    "Camera_Settings.cpp"
    "DriverModules/HVDCDriver/Digipot_MCP4017.cpp"
    "DriverModules/HVDCDriver/HVDC_ADC_Settings.cpp"
    "DriverModules/BatteryDriver/BQ25601RTWR.cpp"
    
    INCLUDE_DIRS
    "."
    "DriverModules/WifiDriver"
    "DriverModules/LEDdimmerDriver"
    "DriverModules/BatteryDriver"
    "DriverModules/HVDCDriver"
  
    )
```

Abhay-2412, 2023-09-14:
Hello, i did some searching i got to know it's because psram is not initalised but in my code i initialised psram

 ```
/**
 *  @file GazelleMiniFw.ino
 *  @brief This source file contains main function
 *         
 *
 *  @author

 *
 *  @date
 */

/* Includes ------------------------------------------------------------------*/

#include "WiFi.h"
#include "esp_wifi.h"
#include "esp_camera.h"
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"
//#include <StringArray.h>
#include <FS.h>
#include "DriverModules/LEDdimmerDriver/PCA9352_LED_dimmer.h"
#include <Wire.h>
#include "PCA9352_LED_dimmer.h"
#include "DriverModules/HVDCDriver/HVDC_ADC_Settings.h"
#include "DriverModules/HVDCDriver/Digipot_MCP4017.h"
#include "DriverModules/BatteryDriver/BQ25601RTWR.h"
#include <WiFiClient.h>
#include<Camera_Settings.h>
#include"DriverModules/WifiDriver/WIFI_Client_Settings.h"

const int potPin = 39;

// variable for storing the potentiometer value
int potValue = 0;
#define HALL_GPIO (12)
// ADC conversion factor
float CONVERSION_ADC_FACTOR = 0;

// Camera configuration variables
camera_config_t config;
boolean takeNewPhoto = false;

//Variables for HB variant test continuous transfer
bool contTransfer = false;
int16_t counter = 0;
bool SCD_TEST = false;
bool COVID_TEST = false;
bool HB_UV = false;
bool HB_WHITE =false;
bool COVID_UV = false;
bool COVID_WHITE =false;
//Wi-Fi functionality variables
const char* wifi_ssid = "Gazelle_mini";
const char* wifi_password = "1234567890";
const uint16_t port = 48502;
char* host = "192.168.149.237";
WiFiClient client;
char cCommand = 0;

// Photo File Name to save in SPIFFS
#define FILE_PHOTO "/photo.jpg"

// Time measurement variables
unsigned long previousMillis = 0;
unsigned long interval = 30000;
const int ledPin_12VEN= 2;
const int ledPin_covid_uv = 4;
//HVDC I2C functionality variables
#define I2C_Freq 100000
#define I2C_SDA 0
#define I2C_SCL 32
TwoWire I2C_0 = TwoWire(0);

//void IRAM_ATTR isr() {
//  Serial.printf("\nHall sensor\n");
//}
// setting PWM properties
const int freq = 5000;
const int ledChannel = 0;
const int pwmresolution = 8;
int dutyCycle = 0;
String stringExit = String("Stop Transfer");
String stringSCD = String("SCDStart Transfer0");
String stringCOVID = String("COVIDStart Transfer0");

void initFrontlightPWM(void)
{
  // configure LED PWM functionalitites
  ledcSetup(ledChannel, freq, pwmresolution);
  
  // attach the channel to the GPIO to be controlled
  ledcAttachPin(ledPin_covid_uv, ledChannel);
  delay(100);
  dutyCycle = 0;
  ledcWrite(ledChannel, dutyCycle);
}

void setup() 
{

  //RAM intilization
  

psramInit();

  // Serial port for debugging purposes
  Serial.begin(230400);
     //I2C initilization
  I2C_0.begin(I2C_SDA , I2C_SCL,I2C_Freq);
  int8_t errorLEDdimmer = setBQ25601_CHARGER_CONTROL_1();
//  errorLEDdimmer = setBQ25601_CHARGER_CONTROL_3();
  errorLEDdimmer = setInputCurrentLimit();
  errorLEDdimmer = setBQ25601_CHARGE_CURRENT_LIMIT();
  errorLEDdimmer = setBQ25601_PRECHG_AND_TERM_CURR_LIM();
  // LED dimmer initilization
  errorLEDdimmer = initPCA9352_LED_dimmer();  
  if(errorLEDdimmer > ZERO)
  {
    Serial.printf("Error in initilization of LED dimmer = %d\n", errorLEDdimmer);
  }

  errorLEDdimmer = switchOn_HVDC_EN();
  if(errorLEDdimmer > ZERO)
  {
    Serial.printf("Error in initilization of LED dimmer = %d\n", errorLEDdimmer);
  }
  pinMode(ledPin_12VEN,OUTPUT);
  digitalWrite(ledPin_12VEN, LOW);
  
  errorLEDdimmer = switchOff_HVDC_EN();
  if(errorLEDdimmer > ZERO)
  {
    Serial.printf("Error in initilization of LED dimmer = %d\n", errorLEDdimmer);
  }
  float RES = 8600;
  WriteRESValue(RES);
  delay(300);
  CONVERSION_ADC_FACTOR = calculateADCCF();
  delay(300);
  errorLEDdimmer = switchOn_HVDC_EN();
  if(errorLEDdimmer > ZERO)
  {
    Serial.printf("Error in initilization of LED dimmer = %d\n", errorLEDdimmer);
  }

initFrontlightPWM();
delay(200);

////  // Camera initilization
  esp_err_t error = initCameraSettings();
  if (error != ESP_OK) 
  {
    Serial.printf("Camera init failed with error 0x%x", error);
  }

  
  //Wi-Fi initilization
  WiFi.softAP(wifi_ssid, wifi_password);
  IPAddress IP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(IP);
  
//  initWiFi();
  initClientConnection();
  
//  initSPIFFS();

}
//contineous execution 
void loop() 
{
//  check for command
  commandReceive();
  
  delay(100);
}

```

Abhay-2412, 2023-09-15:
Hello i enabled PSRAM through menuconfig now its working fine thanks for your help

closed 2023-09-15 by Abhay-2412