espressif/esp32-camera #566: ESP32S3 BT601 Video Capture Issue

farhadvahdani, 2023-09-11:
Hi,

I am trying to capture BT601 video data on the ESP32S3. I have a board that converts the video signal from an analog camera to digital data in the BT601 protocol. The digital signals contain 8 bits of data and a V_sync and H_sync signal.

I have read the user manual for the ESP32S3 and it says that the LCD_CAM module can receive data via BT601. However, I am not sure how to do this, because the LCD_CAM module has a CAM_H_ENABLE signal, but in the BT601 protocol there is no signal like this.

Can you please help me with the following:

Is it possible to capture BT601 video data by the ESP32S3 at all?

If it is possible, are there any tutorials or help you can give me?

How do I deal with the CAM_H_ENABLE signal?

Thank you for your help.

Best regards,

me-no-dev, 2023-09-11:
The important signals are V sync, data and PCLK. Does your module have that? Is there any info for the module?

farhadvahdani, 2023-09-12:
Hi @me-no-dev  Yes, my module has pclk and vsync, and they are both working properly. My pclk is 37 MHz.

farhadvahdani, 2023-09-12:
I recently modified the esp32-camera code, but I lost xclk and i2c communication. Now, I am receiving data that is about 1/4 the size of my original data, and it is not accurate.

me-no-dev, 2023-09-12:
Your PCLK is a bit high. Any chance you can get it down to 20MHz or less?

farhadvahdani, 2023-09-12:
@me-no-dev No, I can't lower it because my analog camera is a PAL type with a resolution of 960*625, and when it is digitized, Pclk becomes 37 MHz. But well, the technical manual said that if it is below 40 MHz, there will be no problem.

farhadvahdani, 2023-09-12:
@me-no-dev     Unfortunately, there is no such device available for purchase, and I built it myself to display an analog camera on an RGB LCD. I can explain it to you in more detail if you want.

farhadvahdani, 2023-09-12:
@me-no-dev  Another question is that we should set the pclk inside the lcd_cam settings. Because there is only one clock setting section in the cam_contrl register, which is for setting the clock, and nothing special happens even if it is changed.

me-no-dev, 2023-09-12:
Since the ESP is a "Slave" in this case, PCLK does not need to be set. There are other controls. Do you have a diagram of what the output signal looks like?

farhadvahdani, 2023-09-12:
you can see the signal with below link
https://drive.google.com/drive/folders/1BmXVMsANgiRItbwLHShNKgGsWn3lUHcn?usp=sharing

me-no-dev, 2023-09-12:
access requested :)

farhadvahdani, 2023-09-12:
@me-no-dev   can you see??

me-no-dev, 2023-09-12:
yes! I suggest that you play with the following values in the driver: https://github.com/espressif/esp32-camera/blob/master/target/esp32s3/ll_cam.c#L267-L272 clk/h/v inv and filter could help grab better images. Also add `. fb_location = CAMERA_FB_IN_DRAM,` to the camera config

farhadvahdani, 2023-09-12:
@me-no-dev   The internal RAM of the ESP32S3 is about 500 KB, and my frame size is 1200 KB, which is too big to fit in internal RAM. Therefore, I cannot use this element".fb_location = CAMERA_FB_IN_DRAM". I also tried to change the LCD_CAM settings that you mentioned, but I did not get any positive results.

farhadvahdani, 2023-09-12:
@me-no-dev 
I have tried almost all the LCD_CAM settings and did not get any positive results. Another question is that I am using the WROOM1-N8R2, which has a quad-connected PSRAM. Could it be that it cannot write to the PSRAM at a sufficient speed?

closed 2023-11-19 by github-actions[bot]