espressif/esp32-camera #567: Any GPIO wiring guidelines?

xXDangerous, 2023-09-11:
Hello,
I'm having some issues getting my OV5640 fully to work. Quite sure my wiring is off but I can't figure out where.
I bought a OV5640 module off of AliExpress, as well as a camera adapter (DevEBox) and a chinese ESP32-S (apparently a NodeMCU32 v1.3 clone).
So far, the way I wired things up allows me to communicate with the camera, detect and configure it and receive frames.. but the frames are broken. When trying to manually save a HEX Dump of the retrieved JPEG frames, they turn out to be corrupted.

I read in some other PRs here that it could be a power source issue, so the wiring should be checked. In another PR someone was complaining about getting some cam_hal messages on which he got the reply that it might be a VSync overflow or something.

Are there any guidelines as to how to properly connect the devices? Which PIN from the camera should be connected to what on the ESP32? Any particular requirement for each pin?

Datasheet for my board should be this one: https://docs.ai-thinker.com/_media/nodemcu32-s_specification_v1.3.pdf

And the GPIO's I assigned are:
```
#define CAM_PIN_PWDN -1
#define CAM_PIN_RESET 15
#define CAM_PIN_XCLK 18
#define CAM_PIN_SIOD 21
#define CAM_PIN_SIOC 22

#define CAM_PIN_D7 14
#define CAM_PIN_D6 27
#define CAM_PIN_D5 26
#define CAM_PIN_D4 25
#define CAM_PIN_D3 33
#define CAM_PIN_D2 32
#define CAM_PIN_D1 35
#define CAM_PIN_D0 34
#define CAM_PIN_VSYNC 4
#define CAM_PIN_HREF 2
#define CAM_PIN_PCLK 19
```

Pixel Format: PIXFORMAT_JPEG
Frame Size: FRAMESIZE_QQVGA
Jpeg Quality: 40%
FB Location: CAMERA_FB_IN_DRAM
FB Count: 1

Thank you in advance to anyone willing to help out.

me-no-dev, 2023-09-11:
Can you try with larger frame size? Like QVGA should be OK. QQVGA is a bit too small and we sometimes have issues properly grabbing the frame (yeah, a bit counter intuitive)

me-no-dev, 2023-09-11:
Other than that, your IO selection is OK. You can use the pins 34 and above for any pin but PWDN, RESET, XCLK, SIOD and SIOC (they must be output capable). That could free some other pins that you could reuse for other things.

BTW You did not mention what XCLK frequency you have selected? 10/20 MHz should be OK.

xXDangerous, 2023-09-11:
Did another test with 40 MHz XCLK and fetched 3 frames one after another. All broken, too. If the IO selection is fine, then it might be the soldering or a power issue. May have to check it tomorrow..

closed 2023-11-19 by github-actions[bot]

Hema2-official, 2024-02-16:
Hey, @xXDangerous!

I've been trying to connect my DevEBox camera adapter to my ESP32-S3 development board, but I just couldn't figure out the pinout of the adapter itself, and I didn't find any good sources either. You're the only person so far I could find.
Can you please help me out with this?