espressif/esp32-camera #570: Impossible to change the configuration after the first photo (Freenove CAM S3)

remibert, 2023-09-17:
Hello,

I do the initialization sequence, shooting configuration, and taking a photo and then taking a photo in a loop, it works. 

If after I then want to reconfigure anything, I systematically get an error (sccb: SCCB_Write Failed addr:0x30, reg:0x.., data:0x.., ret:263, 263=timeout), then taking a photo works, but in the previous configuration. I tried to call the reset function, but it had no effect. 

To be able to modify anything, I must reboot the board, and redo the init sequence. I tried to "deinit" and retry "init", but this resulted in a memory allocation problem (fragmentation certainly).

It seems that switching to image acquisition switches to a particular mode, and then it is impossible to return without rebooting. 

I have an esp32 freenove s3 camera, esp-idf 5.0.2, esp-camera latest version. The operation that I describe works perfectly well with an esp32-cam board.

Is there a call to an esp32-camera API to avoid this kind of problem?

Best regards.

WangYuxin-esp, 2023-09-22:
Hello，May I know which two working modes you want the camera to work in?

WangYuxin-esp, 2023-09-25:
For any parameters that affect the size of data output, such as quality factor, resolution size, and output format. We suggest doing anti initialization to change them. The current driver implementation is very simple, so there is no support for quickly switching between such changes. We will consider adding features that meet this requirement later.

remibert, 2023-09-25:
No problem if it's complicated, I'll leave the code deinit init. Note for those who have the same problem as me, on the reallocation problem after a deinit, init, I made a small modification to keep the memory allocated, see here. https://github.com/remibert/pycameresp/blob/main/patch/S3/c/esp32-camera/driver/cam_hal.c (line 251-253, line 300-310, line 462-463).

closed 2023-11-25 by remibert