espressif/esp32-camera #573: Does SC101IOT support a resolution of 1280 x 720?

skylli, 2023-10-01:
Does SC101IOT support a resolution of 1280 x 720? I ran the 'take picture' example with the following configuration:
.xclk_freq_hz = 20000000,
.pixel_format = PIXFORMAT_YUV422,
.frame_size = FRAMESIZE_HD.
During the program execution, the log continuously outputs: cam_hal: EV-EOF-OVF. I am using ESP32S3. Could you please let me know what is the highest supported resolution for SC101IOT? If it supports FRAMESIZE_HD, how should I configure it or which example should I run?
Here is my code:
```

esp_err_t init_camera(void)
{
    camera_config_t camera_config = {

        .pin_pwdn  = CAM_PIN_PWDN,
        .pin_reset = CAM_PIN_RESET,
        .pin_xclk = CAM_PIN_XCLK,
        .pin_sccb_sda = CAM_PIN_SIOD,
        .pin_sccb_scl = CAM_PIN_SIOC,

        .pin_d7 = CAM_PIN_D7,
        .pin_d6 = CAM_PIN_D6,
        .pin_d5 = CAM_PIN_D5,
        .pin_d4 = CAM_PIN_D4,
        .pin_d3 = CAM_PIN_D3,
        .pin_d2 = CAM_PIN_D2,
        .pin_d1 = CAM_PIN_D1,
        .pin_d0 = CAM_PIN_D0,
        .pin_vsync = CAM_PIN_VSYNC,
        .pin_href = CAM_PIN_HREF,
        .pin_pclk = CAM_PIN_PCLK,

        .xclk_freq_hz = 20000000,//CONFIG_XCLK_FREQ,
        .ledc_timer = LEDC_TIMER_0, // This is only valid on ESP32/ESP32-S2. ESP32-S3 use LCD_CAM interface.
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = PIXFORMAT_YUV422,//PIXFORMAT_YUV422, //PIXFORMAT_YUV422,//PIXFORMAT_JPEG,
        .frame_size = FRAMESIZE_HD, //FRAMESIZE_VGA,//FRAMESIZE_UXGA,//FRAMESIZE_VGA,

        .jpeg_quality = 30,//10,
        .fb_count = 2,
        .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
        .fb_location = CAMERA_FB_IN_PSRAM
        };//CAMERA_GRAB_LATEST. Sets when buffers should be filled
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Camera init failed with error 0x%x", err);
        return err;
    }
    
    
    sensor_t *s = esp_camera_sensor_get();
    s->set_vflip(s, 1);//flip it back
    //initial sensors are flipped vertically and colors are a bit saturated

    // s->set_sharpness(s, 2);
    // s->set_awb_gain(s, 2);
    return ESP_OK;
}
httpd_uri_t uri_get = {
    .uri = "/",
    .method = HTTP_GET,
    .handler = jpg_stream_httpd_handler,
    .user_ctx = NULL};
    
static esp_err_t jpg_httpd_handler(httpd_req_t *req)
{
  camera_fb_t *fb = NULL;
  esp_err_t res = ESP_OK;
  size_t fb_len = 0;
  int64_t fr_start = esp_timer_get_time();

  fb = esp_camera_fb_get();
  if (!fb)
  {
    ESP_LOGE(TAG, "Camera capture failed");
    httpd_resp_send_500(req);
    return ESP_FAIL;
  }
  res = httpd_resp_set_type(req, "image/jpeg");
  if (res == ESP_OK)
  {
    res = httpd_resp_set_hdr(req, "Content-Disposition", "inline; filename=capture.jpg");
  }

  if (res == ESP_OK)
  {
    fb_len = fb->len;
    res = httpd_resp_send(req, (const char *)fb->buf, fb->len);
  }
  esp_camera_fb_return(fb);
  int64_t fr_end = esp_timer_get_time();
  ESP_LOGI(TAG, "JPG: %ldKB %ldms", (uint32_t)(fb_len / 1024), (uint32_t)((fr_end - fr_start) / 1000));
  return res;
}
```

me-no-dev, 2023-10-02:
you can try to lower the XCLK, since 20MHz might be a bit too fast for the esp to be able to process all data from the sensor. High resolutions are generally much more possible with JPEG mode sensors. We have success grabbing a 5MP images with JPEG.

skylli, 2023-10-06:
I have been experiencing continuous "cam_hal: EV-EOF-OVF" errors, and I am unable to capture images even when I reduce XCLK. According to the sc101iot.c file, the SC101IOT sensor only supports the PIXFORMAT_YUV422 format.
```c

static int set_pixformat(sensor_t *sensor, pixformat_t pixformat)
{
    int ret=0;
    sensor->pixformat = pixformat;

    switch (pixformat) {
    case PIXFORMAT_RGB565:
    case PIXFORMAT_RAW:
    case PIXFORMAT_GRAYSCALE:
        ESP_LOGE(TAG, "Not support");
        break;
    case PIXFORMAT_YUV422: // For now, sc101 sensor only support YUV422.
        break;
    default:
        ret = -1;
    }

    return ret;
}
```
 I am curious to know how testing was done to support a resolution of 1280 x 720 for the SC101IOT sensor.

WangYuxin-esp, 2023-10-09:
Sorry for the late reply. Yes, ESP32S3 currently supports SC101iot. Here are some [instructions](https://github.com/espressif/esp32-camera/blob/master/sensors/private_include/sc101iot_settings.h#L56) after testing.
Please note that you must use ESP32S3 instead of ESP32, as the DVP interface of ESP32 has a low speed and is not very suitable for handling YUV422 format data.

skylli, 2023-10-22:
Thank you for your response. I've been using frame_size=FRAMESIZE_HD (1280x720), and it's working, but I'm only getting 0.5 frames per second (fps). Below is my code. Can you please provide guidance on how to improve the fps?
```
    .pin_pclk = CAM_PIN_PCLK,

        .xclk_freq_hz = 10000000,//6000000,//CONFIG_XCLK_FREQ,
        .ledc_timer = LEDC_TIMER_0, // This is only valid on ESP32/ESP32-S2. ESP32-S3 use LCD_CAM interface.
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = PIXFORMAT_YUV422,//PIXFORMAT_YUV422, //PIXFORMAT_YUV422,//PIXFORMAT_JPEG,
        .frame_size = FRAMESIZE_HD, //FRAMESIZE_SVGA, // FRAMESIZE_QQVGA,       //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

        // .jpeg_quality = 12,  //0-63 lower number means higher quality
        // .fb_count = 2,      //if more than one, i2s runs in continuous mode. Use only with JPEG

        // .grab_mode = CAMERA_GRAB_LATEST
        .jpeg_quality = 50,//10,
        .fb_count = 1,
        .fb_location = CAMERA_FB_IN_PSRAM,
        .grab_mode = CAMERA_GRAB_LATEST,
        .conv_mode = YUV422_TO_RGB565,
```

WangYuxin-esp, 2023-10-23:
Yes, because the data size in YUV422 format is relatively large, the maximum frame rate may only reach 3fps when working at 720p resolution. If you need a higher resolution, it is recommended to choose a sensor that can output JPEG format.
In the current configuration, I suggest setting the `fb_count` to 2, which may lead to an improvement in frame rate.

skylli, 2023-10-23:
Thank you for your response. 3fps is sufficient for my product, but when I set xclk_freq_hz to 1000000 and fb_count to 2, I'm only getting 0.2fps, and I keep receiving EV-EOP-OVF warnings in the logs as shown below:
[image]

I'm curious about how you achieved 3fps. Below is my configuration:
```c
esp_err_t init_camera(void)
{
    camera_config_t detection_config = {
        .pin_pwdn  = CAM_PIN_PWDN,
        .pin_reset = CAM_PIN_RESET,
        .pin_xclk = CAM_PIN_XCLK,
        .pin_sccb_sda = CAM_PIN_SIOD,
        .pin_sccb_scl = CAM_PIN_SIOC,

        .pin_d7 = CAM_PIN_D7,
        .pin_d6 = CAM_PIN_D6,
        .pin_d5 = CAM_PIN_D5,
        .pin_d4 = CAM_PIN_D4,
        .pin_d3 = CAM_PIN_D3,
        .pin_d2 = CAM_PIN_D2,
        .pin_d1 = CAM_PIN_D1,
        .pin_d0 = CAM_PIN_D0,
        .pin_vsync = CAM_PIN_VSYNC,
        .pin_href = CAM_PIN_HREF,
        .pin_pclk = CAM_PIN_PCLK,

        .xclk_freq_hz = 1000000,//6000000,//CONFIG_XCLK_FREQ,
        .ledc_timer = LEDC_TIMER_0, // This is only valid on ESP32/ESP32-S2. ESP32-S3 use LCD_CAM interface.
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = PIXFORMAT_YUV422,//PIXFORMAT_YUV422, //PIXFORMAT_YUV422,//PIXFORMAT_JPEG,
        .frame_size = FRAMESIZE_HD, //FRAMESIZE_SVGA, // FRAMESIZE_QQVGA,       //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

        // .jpeg_quality = 12,  //0-63 lower number means higher quality
        // .fb_count = 2,      //if more than one, i2s runs in continuous mode. Use only with JPEG

        // .grab_mode = CAMERA_GRAB_LATEST
        .jpeg_quality = 20,//10,
        .fb_count = 2,
        .fb_location = CAMERA_FB_IN_PSRAM,
        .grab_mode = CAMERA_GRAB_LATEST,
        .conv_mode = YUV422_TO_RGB565,
        };//CAMERA_GRAB_LATEST. Sets when buffers should be filled
    
    
    
    esp_err_t err = esp_camera_init(&detection_config);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "detect Camera init failed with error 0x%x", err);
        return err;
    }
    sensor_t *s = esp_camera_sensor_get();
    s->set_vflip(s, 1);//flip it back
    return ESP_OK;
}
```

I would greatly appreciate your guidance on achieving the desired 3fps with the specified parameters.

WangYuxin-esp, 2023-10-24:
From the test example, it seems that you have also used the JPEG encoding function. Unfortunately, the current JPEG encoding is implemented by software, and its encoding rate for 720p resolution images is limited.

WangYuxin-esp, 2023-10-24:
If you need a high resolution and frame rate, I suggest using a sensor with jpeg encoding function. You can also follow our new chip, ESP32-P4, which will have a hardware encoder and support cameras with MIPI-CSI and DVP interfaces.

skylli, 2023-10-26:
Thank you very much for your patient answer..
[image]

closed 2023-10-26 by skylli