espressif/esp32-camera #575: VSYNC in ESP32 OV5640 Camera driver

Abhay-2412, 2023-10-04:
Why ESP32 taking two VSYNC cycles for giving DMA interrupt to read image in case of frame buffer count 1?

and in case of frame buffer count 2, it is taking one VSYNC.

closed 2023-10-04 by me-no-dev

Abhay-2412, 2023-10-05:
Thank you for the explanation. I just wanted to clarify one point. When you mentioned 'two VSYNCs,' did you mean that the buffer waits for the next VSYNC after returning, so it appears as if there are two VSYNCs in quick succession? I want to make sure I understand the timing correctly. Could you please provide some additional details on how this process works? Thank you

me-no-dev, 2023-10-05:
if you return the buffer right after VSYNC, it will look like it takes two VSYNCs to start getting the frame. Think of it like this: when you return the frame, it waits for the next VSYNC and starts then. Can not start in the middle of the frame