espressif/esp32-camera #577: ESP32 Camera driver VSYNC query

Abhay-2412, 2023-10-05:
We are new to ESP32 and Camera. We are facing issue in getting VSYNC event. We observed that if we change fb_count (frame buffer count) value the VSYNC behavior is changing.
Our understanding is, VSYNC event is received when capture is completed and captured data is available in DMA for processing.
We observed that:
1. If fb_count = 1, it is taking 2 VSYNC events  to raise DMA event.
2. If fb_count = 2, it is taking 1 VSYNC event to raise the DMA event.

We observed same behavior with ESP32 Wrover and ESP32-S3 Wroom.
We  observed this on Oscilloscope.

two queries are there:
1. Can you please help us to understand this behavior? 
2. why 2 VSYNC events are required to get DMA interrupt?

me-no-dev, 2023-10-05:
Please do not post new issues for the same question.

me-no-dev, 2023-10-05:
> Our understanding is, VSYNC event is received when capture is completed and captured data is available in DMA for processing.

VSYNC is received when the camera STARTS streaming, not when it's done.

closed 2023-10-05 by me-no-dev