espressif/esp32-camera #579: Support OV426

Pjcaogthb, 2023-10-06:
Hi,

How can I add OV426 to ESP32 camera component?

I can pay some money for the labor.

Thanks.
PJ

WangYuxin-esp, 2023-10-09:
It is not difficult to add the OV426 driver in the current framework, but camera manufacturers need to provide register configuration files for the corresponding sensor. You can submit a business application to the [official website](https://www.espressif.com/en/home).

WangYuxin-esp, 2023-10-09:
Of course, if you already have a register configuration for the corresponding sensor, please provide the attachment (including register configuration and technical manual for the sensor), and I can write a test example for you.

Pjcaogthb, 2023-10-09:
It looks like we need to generate a file similar to ov2640.c, for example ov426.c
Add ov426 in sensor.h for camera_pid_t in sensors.h:
typedef enum {
    OV9650_PID = 0x96,
    OV7725_PID = 0x77,
    _**OV426_PID =0x3C**_
    OV2640_PID = 0x26,
    OV3660_PID = 0x3660,
    OV5640_PID = 0x5640,
    OV7670_PID = 0x76,
    NT99141_PID = 0x1410,
    GC2145_PID = 0x2145,
    GC032A_PID = 0x232a,
    GC0308_PID = 0x9b,
    BF3005_PID = 0x30,
    BF20A6_PID = 0x20a6,
    SC101IOT_PID = 0xda4a,
    SC030IOT_PID = 0x9a46,
    SC031GS_PID = 0x0031,
} camera_pid_t;
And also other related data structure: such as camera_model_t, all the way to  framesize_t.

I use Elipse ESP-IDF, the espressif__esp32-camera is added through IDF component manager automatically and is not allow for any modifications. It is controlled by this line in the idf_component.yml file:
## IDF Component Manager Manifest File
dependencies:
  ##espressif/esp32-camera: "==2.0.5"

I commented the line and rename the folder "managed_components" to "components" and it allow me to modify the files in the folder.

I am not sure if I am in a correct approach, or there is a better way to do this.

Thanks.
PJ

Pjcaogthb, 2023-11-16:
Hi,

The sensor output data is in Bayer pattern, what does " const bool
support_jpeg; " mean in "camera_sensor_info_t"
structure?

The following are screenshots from OV2640 datasheet, and it doesn't say
support jpeg. The OV426 has the same data pattern and I should also set it
to "True" for this field as following:
{CAMERA_OV2640, "OV2640", OV2640_SCCB_ADDR, OV2640_PID, FRAMESIZE_UXGA,
true},
{CAMERA_OV426, "OV426", OV426_SCCB_ADDR, OV426_PID, FRAMESIZE_OV426, true},

Is that correct?

Thank you very much for the support!
PJ

[image: image.png]

me-no-dev, 2023-11-16:
I did not read anywhere in what little info I found about OV426 that mentions image compression (JPEG). @Pjcaogthb are you positive that it's supported?

Pjcaogthb, 2023-11-16:
I agree with you. From the public OV426 datasheet, I didn't find any JPEG
support information.

Does the program also support OV426 output data, which is in the raw Bayer
pattern? Or I have to convert it into JPEG format before sending it to
ESP32?

Thanks.
PJ

me-no-dev, 2023-11-17:
I'm not familiar with the bayer output format. What is it? How many bytes per pixel?

Pjcaogthb, 2023-11-17:
According to OV2640 datasheet, it is the 8/10 bit raw RGB data. If you use
8 bits, then one byte per pixel. See Table 2 on its datasheet:
[image: image.png]
The odd row is blue and green alternatively and the even row is for green
and red color.

Thanks.
PJ

me-no-dev, 2023-11-17:
for 1 byte per pixel, you could use the GRAYSCALE mode like OV3360, if 2 bytes per pixel, then RGB565 should do.
As long as the other signals are also ok, you should be able to grab frames.

lance-js, 2024-01-30:
Hi @me-no-dev is there any update on providing official support for the OV426 in this package? I am using the OV426 so I would prefer to use the official package rather than maintain a local fork of it.

me-no-dev, 2024-01-30:
@lance-js we do not have the sensor, but you are free to create a pull request with your implementation and we can discuss and merge it?

Pjcaogthb, 2024-04-01:
> @Pjcaogthb as you mentioned you had a working implementation with the OV426? Would you be create a pull request and I'd be happy to review (along with @me-no-dev hopefully) and I can help test on my custom hardware board. I am still in progress working on adding the OV426 driver support and would prefer not to reinvent the wheel. The OV426 is a bit different as its not a sensor in its itself as it just an ISP and you can connect either OVM6946 or OVM6948 camera sensors to it. It also has a fixed 10 bit DVP interface, whereas this package is only currently supporting sensors/ISP's with a 8 bit DVP interface, so this case need special handling.

I am a new user to Github. What is a pull request? How should I create it ?

lance-mlabs, 2024-04-02:
@Pjcaogthb to create a pull request I believe you need to fork this repo then commit the changes and the file a pull request against this repository. Instructions here: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

closed 2024-06-09 by github-actions[bot]

typetrade, 2025-02-28:
> I am a new user to Github. What is a pull request? How should I create it ?

Where can I see your code？

Pjcaogthb, 2025-02-28:
I am not a frequent user. I don't know either.

Thanks.
PJ