espressif/esp32-camera #580: Esp32s3 LCD_ Camera cannot trigger a DMA interrupt (IDFGH-11190)

planeboy817, 2023-10-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Esp32s3 LCD_ Camera cannot trigger a DMA interrupt
In the case of esp32s3 camera, it was not possible to trigger the DMA interrupt and LCD_ Camera peripheral's vs pin interrupt, look for ll_ Cam. h did not find any errors and cannot find answers to related issues online. May I ask if it is due to the lack of flash and PSRAM versions of the chip in ESP32s3, which do not have DMA functionality, or other bugs （idf    version 5.1）

`static esp_err_t ll_cam_dma_init(cam_obj_t *cam)
{
    for (int x = (SOC_GDMA_PAIRS_PER_GROUP - 1); x >= 0; x--) {
        if (GDMA.channel[x].in.link.addr == 0x0) {
            cam->dma_num = x;
            ESP_LOGI(TAG, "DMA Channel=%d", cam->dma_num);
            break;
        }
        if (x == 0) {
            cam_deinit();
            ESP_LOGE(TAG, "Can't found available GDMA channel");
			return ESP_FAIL;
        }
    }

    if (REG_GET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_DMA_CLK_EN) == 0) {
        REG_CLR_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_DMA_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_DMA_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_DMA_RST);
        REG_CLR_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_DMA_RST);
    }

    GDMA.channel[cam->dma_num].in.int_clr.val = ~0;
    GDMA.channel[cam->dma_num].in.int_ena.val = 0;

    GDMA.channel[cam->dma_num].in.conf0.val = 0;
    GDMA.channel[cam->dma_num].in.conf0.in_rst = 1;
    GDMA.channel[cam->dma_num].in.conf0.in_rst = 0;

    //internal SRAM only
    if (!cam->psram_mode) {
        GDMA.channel[cam->dma_num].in.conf0.indscr_burst_en = 1;
        GDMA.channel[cam->dma_num].in.conf0.in_data_burst_en = 1;
    }

    GDMA.channel[cam->dma_num].in.conf1.in_check_owner = 0;
    // GDMA.channel[cam->dma_num].in.conf1.in_ext_mem_bk_size = 2;

    GDMA.channel[cam->dma_num].in.peri_sel.sel = 5;
    //GDMA.channel[cam->dma_num].in.pri.rx_pri = 1;//rx prio 0-15
    //GDMA.channel[cam->dma_num].in.sram_size.in_size = 6;//This register is used to configure the size of L2 Tx FIFO for Rx channel. 0:16 bytes, 1:24 bytes, 2:32 bytes, 3: 40 bytes, 4: 48 bytes, 5:56 bytes, 6: 64 bytes, 7: 72 bytes, 8: 80 bytes.
    //GDMA.channel[cam->dma_num].in.wight.rx_weight = 7;//The weight of Rx channel 0-15
    return ESP_OK;
}
’

’esp_err_t ll_cam_config(cam_obj_t *cam, const camera_config_t *config)
{
    esp_err_t ret = ESP_OK;
    if (REG_GET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_LCD_CAM_CLK_EN) == 0) {
        REG_CLR_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_LCD_CAM_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_CLK_EN1_REG, SYSTEM_LCD_CAM_CLK_EN);
        REG_SET_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_LCD_CAM_RST);
        REG_CLR_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_LCD_CAM_RST);
        // printf("地址：%d\n",DR_REG_SYSTEM_BASE);
    }
    /*lcd摄像头的一些寄存器初始化操作 dma传输和dma中断的设置*/
    printf("配置摄像头\n");
    // lcd_cam_lc_reg_date.val=33566752;
    LCD_CAM.cam_ctrl.val = 0;

    LCD_CAM.cam_ctrl.cam_clkm_div_b = 0;
    LCD_CAM.cam_ctrl.cam_clkm_div_a = 0;
    LCD_CAM.cam_ctrl.cam_clkm_div_num = 160000000 / config->xclk_freq_hz;
    LCD_CAM.cam_ctrl.cam_clk_sel = 3;//Select Camera module source clock. 0: no clock. 1: APLL. 2: CLK160. 3: no clock.

    LCD_CAM.cam_ctrl.cam_stop_en = 0;
    LCD_CAM.cam_ctrl.cam_vsync_filter_thres = 4; // Filter by LCD_CAM clock
    LCD_CAM.cam_ctrl.cam_update = 0;
    LCD_CAM.cam_ctrl.cam_byte_order = cam->swap_data;
    LCD_CAM.cam_ctrl.cam_bit_order = 0;
    LCD_CAM.cam_ctrl.cam_line_int_en = 0;
    LCD_CAM.cam_ctrl.cam_vs_eof_en = 0; //1: CAM_VSYNC to generate in_suc_eof. 0: in_suc_eof is controlled by reg_cam_rec_data_cyclelen

    LCD_CAM.cam_ctrl1.val = 0;
    LCD_CAM.cam_ctrl1.cam_rec_data_bytelen = LCD_CAM_DMA_NODE_BUFFER_MAX_SIZE - 1; // Cannot be assigned to 0, and it is easy to overflow
    LCD_CAM.cam_ctrl1.cam_line_int_num = 0; // The number of hsyncs that generate hs interrupts
    LCD_CAM.cam_ctrl1.cam_clk_inv = 0;
    LCD_CAM.cam_ctrl1.cam_vsync_filter_en = 1;
    LCD_CAM.cam_ctrl1.cam_2byte_en = 0;
    LCD_CAM.cam_ctrl1.cam_de_inv = 0;
    LCD_CAM.cam_ctrl1.cam_hsync_inv = 0;
    LCD_CAM.cam_ctrl1.cam_vsync_inv = 0;
    LCD_CAM.cam_ctrl1.cam_vh_de_mode_en = 0;

    LCD_CAM.cam_rgb_yuv.val = 0;

#if CONFIG_CAMERA_CONVERTER_ENABLED
    if (config->conv_mode) {
        ret = ll_cam_converter_config(cam, config);
        if(ret != ESP_OK) {
            return ret;
        }
    }
#endif

    LCD_CAM.cam_ctrl.cam_update = 1;
    LCD_CAM.cam_ctrl1.cam_start = 1;

    ret = ll_cam_dma_init(cam);

    return ret;
}’
’void ll_cam_vsync_intr_enable(cam_obj_t *cam, bool en)
{
    LCD_CAM.lc_dma_int_clr.cam_vsync_int_clr = 1;
    if (en) {
        LCD_CAM.lc_dma_int_ena.cam_vsync_int_ena = 1;
        //xiamianjiachulaide
        LCD_CAM.lc_dma_int_ena.lcd_vsync_int_ena = 1;
         printf("中断使能\n");
    } else {
         #**printf("中断失能\n");**
        LCD_CAM.lc_dma_int_ena.cam_vsync_int_ena = 0;
    }
}`

me-no-dev, 2023-10-09:
This issue is not very clear to me, but ESP32-S3 does have DMA and does work fine with the camera when PSRAM is not found (if it's buffers can fit into internal RAM). You did not say which sensor you are using or what settings you used to start the camera driver, so I am unable to help any further than that.

WangYuxin-esp, 2023-10-09:
I tested the esp32-camera example on the v5.1 version of IDF and it worked properly.
IDF commit: 012fafb82781
Therefore, this issue should not be a version compatibility issue. I suggest checking if the sensor is working properly.

planeboy817, 2023-10-21:
> This issue is not very clear to me, but ESP32-S3 does have DMA and does work fine with the camera when PSRAM is not found (if it's buffers can fit into internal RAM). You did not say which sensor you are using or what settings you used to start the camera driver, so I am unable to help any further than that.
> 
> 这个问题对我来说不是很清楚，但是 esp32-s3确实有 DMA，并且在没有找到 PSRAM (如果它的缓冲区可以放入内部 RAM)的情况下也能正常工作。你没有说你正在使用哪个传感器，或者你用什么设置来启动相机驱动程序，所以我不能帮助任何进一步的。

Okay, I'll try again. Thank you for your reply

planeboy817, 2023-10-21:
> I tested the esp32-camera example on the v5.1 version of IDF and it worked properly. IDF commit: 012fafb82781 Therefore, this issue should not be a version compatibility issue. I suggest checking if the sensor is working properly.
> 
> 我在 IDF 的 v 5.1版本上测试了 esp32-camera 示例，它工作正常。IDF commit: 012fa fb82781因此，这个问题不应该是版本兼容性问题。我建议检查传感器是否工作正常。

I used a logic analyzer and the test results showed that the camera was working normally, but it was still unable to trigger the vs interrupt and DMA interrupt

WangYuxin-esp, 2023-10-23:
Can this device work properly when only using CAM?

planeboy817, 2023-10-23:
> Can this device work properly when only using CAM?

The device works normally, d0-d7 should be able to return data, but the dma transfer does not seem to have started, the ll_cam.c file has not been modified

planeboy817, 2023-10-23:
> What is the frequency of the pixel clock when you connected the LA?
The xclk pin is connected to a 12M crystal oscillator, with the XCLK value set to -1 and the pixel clock set to 12M

closed 2023-12-29 by github-actions[bot]