espressif/esp32-camera #582: Capturing frame and send it to the application within one second

Abhay-2412, 2023-10-11:
Hello,I am currently working on developing a functionality that involves utilizing the capabilities of an OV5640 camera to capture frames. The objective is to transmit these frames to an application running on a specific IP address via a WiFi connection. To achieve this, I am employing the `client.write()` method to facilitate the transmission of the captured images.

```
void capturePhoto( WiFiClient client ) {
  int ret = 1;
  sensor_t * s = esp_camera_sensor_get();
  camera_fb_t * fb = NULL; // pointer
  bool ok = 0; // Boolean indicating if the picture has been taken correctly

  do {

      fb = esp_camera_fb_get();

    if (!fb) {
      Serial.println("Camera capture failed");
      return;
    }
    // check if file has been correctly saved in SPIFFS
    ok = checkPhoto(fb);
    if (ok)
    {
      client.print("begin");
      client.write(fb->buf,fb->len);
      client.print("end");
      esp_camera_fb_return(fb);
    }
  } while ( !ok );
}
```

In my ongoing project, I have observed that I am able to successfully receive images in my application. These images are captured by an OV5640 camera and consist of frames depicting a running stopwatch. However, during my analysis, I have noticed that while most frames are received within a time frame of 750 milliseconds, there are occasional instances where the processing time extends to 1500 milliseconds. Consequently, I suspect that these delays might be causing the loss of certain frames.

To provide some context, I am currently utilizing four frame buffers in my implementation. Furthermore, I have been employing the client.write() function for transmitting the captured frames. Although most of the transmission processes take less than a second, there have been a few exceptional cases where the transmission time exceeded 9 seconds.

Considering these observations, I would greatly appreciate any assistance in confirming whether the usage of the `client.write()` function could potentially impact the frame buffer mechanism, thereby leading to frame loss. Additionally, I am open to suggestions and alternatives that could help mitigate this issue. Thank you for your valuable input!

Abhay-2412, 2023-10-12:
I am currently facing a significant challenge and require assistance. I am experiencing inconsistencies in frame retrieval, occasionally missing three or more frames. I suspect that the `client.write()` function may be causing the slowdown. I would greatly appreciate it if someone could confirm this suspicion or suggest any alternative approaches. Thank you kindly for your support.

KrazyIvan-, 2023-12-05:
client.write() sure can, its dead lock untill send all data or timeout
750 milliseconds(5Mpix? image) mean you try to pushe like 500Kb file, defenetly hard for esp32 via TCP at once (because of how TCP works, TCP congestion...) 
to find if thats your problem
try : send(client->fd(), (void*) fb->buf,fb->len, MSG_DONTWAIT);  // you will suffer from fail transffer of image
[instead this line: client.write(fb->buf,fb->len);]

Abhay-2412, 2023-12-06:
I am trying to send images of YUV422 format with a frame size of 640x240, and I am also applying a crop. The data size is about 307.2 KB. We were wondering if we can able to acheive it if we use ESP32-S3 instead of the ESP32 for this purpose.

KrazyIvan-, 2023-12-06:
it should not be a problem.

307.2 KB should be able to get within less than 200ms including wifi data send.
I have no idea how you ended up with 750ms and 1500ms, maybe because you saved in SPIFFS?

It's better to test every step of your code and find where to get with this huge delay. Definitely not an ESP32-S3 issue

Abhay-2412, 2023-12-06:
We are currently facing challenges in achieving the goal of transferring 480 frames within a 480-second timeframe. Despite implementing logging around the client.write() operation, we decided to skip some frames which have successfully mitigated spikes, the objective remains to consistently attain the transmission of each frame within a one-second interval.

closed 2024-02-12 by github-actions[bot]