espressif/esp32-camera #583: sc031gs: frame size err

huangdiajd, 2023-10-11:
Using the SC031GS camera, the frame_size was found to be wrong. How large frame_size should be set for SC031GS. I see that the error message should be set to 200*200, but I set 174*144, which is not working. Of course, I also tried to change 174*144 to 200*200, but the same error will appear. How should I change the appropriate frame_size

I (2715) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (2725) s3 ll_cam: DMA Channel=4
I (2735) cam_hal: cam init ok
I (2735) sccb: pin_sda 21 pin_scl 14
I (2745) sccb: sccb_i2c_port=1
I (2755) camera: Detected camera at address=0x30
I (2755) ov2640: Mismatch PID=0x0
I (2755) camera: Detected SC031GS camera
I (2755) camera: Camera PID=0x31 VER=0x00 MIDL=0x00 MIDH=0x00
I (2845) s3 ll_cam: node_size: 2500, nodes_per_line: 1, lines_per_node: 25
I (2845) s3 ll_cam: dma_half_buffer_min:  2500, dma_half_buffer: 10000, lines_per_half_buffer: 100, dma_buffer_size: 10000
I (2855) cam_hal: buffer_size: 10000, half_buffer_size: 10000, node_buffer_size: 2500, node_cnt: 4, total_cnt: 1
I (2865) cam_hal: Allocating 10016 Byte frame buffer in PSRAM
I (2875) cam_hal: Frame[0]: Offset: 16, Addr: 0x3C0D0A50
I (2875) cam_hal: Allocating 10016 Byte frame buffer in PSRAM
I (2885) cam_hal: Frame[1]: Offset: 16, Addr: 0x3C0D3190
I (2885) cam_hal: cam config ok
E (2895) sc031gs: Only support 200*200 for now, contact us if you want to use other resolutions
E (2905) sc031gs: frame size err
E (2905) camera: Failed to set frame size
E (2915) ESP32-CAM text: Camera Init Failed

After changing it to 200*200, I connected to the WiFi, but I could not connect to the website 192.168.4.1, which would cause an error

app task running...
app task running...
app task running...
app task running...
app task running...
I (19845) wifi:<ba-add>idx:3 (ifx:1, 32:50:13:6a:f7:fd), tid:6, ssn:19, winSize:64
app task running...
W (19935) cam_hal: Failed to get the frame on time!
E (19935) camera http: Camera capture failed
W (19935) httpd_uri: httpd_uri: uri handler execution failed

WangYuxin-esp, 2023-10-11:
Please use the new [branch](https://github.com/espressif/esp32-camera/pull/581) for testing.

huangdiajd, 2023-10-11:
I changed it to VGA, and then it gave an error when connecting to the web page. Can you help me see what is the reason
app task running...
app task running...
app task running...
I (15775) wifi:<ba-add>idx:3 (ifx:1, 32:50:13:6a:f7:fd), tid:6, ssn:16, winSize:64
W (16175) cam_hal: Failed to get the frame on time!
E (16175) camera http: Camera capture failed
W (16175) httpd_uri: httpd_uri: uri handler execution failed

> Please use the new [branch](https://github.com/espressif/esp32-camera/pull/581) for testing.

WangYuxin-esp, 2023-10-11:
The current xclk can be used is 10M.

huangdiajd, 2023-10-12:
> The current xclk can be used is 10M.
I have changed xclk from 16M to 10M, but it will show the following message

I (2724) sccb: pin_sda 21 pin_scl 14
I (2734) sccb: sccb_i2c_port=1
I (2744) camera: Detected camera at address=0x30
I (2744) ov2640: Mismatch PID=0x0
I (2744) camera: Detected SC031GS camera
I (2744) camera: Camera PID=0x31 VER=0x00 MIDL=0x00 MIDH=0x00
I (3074) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (3074) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (3074) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 20
I (3084) cam_hal: Allocating 307200 Byte frame buffer in PSRAM
I (3094) cam_hal: Allocating 307200 Byte frame buffer in PSRAM
I (3104) cam_hal: cam config ok
I (3174) ESP32-CAM text: Camera Init OK.
I (3174) main_task: Returned from app_main()
app task running...
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF

huangdiajd, 2023-10-12:
> The current xclk can be used is 10M.
This is my camera configuration 
// 摄像头配置
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D9,
    .pin_d6 = CAM_PIN_D8,
    .pin_d5 = CAM_PIN_D7,
    .pin_d4 = CAM_PIN_D6,
    .pin_d3 = CAM_PIN_D5,
    .pin_d2 = CAM_PIN_D4,
    .pin_d1 = CAM_PIN_D3,
    .pin_d0 = CAM_PIN_D2,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //OV5640 XCLK 16MHz   OV2640 20  
    .xclk_freq_hz = 10000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_GRAYSCALE, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_VGA,    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 12, //0-63 lower number means higher quality
    .fb_count = 2,      //if more than one, i2s runs in continuous mode. Use only with JPEG
    .grab_mode = CAMERA_GRAB_LATEST
};
#endif

WangYuxin-esp, 2023-10-12:
Please take a schematic diagram of the connection between the development board and the sensor you are using. I suspect this may be a hardware issue, as there were no issues with my testing here.

huangdiajd, 2023-10-12:
/
[image]
我最后只能将xclk改到6M，但是这个帧率只有5fps了。然后xclk怎么才能提升或者说031gs这个帧率如何提升.

WangYuxin-esp, 2023-10-12:
May I ask what frame rate do you expect to achieve at VGA resolution?
I think this high-speed sensor signal line needs to be made very well. The current situation is likely a hardware issue.

huangdiajd, 2023-10-12:
> May I ask what frame rate do you expect to achieve at VGA resolution? I think this high-speed sensor signal line needs to be made very well. The current situation is likely a hardwar
目前只有 5fps, 应该要20 fps 。031GS在VGA 下能最大达到多少帧？不过目前我在网页上看到的图像只有纯白色，不形成灰白图像，这个是因为XCLK的原因么？

WangYuxin-esp, 2023-10-12:
20fps 可以的。目前应该已经有 20fps 了。只是你的 XCLK 低导致帧率未及预期。
输出的应是灰度图。不是 XCLK 的原因，是镜头、sensor 打样的问题。这个 sensor 完美的工作需要自己调试自动曝光、增益参数。
你可以通过 
`sensor_t *s = esp_camera_sensor_get();
    s->set_reg(0x0211, 0x0b); `
来控制 sensor 的寄存器。

huangdiajd, 2023-10-13:
> 20fps 可以的。目前应该已经有 20fps 了。只是你的 XCLK 低导致帧率未及预期。 输出的应是灰度图。不是 XCLK 的原因，是镜头、sensor 打样的问题。这个 sensor 完美的工作需要自己调试自动曝光、增益参数。 你可以通过 `sensor_t *s = esp_camera_sensor_get(); s->set_reg(0x0211, 0x0b); ` 来控制 sensor 的寄存器。
[image]
我现在将寄存器增益// {0x3e06, 0x0c},改成{0x3e06, 0xc8}, ，可以达到上图的效果，但是画面有很多毛刺，应该修改哪些寄存器，才能让画质流畅。		
[image]
这是 我的031手册，我应该看哪些寄存器部分，然后修改。

WangYuxin-esp, 2023-10-16:
If the data is already available, even if it is bad, then ESP32 has completed its mission. You can communicate with the technical personnel of the original sensor factory regarding the subsequent image tuning work.

closed 2023-10-18 by huangdiajd

ftbleed, 2023-10-20:
SC031GS可以直接接上ESP32-S3-EYE调试吗？ @WangYuxin-esp

WangYuxin-esp, 2023-10-23:
This is not easy, because this high-speed sensor SC031GS requires a very short FPC. I have integrated SC031GS on my development board.

Eduardo-Brandt, 2023-11-13:
> SC031GS可以直接接上ESP32-S3-EYE调试吗？ @WangYuxin-esp

Very cool work from WangYuxin-esp!, it really enables 640x480 resolution on SC031GS. 
But I´ve got distorced images at 640x480 resolution. **Can it be because of the high PCLK(Pixel Clock) of 25MHz? Any suggestion?**

See some results I´ve gotten:
pic1: 
[image]
pic2: 
[image]
pic3: 
[image]

With lower resolutions, the results are better:
[image]

WangYuxin-esp, 2023-11-14:
> > SC031GS可以直接接上ESP32-S3-EYE调试吗？ @WangYuxin-esp
> 
> Very cool work from WangYuxin-esp!, it really enables 640x480 resolution on SC031GS. But I´ve got distorced images at 640x480 resolution. **Can it be because of the high PCLK(Pixel Clock) of 25MHz? Any suggestion?**
> 
> See some results I´ve gotten (Pictures from my oscilloscope): pic1: [image] pic2: [image] pic3: [image]
> 
> With lower resolutions, the results are better: [image]

If the situation is the same with lowering the xclk, it may be a power issue.

Eduardo-Brandt, 2023-11-16:
I have discovered that the problem was because of the BUS cable lenght. Mainly because of the PCLK (Pixel Clock) pin, that carries about 25MHz clock. I got very better results when I shortened the cables from 10cm to about 15mm of lenght. The 3.3V to 1.8V converters may worsen the problem also.