espressif/esp32-camera #585: cam_hal: EV-EOF-OVF error while cropping

Abhay-2412, 2023-10-17:
Iam trying to get a custom crop(640x240) of capture by my 5MP ov5640 camera but constantly getting `cam_hal: EV-EOF-OVF` below are the logs.

```
16:55:24.707 -> Connected to Abhay iphone13
16:55:24.707 -> IP address: 172.20.10.4
16:55:25.701 -> LED OFF
16:55:25.701 -> WiFi connected
16:55:26.416 -> E (6327) cam_hal: CONFIG_IDF_TARGET_ESP32
16:55:26.416 -> E (6327) cam_hal:  not in cam_obj->jpeg_mode
16:55:26.416 -> E (6327) esp32 ll_cam: node_size: 1280, nodes_per_line: 4, lines_per_node: 1, dma_half_buffer_min:  5120, dma_half_buffer: 40960,lines_per_half_buffer:  8, dma_buffer_size: 81920, image_size: 307200
16:55:26.461 -> E (6341) cam_hal: buffer_size: 81920, half_buffer_size: 40960, node_buffer_size: 1280, node_cnt: 64, total_cnt: 7
16:55:26.493 -> WiFi connected
16:55:26.493 -> OV5640 camera is connected
16:55:26.493 -> Starting web server on port: '80'
16:55:26.541 -> Starting stream server on port: '81'
16:55:26.541 -> Camera Ready! Use 'http://172.20.10.4' to connect
16:55:27.034 -> cam_hal: EV-EOF-OVF
16:55:27.437 -> cam_hal: EV-EOF-OVF
16:55:27.853 -> cam_hal: EV-EOF-OVF
16:55:28.248 -> cam_hal: EV-EOF-OVF
16:55:28.669 -> cam_hal: EV-EOF-OVF
16:55:29.091 -> cam_hal: EV-EOF-OVF
16:55:29.499 -> cam_hal: EV-EOF-OVF
```
I used below snippet for cropping

` s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2560, 1920, 640, 240, false, false);`

and one more thing i observed is when i try below cropping its working fine

` s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 5120, 1920, 640, 240, false, false);`

But logically my camerea is 5MP why that crop is not working and why above crop is working

me-no-dev, 2023-10-18:
Windowing on that sensor is rather complicated and numbers rarely make sense. I applaud the fact that you got it working, generally that is what I did for all other resolutions too. OV2640, for example, makes much more sense.

Abhay-2412, 2023-10-19:
And one more thing  `s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2560, 1920, 640, 240, false, false);` working in esp32s3 without any error so i was wondering is there any technical reason that its working on esp32 s3 but not on esp32

WangYuxin-esp, 2023-10-23:
The DVP interface of ESP32 allows a relatively low rate. Currently, ESP32S3 is recommended for most projects that require the use of cameras.
Also, I always have a question, why choose a camera that can output a resolution of 5M if the resolution you need is 640x240? Choosing an SC030iot or BF3005 will be better and more cost-effective.

KrazyIvan-, 2023-12-04:
try  on esp32
s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2592, 790, 640, 240, false, false);
or
s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2592, 272, 640, 240, false, false);

Abhay-2412, 2023-12-05:
Hello @KrazyIvan- Below are the logs for the crop coordinates you suggested

s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2592, 272, 640, 240, false, false);

```
14:08:11.112 -> WiFi connected
14:08:11.112 -> OV5640 camera is connected
14:08:11.152 -> Starting web server on port: '80'
14:08:11.152 -> Starting stream server on port: '81'
14:08:11.152 -> Camera Ready! Use 'http://172.20.10.5' to connect
14:08:11.307 -> cam_hal: EV-EOF-OVF
14:08:11.380 -> E (6652) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.413 -> cam_hal: EV-EOF-OVF
14:08:11.479 -> E (6769) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.553 -> E (6828) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.623 -> E (6887) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.671 -> E (6945) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.712 -> cam_hal: EV-EOF-OVF
14:08:11.788 -> E (7063) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.819 -> E (7122) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.885 -> E (7180) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.929 -> cam_hal: EV-EOF-OVF
14:08:12.005 -> cam_hal: EV-VSYNC-OVF
14:08:12.111 -> cam_hal: EV-EOF-OVF
14:08:12.193 -> E (7474) cam_hal: FB-SIZE: 307200 != 614400
14:08:12.231 -> cam_hal: EV-EOF-OVF
14:08:12.305 -> E (7592) cam_hal: FB-SIZE: 307200 != 614400
14:08:12.338 -> cam_hal: EV-EOF-OVF
14:08:12.417 -> cam_hal: EV-VSYNC-OVF
```
s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2592, 790, 640, 240, false, false);

```
14:08:11.152 -> Camera Ready! Use 'http://172.20.10.5' to connect
14:08:11.307 -> cam_hal: EV-EOF-OVF
14:08:11.380 -> E (6652) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.413 -> cam_hal: EV-EOF-OVF
14:08:11.479 -> E (6769) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.553 -> E (6828) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.623 -> E (6887) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.671 -> E (6945) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.712 -> cam_hal: EV-EOF-OVF
14:08:11.788 -> E (7063) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.819 -> E (7122) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.885 -> E (7180) cam_hal: FB-SIZE: 307200 != 614400
14:08:11.929 -> cam_hal: EV-EOF-OVF
14:08:12.005 -> cam_hal: EV-VSYNC-OVF
14:08:12.111 -> cam_hal: EV-EOF-OVF
14:08:12.193 -> E (7474) cam_hal: FB-SIZE: 307200 != 614400
14:08:12.231 -> cam_hal: EV-EOF-OVF
14:08:12.305 -> E (7592) cam_hal: FB-SIZE: 307200 != 614400
14:08:12.338 -> cam_hal: EV-EOF-OVF
14:08:12.417 -> cam_hal: EV-VSYNC-OVF
14:08:12.537 -> cam_hal: EV-EOF-OVF
14:08:12.610 -> E (7885) cam_hal: FB-SIZE: 307200 != 614400
14:08:12.680 -> E (7944) cam_hal: FB-SIZE: 307200 != 614400
```

KrazyIvan-, 2023-12-05:
what is your config?

KrazyIvan-, 2023-12-05:
1000, 550, 1640, 790, 0, 0, 2592, 790, 640, 240, false, false
works perfect. 30ms per frame at 20Mhz xclock size 27772bytes

[image]
[image]

Abhay-2412, 2023-12-05:
hey thanks for the details which `pixel_format` you have used if possible please share your camera config settings

KrazyIvan-, 2023-12-05:
camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sccb_sda = SIOD_GPIO_NUM;
  config.pin_sccb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = -1 ;//PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  config.frame_size = FRAMESIZE_UXGA;
  config.jpeg_quality = 4;
 
 esp_camera_init(&config);

you can swich resolution without reset/stop/reinit camamera at any time just :
s->set_framesize(s,  FRAMESIZE_VGA);
or 
s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2592, 790, 640, 240, false, false); 
or
s->set_res_raw(s, 1000, 550, 1640, 790, 0, 0, 2592, 272, 640, 240, false, false); // 2ms/frame but noise image

Chip is ESP32-D0WD-V3, also work with ESP32-D0WD

closed 2024-02-12 by github-actions[bot]